libimobiledevice/idevicerestore #645: I can't restore my MacBook Air M1 Late 2020

eilhanzy, 2024-06-11:
Hello there, thanks in advance for helping me. I got a issue from idevicerestore and I can't restore my device from Raspberry Pi 5. There is a log for this issue:

`idevicerestore 1.0.0-191-g4ed598b
Found device in DFU mode
ECID: 1521845238169630
Identified device as j313ap, MacBookAir10,1
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 14.5
IPSW Product Build: 23F79 Major: 23
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Performing macOS restore
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in dfu mode... d1 9e 6a 05 41 ca 65 6f 47 0f d8 db c1 b0 a7 3c 68 75 43 9b 1e 73 eb fa 98 02 e1 cd 93 88 57 d2 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 3a bb d8 31 25 a1 cf e1 26 b5 2e e8 6d da c9 fc 76 de 50 e4 
Received SHSH blobs
Getting SepNonce in dfu mode... 3a bb d8 31 25 a1 cf e1 26 b5 2e e8 6d da c9 fc 76 de 50 e4 
Received SHSH blobs
Getting SepNonce in dfu mode... 3a bb d8 31 25 a1 cf e1 26 b5 2e e8 6d da c9 fc 76 de 50 e4 
Received SHSH blobs
Extracting iBSS.j313.RELEASE.im4p (Firmware/dfu/iBSS.j313.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Sending iBSS (730801 bytes)...
[==================================================] 100.0%
Nonce: d1 9e 6a 05 41 ca 65 6f 47 0f d8 db c1 b0 a7 3c 68 75 43 9b 1e 73 eb fa 98 02 e1 cd 93 88 57 d2 
Personalizing IMG4 component Ap,LocalPolicy...
Sending Ap,LocalPolicy (3041 bytes)...
[==================================================] 100.0%
INFO: executing command: lpolrestore
iBoot boot-stage=1
Extracting t8103ciofw.im4p (Firmware/t8103ciofw.im4p)...
Personalizing IMG4 component Ap,RestoreCIO...
Sending Ap,RestoreCIO (172825 bytes)...
[==================================================] 100.0%
INFO: executing command: firmware
Extracting t8103tmufw.im4p (Firmware/t8103tmufw.im4p)...
Personalizing IMG4 component Ap,RestoreTMU...
Sending Ap,RestoreTMU (21358 bytes)...
[==================================================] 100.0%
INFO: executing command: firmware
Extracting rans.t8103.release.im4p (Firmware/rans.t8103.release.im4p)...
Personalizing IMG4 component RestoreANS...
Sending RestoreANS (1000676 bytes)...
[==================================================] 100.0%
INFO: executing command: firmware
Extracting ipad13dcp_restore.im4p (Firmware/dcp/ipad13dcp_restore.im4p)...
Personalizing IMG4 component RestoreDCP...
Sending RestoreDCP (9574385 bytes)...
[==================================================] 100.0%
INFO: executing command: firmware
INFO: executing command: setenv auto-boot false
INFO: executing command: saveenv
INFO: executing command: setenvnp boot-args rd=md0 nand-enable-reformat=1 -progress -restore
Extracting applelogo@2x~mac-USBc.im4p (Firmware/all_flash/applelogo@2x~mac-USBc.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (18677 bytes)...
[==================================================] 100.0%
INFO: executing command: setpicture 4
INFO: executing command: bgcolor 0 0 0
Extracting iBEC.j313.RELEASE.im4p (Firmware/dfu/iBEC.j313.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (544794 bytes)...
[==================================================] 100.0%
INFO: device serial number is C02HG32ZQ6L4
Recovery Mode Environment:
iBoot build-version=iBoot-10151.121.1
iBoot build-style=RELEASE
iBoot boot-stage=2
radio-error=2
radio-error-string=2
Sending RestoreLogo...
Extracting applelogo@2x~mac-USBc.im4p (Firmware/all_flash/applelogo@2x~mac-USBc.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (18677 bytes)...
Extracting h13_ane_fw_styx_j5x.im4p (Firmware/ane/h13_ane_fw_styx_j5x.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (4925425 bytes)...
Extracting aopfw-mac13gaop.RELEASE.im4p (Firmware/AOP/aopfw-mac13gaop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (1329137 bytes)...
Extracting AppleAVE2FW_H13G.im4p (Firmware/ave/AppleAVE2FW_H13G.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1373353 bytes)...
Extracting ipad13dcp.im4p (Firmware/dcp/ipad13dcp.im4p)...
Personalizing IMG4 component Ap,DCP2...
Sending Ap,DCP2 (9574385 bytes)...
Extracting armfw_g13g.im4p (Firmware/agx/armfw_g13g.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (2932489 bytes)...
Extracting adc-pallas-j129.im4p (Firmware/isp_bni/adc-pallas-j129.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (13805553 bytes)...
Extracting t8103pmp.im4p (Firmware/pmp/t8103pmp.im4p)...
Personalizing IMG4 component PMP...
Sending PMP (415729 bytes)...
Extracting 090-25576-099.dmg.trustcache (Firmware/090-25576-099.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (19992 bytes)...
Extracting SmartIOFirmware_ASCv4.im4p (Firmware/SmartIOFirmware_ASCv4.im4p)...
Personalizing IMG4 component SIO...
Sending SIO (1163717 bytes)...
Extracting iBootData.j313.RELEASE.im4p (Firmware/all_flash/iBootData.j313.RELEASE.im4p)...
Personalizing IMG4 component iBootData...
Sending iBootData (20094 bytes)...
ramdisk-size=0x20000000
Extracting 090-25576-099.dmg (090-25576-099.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (161490932 bytes)...
Extracting DeviceTree.j313ap.im4p (Firmware/all_flash/DeviceTree.j313ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (52245 bytes)...
Extracting sep-firmware.j313.RELEASE.im4p (Firmware/all_flash/sep-firmware.j313.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (6336710 bytes)...
Extracting kernelcache.release.mac13g (kernelcache.release.mac13g)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (27536070 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device 00008103-0005681C34D3001E has successfully entered restore mode
Hardware Information:
BoardID: 38
ChipID: 33027
UniqueChipID: 1521845238169630
ProductionMode: true
Previous restore exit status: 0x100
Starting Reverse Proxy
idevicerestore: symbol lookup error: idevicerestore: undefined symbol: reverse_proxy_client_create_with_port
`

Duckmilk, 2024-07-07:
@eilhanzy hey buddy, i currently having the same issues with my m1 pro mbp, were u able to figure out what the issue was?

itspluxstahre, 2024-07-19:
> @eilhanzy hey buddy, i currently having the same issues with my m1 pro mbp, were u able to figure out what the issue was?

I found a solution... The package in Fedora 40 (dnf install idevicerestore) that is based on this commit https://github.com/libimobiledevice/idevicerestore/tree/609f7f058487596597e8e742088119fdd46729df was able to restore my mbp m1 pro fully with latest macos