libimobiledevice/idevicerestore #647: iOS 18 Not Supported

keikei14, 2024-06-12:
Hello, I am trying to update my iPhone 11 to iOS 18 using the IPSW files from the official website and it fails. If it helps, I am on Windows 11.
Installation log here:
```
C:\Users\Keian\Desktop\idevicerestore-latest_x86_64-mingw64>idevicerestore.exe -e C:\Users\Keian\Desktop\aria2-1.37.0-win-64bit-build1\iPhone12,1_18.0_22A5282m_Restore.ipsw
idevicerestore 1.0.0-git-4ed598b-dirty
Found device in Recovery mode
ECID: 7987995979530286
Identified device as n104ap, iPhone12,1
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 18.0
IPSW Product Build: 22A5282m Major: 22
INFO: device serial number is FFXJ281HN73D
Device supports Image4: true
Variant: Developer Erase Install (IPSW)
This restore will erase all device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... b8 59 2c c3 da 54 a8 6c 98 9d ba 64 2e 97 16 18 07 9d 5d 9c eb 43 17 4a d1 ff 71 8a 09 a5 92 8f
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 03 b9 1c 3e 6e cd a4 88 48 b7 f9 43 ae b8 31 bc d2 0f cc 0d
Received SHSH blobs
Extracting iBEC.n104.RELEASE.im4p (Firmware/dfu/iBEC.n104.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (1563894 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-11881.0.80.502.1
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@1792~iphone.im4p (Firmware/all_flash/applelogo@1792~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (20404 bytes)...
Extracting h12_ane_fw_metis.im4p (Firmware/ane/h12_ane_fw_metis.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (4874913 bytes)...
Extracting aopfw-iphone12baop.RELEASE.im4p (Firmware/AOP/aopfw-iphone12baop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (2171553 bytes)...
Extracting AppleAVE2FW_H12.im4p (Firmware/ave/AppleAVE2FW_H12.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1405305 bytes)...
Extracting armfw_g12p.im4p (Firmware/agx/armfw_g12p.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (1229033 bytes)...
Extracting adc-zelus-n104.im4p (Firmware/isp_bni/adc-zelus-n104.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (14983841 bytes)...
Extracting t8030pmp.im4p (Firmware/pmp/t8030pmp.im4p)...
Personalizing IMG4 component PMP...
Sending PMP (307873 bytes)...
Extracting 090-27766-071.dmg.trustcache (Firmware/090-27766-071.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (16036 bytes)...
Extracting SmartIOFirmware_ASCv2.im4p (Firmware/SmartIOFirmware_ASCv2.im4p)...
Personalizing IMG4 component SIO...
Sending SIO (1162045 bytes)...
Extracting WirelessPower.iphone12b.im4p (Firmware/WirelessPower/WirelessPower.iphone12b.im4p)...
Personalizing IMG4 component WCHFirmwareUpdater...
Sending WCHFirmwareUpdater (62480 bytes)...
ramdisk-size=0x20000000
Extracting 090-27766-071.dmg (090-27766-071.dmg)...
[... 672 lines cut ...]
restore-step-user-progress = 0
Considering saving tolerated failures: false (null) false
[18:17:04.0030-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030655:16;0x11030647:19}
restore-step-names = {0x11030655:is_host_compatible;0x11030647:cleanup_check_result}
restore-step-uptime = 11
restore-step-user-progress = 0
[18:17:04.0031-GMT]{3>6} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030655:16}
restore-step-names = {0x11030655:is_host_compatible}
restore-step-uptime = 11
restore-step-user-progress = 0
Considering saving tolerated failures: false (null) false
[18:17:04.0031-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030655:16;0x11030648:20}
restore-step-names = {0x11030655:is_host_compatible;0x11030648:cleanup_send_final_status}
restore-step-uptime = 11
restore-step-user-progress = 0

ERROR: Unable to successfully restore device
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(This host version is unsupported. You may need to update your host tools.)
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device
```

doronz88, 2024-06-12:
I also stumbled into that while testing with `pymobiledevice3`.
There are several milestones required for ios18 updates to work:

- Add support for the new async requests
- Since the addition of the new AEA format, we require to send GET/POST requests for the device during both ASR and the Cryptex writes so the encrypted archived can be decoded as stream and written back to the device on-the-fly.
- Possibly more. Still haven't gotten the ASR to fully work.

I'll keep updates regarding how I progress and the protocol changes.

My WIP PR can be found in here:
https://github.com/doronz88/pymobiledevice3/pull/1062/files

doronz88, 2024-06-13:
I'm currently struggling with handling the `StreamedImageDecryptionKey` data request.
If anyone has an idea as to how should the client respond to the following request, please update:

```pyhon
{'DataType': 'StreamedImageDecryptionKey',
 'DataPort': 49172,
 'Arguments': {'RequestURL': 'https://wkms.sd.apple.com/keys/fetch',
  'RequestAdditionalHeaders': {'Content-Type': 'application/json',
   'Accept': 'application/json'},
  'ArchiveIdentifier': '6bb460bfb79c6aed51b48e09fc445c5df6cfaa4dd046d9b8522942b326f68a80',
  'RequestMethod': 'POST',
  'RequestBody': b'{"auth-data":"Sv+DAwEBC2FlYUF1dGhEYXRhAf+EAAEEAQpTZWFsZWRBdHRyAQoAAQNBQUQBCgABCUtleUZhbWlseQEMAAEHVmVyc2lvbgEEAAAA\\/gId\\/4QB\\/gFuQDu3NFbMSGIapa6bXTyrDPrAw9HAzicAn8W26sEQ5E9EPnB9aZKsaYTrKCoXfUwH2pLTmhpQ+DS4qF\\/Cy+LlYbvajtcfZd5+eM818UH2dQ0dyCMHvtvHwUbzcj19xFK5\\/DdoEM78Y6mqWCrQv5+YTRR4ccQLlYgwI6N4zARMlDx70LvqKuSQ3e6lTT9OkveZ2IxZtiJiR2OuKaH1DnkFHuXwLpaxghjgoT2yd4fThoE5cnTo9QHWiPpaBOFkdWMR4CxboPfYwIB8O1ylnvR3+8gN7LtcAW+XqvT+VCVk1IZq24cYFSvVGcpyNw4g2\\/zBKJLY1R5s\\/5OHoUL6sZ974u\\/Us4WIOz\\/X5iY00lCAUExgsvzEc9T6Q3KFQnCTKd3AJxYgO\\/HDBwY\\/dRd9e7xT+AM4pjhL9ibYtGIPqE3f\\/AGgrABLiy4CdCfq8vh36PmN5XbHYDTimyBivBojySMFHDXifjXNQlymd\\/5YGpNWAf+CBBkgyV9SAoSQqi+vLhHq4gFW53QZE\\/LpdbeyUrQpKv0TZyWhuYJRbC6J1wflPN8C1sMw2jUOREVHaXi8dC2PKRkgmbq5ns7OFmkMsCARdP77mT+b8k2D0xV8+Zxn9U40wwYVP68B7M2aWP+\\/pKrNOPwY9IaGbp9p7ycX6pemyAEpEwEfSUFTLURpc2tJbWFnZXMyLVNBS1MtQ29tcGF0aWJsZQECAA==","cert-chain":"MIIBpzCCAU2gAwIBAgIBATAKBggqhkjOPQQDAjBdMTowOAYDVQQDEzFjb20uYXBwbGUucmVzdG9yZWQuSW50ZXJuYWxLZXlXcmFwcGluZ0NlcnRpZmljYXRlMQswCQYDVQQGEwJVUzESMBAGA1UEChMJQXBwbGUgSW5jMB4XDTI0MDYxMzA3MjEyMFoXDTI0MDYxMzA3MjEyMFowXTE6MDgGA1UEAxMxY29tLmFwcGxlLnJlc3RvcmVkLkludGVybmFsS2V5V3JhcHBpbmdDZXJ0aWZpY2F0ZTELMAkGA1UEBhMCVVMxEjAQBgNVBAoTCUFwcGxlIEluYzBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABKWIXkEOs5gfAtda3pvNmoazFB8EHyf1UxSISOLqepXgLFg8eAheVrvzE3TUF4R0li2RQjx6Az8UAZRpOMoq1kMwCgYIKoZIzj0EAwIDSAAwRQIhALI+iQbUPx8l9UMWXOdVYa5UIQnOP4dRjBjJBlAPUjQqAiAtO7zPswJGGfT+ANYUsX8eroLdte\\/1IE1n8qH1zn9qqQ=="}'},
 'MsgType': 'AsyncDataRequestMsg'}
```

The `RequestURL` (https://wkms.sd.apple.com/keys/fetch) isn't reachable.

codebar33, 2024-06-13:
Hi @doronz88,

This might be what you are looking for: https://github.com/dhinakg/aeota

nikias, 2024-06-13:
looks like it uses `https://wkms-public.apple.com/fcs-keys` for the request URL

doronz88, 2024-06-13:
Thanks but I already know of the AEA format.

It turns out the device sends the `StreamedImageDecryptionKey` only if the previous `URLAsset` failed, which I replied incorrectly. It sends this request:

```python
{'DataType': 'URLAsset', 'DataPort': 49172, 'Arguments': {'RequestMethod': 'GET', 'RequestURL': 'https://wkms-public.apple.com/fcs-keys/C76OEoiX5Lfc0nRQtn1cLkOEwDtC8HGIM_M_1rJgQ9g='}, 'MsgType': 'AsyncDataRequestMsg'}
```

And the response is as follows:

```python
{'ResponseBody': b'-----BEGIN PRIVATE KEY-----\nMIGHAgEAMBMGByqGSM49AgEGCCqG'
                 b'SM49AwEHBG0wawIBAQQgmjX0pbe6Ya+02TzN\ncIZXvz/UHwYL7RpQQdk'
                 b'MPWZfOe+hRANCAATsyK1dBsPRUeMyohV3eIPnI4l6K8cP\nVxfFEpDwMC'
                 b'useMEkWE3WL9Aw/M32FNSkiXeCiAz11pNuBUXefNAOIydI\n-----END '
                 b'PRIVATE KEY-----\n',
 'ResponseBodyDone': True,
 'ResponseHeaders': {'Accept-Ranges': 'bytes',
                     'Age': '0',
                     'Cache-Control': 'max-age=600, public',
                     'Content-Length': '241',
                     'Content-Type': 'binary/octet-stream',
                     'Date': 'Mon, 10 Jun 2024 23:21:01 GMT',
                     'Etag': '"b751b41d14623536ba95cdab7bd34550"',
                     'Last-Modified': 'Mon, 10 Jun 2024 19:10:07 GMT',
                     'Server': 'AmazonS3',
                     'Strict-Transport-Security': 'max-age=31536000; '
                                                  'includeSubdomains',
                     'Via': 'https/1.1 usdal4-edge-lx-011.ts.apple.com '
                            '(acdn/153.14426), https/1.1 '
                            'usdal4-edge-bx-027.ts.apple.com (acdn/153.14426)',
                     'cdnuuid': 'c9c3f46b-48ef-4c2b-b4df-6aaddea00785-5254568688',
                     'x-amz-id-2': '9XtNxkFI8aomHCjA63B19baWKD6dWmi0eB/9Uv8smUVo93LA+Y+5WPvE8Frz8Z9Ri7ecYkvKZwQ=',
                     'x-amz-request-id': '6WZDWMKM884ETEH6',
                     'x-amz-server-side-encryption': 'AES256',
                     'x-amz-version-id': 'sFuFSQz7fdn_5tpwDBCn0GDs2GrkDlHC',
                     'x-cache': 'hit-stale, hit-stale'},
 'ResponseStatus': 200}
```

Once I constructed it well, it followed through. Now encountering another error I'm trying to figure out 😅

doronz88, 2024-06-13:
I got till the `Cryptex1,SystemOS` part.
The device requests the following:
```python
{'MsgType': 'DataRequestMsg',
 'DataType': 'SourceBootObjectV4',
 'Arguments': {'Variant': 'Erase', 'ImageName': 'Cryptex1,SystemOS'}}
```
Then I send it the first chunk of size 8192, In response I'm requested to get the decryption key:
```python
{'DataType': 'URLAsset',
 'DataPort': 49193,
 'Arguments': {'RequestMethod': 'GET',
  'RequestURL': 'https://wkms-public.apple.com/fcs-keys/C76OEoiX5Lfc0nRQtn1cLkOEwDtC8HGIM_M_1rJgQ9g='},
 'MsgType': 'AsyncDataRequestMsg'}
```
Which I reply, then finish sending the rest of the cryptex.

However, after it's done I get the following:
```python
{'CHECKPOINT_COMPLETE': True,
 'CHECKPOINT_RESULT': 0,
 'CHECKPOINT_ERROR': '[0]D(failed to install splat)',
 'CHECKPOINT_NAME': 'cleanup_send_final_status',
 'CHECKPOINT_ID': 1608,
 'MsgType': 'CheckpointMsg'}
```

In restored logs I can find
```
Transfer: successfuly unwrapped fcs key.
Transfer: Receiving 8192 bytes, writing 4020240384 bytes
Transfer: Preallocated 4020240384 bytes for /mnt9/cryptex1/current/os.dmg
Transfer: Received 1577058304 of 8192 bytes from host
Transfer: Wrote 4020240384 of 4020240384 bytes to /mnt9/cryptex1/current/os.dmg
Transfer: Unlinking output file /mnt9/cryptex1/current/os.dmg due to error 1112
ramrod_splat_install: install handler failed
[14:43:21.0750-GMT]{3>6} CHECKPOINT FAILURE:(FAILURE:100) (null):[0x06A6] install_splat [0]D(failed to install splat)
```

Any ideas as to what could be the problem?

doronz88, 2024-06-16:
Got it working! :) You're welcome to have a look at my existing PR
Still refactoring it

nikias, 2024-06-16:
Amazing, good job @doronz88