libimobiledevice/idevicerestore #650: idevicerestore: error while loading shared libraries: libirecovery-1.0.so.5: cannot open shared object file: No such file or directory

Aceitunator, 2024-06-25:
After spending hours dealing with broken dependencies (needing at some point an older libimobiledevice-glue-dev, needing an older libimobiledevice-glue needing an older libplist blablabla).

I got this:

user@debian:~$ idevicerestore
idevicerestore: error while loading shared libraries: libirecovery-1.0.so.5: cannot open shared object file: No such file or directory

Maybe this is why the "year of linux" never come but the non userfriendly hell of linux.

At this point im not looking for a fix. Just leaving a stupid advice.

Kydix, 2024-07-23:
same problem for me ...

federoRE, 2024-11-23:
Too many libraries to install that I forgot to run that command. Thanks.