libimobiledevice/idevicerestore #651: iOS 18 beta: fetch Savage ticket error

iFred09, 2024-06-25:
Hello,

I have an error when trying to update my iPhone to iOS 18 beta with the latest build of idevicerestore.

`idevicerestore 1.0.0-199-g28c1dab
Found device in Normal mode
ECID: 626314411458588
Identified device as d83ap, iPhone16,1
Device Product Version: 18.0
Device Product Build: 22A5282m
Extracting BuildManifest from IPSW
IPSW Product Version: 18.0
IPSW Product Build: 22A5297f Major: 22
Device supports Image4: true
Variant: Developer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Checking if device requires stashbag...
Device requires stashbag.
******************************************************************************
* Please enter your passcode on the device.  The device will store a token   *
* that will be used after restore to access the user data partition.  This   *
* prevents an 'Attempting data recovery' process occurring after reboot that *
* may take a long time to complete and will _also_ require the passcode.     *
******************************************************************************
Stashbag created.
Getting ApNonce in normal mode... 20 6c ce 39 c5 bf 11 b1 a1 e7 3c f9 af f7 f4 9c 0d 69 be 9d 8e 84 48 f2 10 d8 3b 68 ee 6e 3d ab
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... fe a3 1c ab da 63 7e a2 8d 4f 76 2b d7 27 16 9d 48 e6 7a ce
Received SHSH blobs
Committing stashbag...
Stashbag committed!
Entering recovery mode...
INFO: device serial number is N7HNW73FCY
Extracting iBEC.d83.RELEASE.im4p (Firmware/dfu/iBEC.d83.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (1579239 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-11881.0.193.0.1
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2556~iphone.im4p (Firmware/all_flash/applelogo@2556~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (29262 bytes)...
Extracting h16x_ane_fw_iaso_d8x.im4p (Firmware/ane/h16x_ane_fw_iaso_d8x.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (1632115 bytes)...
Extracting aopfw-iphone16aop.RELEASE.im4p (Firmware/AOP/aopfw-iphone16aop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (3209075 bytes)...
Extracting AppleAVE2FW_H16.im4p (Firmware/ave/AppleAVE2FW_H16.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1546027 bytes)...
Extracting sptm.t8122.release.im4p (Firmware/sptm.t8122.release.im4p)...
Personalizing IMG4 component Ap,RestoreSecurePageTableMonitor...
Sending Ap,RestoreSecurePageTableMonitor (155901 bytes)...
Extracting txm.iphoneos.release.im4p (Firmware/txm.iphoneos.release.im4p)...
Personalizing IMG4 component Ap,RestoreTrustedExecutionMonitor...
Sending Ap,RestoreTrustedExecutionMonitor (156044 bytes)...
Extracting armfw_g16p.im4p (Firmware/agx/armfw_g16p.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (4212955 bytes)...
Extracting adc-aceso-d8x.im4p (Firmware/isp_bni/adc-aceso-d8x.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (25388918 bytes)...
Extracting D83_MtpFirmware.im4p (Firmware/D83_MtpFirmware.im4p)...
Personalizing IMG4 component MtpFirmware...
Sending MtpFirmware (6878237 bytes)...
Extracting t8130pmp.im4p (Firmware/pmp/t8130pmp.im4p)...
[... 414 lines cut ...]
Checkpoint completed id: 0x131F (update_canary) result=0
Checkpoint started   id: 0x130D (update_savage)
Updating Savage (60)
Sending ReceiptManifest data now...
Done sending ReceiptManifest data
About to send BuildIdentity Dict...
Sending BuildIdentityDict now...
Done sending BuildIdentityDict
About to send Savage,BA-Prod-Patch...
Sending Savage,BA-Prod-Patch now (114 bytes)
Done sending Savage,BA-Prod-Patch
About to send Savage,BE-Prod-Patch...
Sending Savage,BE-Prod-Patch now (514 bytes)
Done sending Savage,BE-Prod-Patch
Sending Savage TSS request...
ERROR: Unable to fetch Savage ticket
ERROR: restore_send_firmware_updater_data: Couldn't get Savage firmware data
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device`

nikias, 2024-06-25:
Can you run with `-dd`? We want to see the TSS request that is failing.

iFred09, 2024-06-25:
here it is :

```
Sending BuildIdentityDict now...
Done sending BuildIdentityDict
restore_handle_data_request_msg: type = SourceBootObjectV4
DEBUG: restore_send_source_boot_object_v4: Got SourceBootObjectV4 request:
printing 438 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>MsgType</key>
	<string>DataRequestMsg</string>
	<key>DataType</key>
	<string>SourceBootObjectV4</string>
	<key>Arguments</key>
	<dict>
		<key>ImageName</key>
		<string>Savage,BA-Prod-Patch</string>
		<key>IsRecoveryOS</key>
		<false/>
	</dict>
</dict>
</plist>
About to send Savage,BA-Prod-Patch...
DEBUG: tss_response_get_path_by_entry: No entry 'Savage,BA-Prod-Patch' in TSS response
NOTE: No path for component Savage,BA-Prod-Patch in TSS, will fetch from build identity
Sending Savage,BA-Prod-Patch now (114 bytes)
Done sending Savage,BA-Prod-Patch
restore_handle_data_request_msg: type = SourceBootObjectV4
DEBUG: restore_send_source_boot_object_v4: Got SourceBootObjectV4 request:
printing 438 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>MsgType</key>
	<string>DataRequestMsg</string>
	<key>DataType</key>
	<string>SourceBootObjectV4</string>
	<key>Arguments</key>
	<dict>
		<key>ImageName</key>
		<string>Savage,BE-Prod-Patch</string>
		<key>IsRecoveryOS</key>
		<false/>
	</dict>
</dict>
</plist>
About to send Savage,BE-Prod-Patch...
DEBUG: tss_response_get_path_by_entry: No entry 'Savage,BE-Prod-Patch' in TSS response
NOTE: No path for component Savage,BE-Prod-Patch in TSS, will fetch from build identity
Sending Savage,BE-Prod-Patch now (514 bytes)
Done sending Savage,BE-Prod-Patch
restore_handle_data_request_msg: type = FirmwareUpdaterData
DEBUG: restore_send_firmware_updater_data: Got FirmwareUpdaterData request:
printing 2327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>MsgType</key>
	<string>DataRequestMsg</string>
	<key>DataType</key>
	<string>FirmwareUpdaterData</string>
	<key>Arguments</key>
	<dict>
		<key>MessageArgInfo</key>
		<dict>
[... 174 lines cut ...]
< Server: Apple
< Date: Tue, 25 Jun 2024 18:18:06 GMT
< Content-Type: text/html
< Content-Length: 69
< Connection: keep-alive
< Host: gs.apple.com
< Strict-Transport-Security: max-age=31536000; includeSubdomains
< X-Frame-Options: SAMEORIGIN
<
* Connection #0 to host gs.apple.com left intact
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
ERROR: Unable to fetch Savage ticket
ERROR: restore_send_firmware_updater_data: Couldn't get Savage firmware data
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
ERROR: Unable to send data to ASR. Sent 107784 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
No data to read (timeout)
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device
```

nikias, 2024-06-26:
I pushed a commit that will use the `DeviceGeneratedRequest` to work around this problem for now.