libimobiledevice/idevicerestore #655: ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.

felipedbene, 2024-07-05:
I am getting this on my 2023 Macbook Pro M2, that machine has the hard-drive wipped and I just wanted to install Somona again.

➜  ~ idevicerestore -e --latest -d
idevicerestore 1.0.0-202-g26613f9
irecv_event_cb: device 001c30603ec0c01e (udid: N/A) connected in Recovery mode
Found device in Recovery mode
ECID: 7934489275711518
Identified device as j414sap, Mac14,9
Device Product Version: N/A
Device Product Build: N/A
* Host api.ipsw.me:443 was resolved.
* IPv6: 2606:4700:20::681a:89c, 2606:4700:20::681a:99c, 2606:4700:20::ac43:47bb
* IPv4: 172.67.71.187, 104.26.9.156, 104.26.8.156
*   Trying [2606:4700:20::681a:89c]:443...
* Connected to api.ipsw.me (2606:4700:20::681a:89c) port 443
* ALPN: curl offers h2,http/1.1
* SSL connection using TLSv1.3 / AEAD-CHACHA20-POLY1305-SHA256 / [blank] / UNDEF
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=ipsw.me
*  start date: Jun 30 00:35:28 2024 GMT
*  expire date: Sep 28 00:35:27 2024 GMT
*  subjectAltName: host "api.ipsw.me" matched cert's "*.ipsw.me"
*  issuer: C=US; O=Google Trust Services; CN=WE1
*  SSL certificate verify ok.
* using HTTP/2
* [HTTP/2] [1] OPENED stream for https://api.ipsw.me/v4/device/Mac14,9
* [HTTP/2] [1] [:method: GET]
* [HTTP/2] [1] [:scheme: https]
* [HTTP/2] [1] [:authority: api.ipsw.me]
* [HTTP/2] [1] [:path: /v4/device/Mac14,9]
* [HTTP/2] [1] [user-agent: InetURL/1.0 idevicerestore/1.0.0-202-g26613f9]
* [HTTP/2] [1] [accept: */*]
> GET /v4/device/Mac14,9 HTTP/2
Host: api.ipsw.me
User-Agent: InetURL/1.0 idevicerestore/1.0.0-202-g26613f9
Accept: */*

< HTTP/2 200
< date: Fri, 05 Jul 2024 00:24:39 GMT
< content-type: application/json
< vary: Accept-Encoding
< vary: Origin
< access-control-allow-methods: GET, OPTIONS
< access-control-allow-origin: *
< cache-control: public, max-age=14400
< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< expires: Fri, 05 Jul 2024 00:25:39 GMT
< last-modified: Mon, 13 May 2024 17:46:20 GMT
< x-content-type-options: nosniff
< x-data-location: database
< x-frame-options: DENY
< x-frame-options: DENY
< x-xss-protection: 1; mode=block
< strict-transport-security: max-age=15552000; preload
< cf-cache-status: EXPIRED
< report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=9w994oZBcfADRYtpKLyYL5jzC5CEICC0OA4KXgdmiIG6c7VIEGmhSrMRljSCpq1OrF9ija0xAQRRHRVnyMp%2Fqua0tp2hTSNSZT3LA5ICr9o%2BObgJ5X6nE0y5FdxobrgNa2I6JGcGGn7M"}],"group":"cf-nel","max_age":604800}
< nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
< server: cloudflare
< cf-ray: 89e3369d593de153-ORD
< alt-svc: h3=":443"; ma=86400
<
* Connection #0 to host api.ipsw.me left intact
The following firmwares are currently being signed for Mac14,9:
  [1] 14.5 (build 23F79)
  [2] 14.4.1 (build 23E224)
  [3] 14.4 (build 23E214)
  [4] 14.3.1 (build 23D60)
  [5] 14.3 (build 23D56)
  [6] 14.2.1 (build 23C71)
  [7] 14.2 (build 23C64)
  [8] 14.1.2 (build 23B92)
  [9] 14.1.1 (build 23B81)
  [10] 14.1 (build 23B74)
  [11] 14.0 (build 23A344)
  [12] 13.6 (build 22G120)
  [13] 13.5.2 (build 22G91)
  [14] 13.5.1 (build 22G90)
  [15] 13.5 (build 22G74)
  [16] 13.4.1 (build 22F82)
  [17] 13.4 (build 22F66)
  [18] 13.3.1 (build 22E261)
  [19] 13.3 (build 22E252)
  [20] 13.2.1 (build 22D68)
  [21] 13.2 (build 22D49)
Select the firmware you want to restore: 1
Selected firmware 14.5 (build 23F79)
Verifying 'UniversalMac_14.5_23F79_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
IPSW Product Version: 14.5
IPSW Product Build: 23F79 Major: 23
INFO: device serial number is PD6PFPDHPJ
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Performing macOS restore
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... 83 d9 36 52 bd 96 cd ff 65 76 82 82 35 97 05 01 6f 8d ac 40 8a 98 2a 54 f8 e9 3c f4 21 6f f7 c8
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... e3 ff 6f 6f 7c 48 a0 13 38 39 7f 38 95 9c a5 4a fe 56 be 0d
Received SHSH blobs
Getting SepNonce in recovery mode... e3 ff 6f 6f 7c 48 a0 13 38 39 7f 38 95 9c a5 4a fe 56 be 0d
Received SHSH blobs
Getting SepNonce in recovery mode... e3 ff 6f 6f 7c 48 a0 13 38 39 7f 38 95 9c a5 4a fe 56 be 0d
Received SHSH blobs
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.j414s.RELEASE.im4p (Firmware/dfu/iBEC.j414s.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (887516 bytes)...
Waiting for device to disconnect...

Leon-cam, 2024-07-05:
I got the same error with the following debug log when restoring an iPad.

#NOTE: No path for component iBEC in TSS, will fetch from build_identity

Extracting iBEC.j310.RELEASE.im4p (Firmware/dfu/iBEC.j310.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (984692 bytes)...
ERROR: Unable to send iBEC component: Unable to upload data to device
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

felipedbene, 2024-07-05:
In my case I can in the debug it tries to send, it doesnt error at all but the device doesnt seem to be responding. I've been using a generic ThunderBolt 4 cable, I praying this is the issue. I've an official apple as the official procedure suggest, early waiting to test it all again with it.

Leon-cam, 2024-07-22:
I've encountered this issue before. After I updated the libirecovery and rebuilt the whole project, the issue was resolved. Hope that would help.

felipedbene, 2024-07-22:
I've found the reason, I was putting the machine into recovery (not DFU). After setting up dfu correctly, even Apple official tool worked.

osmanbowser, 2024-12-10:
> I've found the reason, I was putting the machine into recovery (not DFU). After setting up dfu correctly, even Apple official tool worked.

You saved me a lot of trouble! I ran into the same issue with the same computer.

For reference, this is how you