libimobiledevice/idevicerestore #656: APFS Image Inversion Error During iPad Restore

Leon-cam, 2024-07-05:
### Description

I encountered an issue while attempting to restore an iPad using the `idevicerestore` tool. The process fails at the step involving the inversion of the APFS image.

### Environment

- **Tool**: idevicerestore
- **Device**: iPad13,18
- **iOS Version**: 17.5.1 (build 21F91)
- **Host OS**:  Windows
- **IPSW File**: iPad_Fall_2022_17.5.1_21F91_Restore.ipsw

### Steps to Reproduce

1. Connect the iPad in Recovery mode.
2. Run the `idevicerestore` command with the appropriate IPSW file.
3. Observe the process fail with the error related to inverting the APFS image.

### Log Output

Below is an excerpt from the log file where the error occurs:
Checkpoint FAILURE id: 0x6B4 result=1101: [0]D(failed to invert APFS image)
Checkpoint FAILURE id: 0x67B result=1101: [0]D(failed to invert APFS image)

The complete log file is attached.
[complete_restore_ios_log.txt](https://github.com/user-attachments/files/16108567/complete_restore_ios_log.txt)

### Additional Information

- The iPad can be restored using iTunes on Windows.

Leon-cam, 2024-07-05:
### Description
This issue was changed after updating the following dependencies from master branches of the following modules :
- libusbmuxd
- libimobiledevice
- libplist
- imobiledevice-glue
- libtatsu
- libirecovery
- idevicerestore

### Errors:

Tag found
Unhandled progress operation 55 (55)
restore_handle_data_request_msg: type = FUDData
restore_handle_data_request_msg: type = FUDData
Tag found
ERROR: Unable to send data to ASR. Sent 0 of 131092 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131092 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131092 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131092 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
ERROR: Unable to restore device

### Log Output
The complete log file is attached.
[restore_ipad_log.txt](https://github.com/user-attachments/files/16111321/restore_ipad_log.txt)

Leon-cam, 2024-07-08:
After making a few tweaks in the code and running idevicerestore with an administrative user on Windows, I encountered the following error:

> Debug: restore_handle_data_request_msg: type = FUDData
info: Extracting 092-64485-001.dmg.mtree (Firmware/092-64485-001.dmg.mtree)...
info: Personalizing IMG4 component Ap,SystemVolumeCanonicalMetadata...
debug: Tag found
info: Sending IsFUDFirmware for Ap,SystemVolumeCanonicalMetadata...
[libusbmuxd] usbmuxd_send: Error 1 when sending: **Operation not permitted**
error: ERROR[-2]: Unable to send data to ASR. Sent 0 of 131092 bytes.
error: Unable to send filesystem payload chunk, retrying...
[libusbmuxd] usbmuxd_send: Error 1 when sending: **Operation not permitted**
error: ERROR[-2]: Unable to send data to ASR. Sent 0 of 131092 bytes.
error: Unable to send filesystem payload chunk, retrying...
[libusbmuxd] usbmuxd_send: Error 1 when sending: **Operation not permitted**
error: ERROR[-2]: Unable to send data to ASR. Sent 0 of 131092 bytes.
error: Unable to send filesystem payload chunk, retrying...
[libusbmuxd] usbmuxd_send: Error 1 when sending: **Operation not permitted**
error: ERROR[-2]: Unable to send data to ASR. Sent 0 of 131092 bytes.
error: Unable to send filesystem payload chunk, retrying...
error: ERROR: Unable to send payload to ASR
error: ERROR: Unable to send filesystem

ERROR -2 means IDEVICE_E_UNKNOWN_ERROR

nikias, 2024-07-08:
Question, have you only tried with the compressed ipsw file? Try if it works with a fully extracted ipsw, just pass the path to the directory with all the files in it to idevicerestore as ipsw "file".

Leon-cam, 2024-07-09:
Yes, previously I only tried with the compressed IPSW file. After specifying the unzipped directory, it successfully went through the restoring process. Many thanks for your information.

nikias, 2024-07-09:
I pushed a fix to master. It was a concurrency issue with zip file access.

adventurousAardvark, 2024-07-10:
Hi @nikias thanks for applying the fix.  Do you know if this patch is universal?  Unfortunately, I am having the same issue with Mac mini (M1) after patch.

I have tried both the compressed ipsw file and the extracted "ipsw" directory. 

IPSW file version: UniversalMac_15.0_24A5279h_Restore.ipsw 

Mac mini M1 running Ventura.

Do you have any other suggestions on how to remedy this issue?  

```
<dict>
	<key>Command</key>
	<string>Payload</string>
</dict>
</plist>
Filesystem validated
Sending filesystem now...
Restoring image (13)
ERROR: Unable to send data to ASR. Sent 130912 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
Checkpoint completed id: 0x675 (init_gigalocker) result=0
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device

```

Thank you!

Leon-cam, 2024-07-11:
Thanks, @nikias! It’s working now without the need to unzip the file beforehand.

closed 2024-08-08 by Leon-cam