libimobiledevice/idevicerestore #658: IPhone XS restoration to iOS 18 beta3 failed, It is stuck in the loop of sending Yonkers TSS and cannot come out

xdayong, 2024-07-15:
[xs_ios18_beta3.log](https://github.com/user-attachments/files/16229849/xs_ios18_beta3.log)

iFred09, 2024-07-17:
hello,

I have the same loop problem when I'm trying to update to iOS 18b3 with the latest build of idevicerestore.
I have a error in the end: 

```
    };
    "Yonkers,SysTopPatch9" =     {
        Digest = {length = 32, bytes = 0x0ab6d636 6ae2ff41 dd21c9ad 621ae038 ... 70cbe6f6 73bd4489 };
        EPRO = 1;
        FabRevision = 54529;
        Info =         {
            IsFTAB = 0;
            IsLoadedByiBoot = 0;
            IsLoadedByiBootStage1 = 0;
            IsiBootEANFirmware = 0;
            IsiBootNonEssentialFirmware = 0;
            Path = "Firmware/Sumter/Sumter.Ymgt_Prod.fw";

_logSinkCallback: updater_log Savage (RestoreInfo): SavageUpdaterCreateRequest [input]: DeviceInfo {
    YonkersDeviceInfo =     {
        "Yonkers,AllowOfflineBoot" = 1;
        "Yonkers,ChipID" = 1300;
        "Yonkers,ECID" = {length = 16, bytes = 0xf17e04d2fcb59cfc86892120be2a38ab};
        "Yonkers,FabRevision" = 54529;
        "Yonkers,Nonce" = {length = 16, bytes = 0x3b89005e4dab83ba32b3f3a14fd4b1c3};
        "Yonkers,ProductionMode" = 1;
        "Yonkers,ReadECKey" = 1;
        "Yonkers,ReadFWKey" = 1;
        YonkersIsProvisioned = 1;
        YonkersIsYmgt = 1;
    };
}

_logSinkCallback: updater_log Savage (RestoreInfo): SavageUpdaterCreateRequest [input]: FirmwareData {
    YonkersFirmware = {length = 776, bytes = 0x02000000 48000000 08030000 01000000 ... 61b2c117 c7ecb233 };
    YonkersFirmwareAlt = {length = 776, bytes = 0x02000000 48000000 08030000 01000000 ... 446706d8 61f433ce };
}

_logSinkCallback: updater_log Savage (RestoreInfo): SavageUpdaterCreateRequest [input]: SavageMeasurementDict - No C string description available.

_logSinkCallback: updater_log Savage (RestoreInfo): SavageUpdaterCreateRequest [input]: YonkersMeasurementDict - {
    MeasurementDictPatch =     {
        Digest = {length = 32, bytes = 0x0ab6d636 6ae2ff41 dd21c9ad 621ae038 ... 70cbe6f6 73bd4489 };
    };
}

_logSinkCallback: updater_log Savage (RestoreInfo): SavageUpdaterCreateRequest [input]: SEP Digest from ReceiptManifest - <CFData 0x79817de90 [0x1088e7798]>{length = 48, capacity = 48, bytes = 0xd0008401340167c1fa9e2515a7b44995 ... 30dad0b1ee5dfedf}

_logSinkCallback: updater_log Savage (RestoreInfo): SavageUpdaterCreateRequest [output]: {
    "@Yonkers,Ticket" = 1;
    "Yonkers,AllowOfflineBoot" = 1;
    "Yonkers,BoardID" = 0;
    "Yonkers,ChipID" = 1300;
    "Yonkers,ECID" = {length = 16, bytes = 0xf17e04d2fcb59cfc86892120be2a38ab};
    "Yonkers,Nonce" = {length = 16, bytes = 0x3b89005e4dab83ba32b3f3a14fd4b1c3};
    "Yonkers,PatchEpoch" = 0;
    "Yonkers,ProductionMode" = 1;
    "Yonkers,ReadECKey" = 1;
    "Yonkers,ReadFWKey" = 1;
    "Yonkers,SysTopPatch9" =     {
        Digest = {length = 32, bytes = 0x0ab6d636 6ae2ff41 dd21c9ad 621ae038 ... 70cbe6f6 73bd4489 };
        EPRO = 1;
        FabRevision = 54529;
    };
}

_update_generic_firmware: response = {
    "@ServerVersion" = "2.1.0";
    FirmwareData =     {
[... 960 lines cut ...]
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[07:30:23.0296-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x1103067B:65;0x11030635:73;0x1103130D:93;0x110306C2:97}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030635:update_device_firmware;0x1103130D:update_savage}
restore-step-uptime = 1504
restore-step-user-progress = 79
send_crash_logs is not supported by the host.
[07:30:23.0296-GMT]{3>6} CHECKPOINT END: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x1103067B:65;0x11030635:73;0x1103130D:93}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030635:update_device_firmware;0x1103130D:update_savage}
restore-step-uptime = 1504
restore-step-user-progress = 79
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[07:30:23.0296-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103067B:65;0x11030635:73;0x1103130D:93;0x11030648:98}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030635:update_device_firmware;0x1103130D:update_savage}
restore-step-uptime = 1504
restore-step-user-progress = 79

ERROR: Unable to successfully restore device
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(failed to update device firmware)[1]D(ramrod firmware update failed)[2]D(_update_generic_firmware: Exceeded maximum loops for Savage)
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device
```

pekn, 2024-07-18:
This happens with iPhone 15 as well.. similar problem, yonkers is just looped.

xxCabin, 2024-07-30:
I also encountered it, have you solved it.

yinyajiang, 2024-08-02:
ios18 beta4 has the same problem

xlzhen-940218, 2024-08-15:
tatsu VersionInfo update to 1028.0.0.0.1 fix loop send Yonkers TSS problem

xdayong, 2024-08-15:
> tatsu VersionInfo update to 1028.0.0.0.1 fix loop send Yonkers TSS problem
Oh, I haven't seen this version

yinyajiang, 2024-08-17:
> tatsu VersionInfo update to 1028.0.0.0.1 fix loop send Yonkers TSS problem

I tried 1033.0.2 and 1028.0.0.0.1, both didn't work on ios beta6.

xlzhen-940218, 2024-08-19:
> > tatsu VersionInfo update to 1028.0.0.0.1 fix loop send Yonkers TSS problem
> 
> I tried 1033.0.2 and 1028.0.0.0.1, both didn't work on ios beta6.

need more code.please see 3utools code

Flo22100, 2024-08-21:
I got the same problem, how do you update tatsu ?

Thanks

yinyajiang, 2024-08-28:
> > > tatsu VersionInfo update to 1028.0.0.0.1 fix loop send Yonkers TSS problem
> > 
> > 
> > I tried 1033.0.2 and 1028.0.0.0.1, both didn't work on ios beta6.
> 
> need more code.please see 3utools code

Do you know what code needs to be modified? I'm not very good at reverse engineering.

irainbw, 2024-08-30:
> tatsu VersionInfo update to 1028.0.0.0.1 fix loop send Yonkers TSS problem

I have the same problem, please tell me how to solve it

xdayong, 2024-09-02:
> > > > tatsu VersionInfo update to 1028.0.0.0.1 fix loop send Yonkers TSS problem
> > > 
> > > 
> > > I tried 1033.0.2 and 1028.0.0.0.1, both didn't work on ios beta6.
> > 
> > 
> > need more code.please see 3utools code
> 
> Do you know what code needs to be modified? I'm not very good at reverse engineering.

Oh, I ha

> > > tatsu VersionInfo update to 1028.0.0.0.1 fix loop send Yonkers TSS problem
> > 
> > 
> > I tried 1033.0.2 and 1028.0.0.0.1, both didn't work on ios beta6.
> 
> need more code.please see 3utools code

Can you provide the 3utools code's link?

nikias, 2024-09-02:
Do you guys have the latest libtatsu (1.0.3) ?

yinyajiang, 2024-09-02:
> Do you guys have the latest libtatsu (1.0.3) ?

Yes, I am using the latest version of libsatsu

yinyajiang, 2024-09-02:
> Do you guys have the latest libtatsu (1.0.3) ?

I tried changing the tatsu client version to 1028.0.0.0.1 and 1033.0.2 (same as in pymobiledevice3) but it didn't work.

xdayong, 2024-09-04:
> Do you guys have the latest libtatsu (1.0.3) ?

Yes, I use the latest, and libimobiledevice ,libplist and other libraries are also the latest.

nikias, 2024-09-05:
Ok, just making sure. Can you run idevicerestore with -d -d so we can see the TSS request (and the reply)?

xdayong, 2024-09-06:
> Ok, just making sure. Can you run idevicerestore with -d -d so we can see the TSS request (and the reply)?

[ios18.txt](https://github.com/user-attachments/files/16902209/ios18.txt)

The file is TSS information. My device is iPhone XS , IPSW is iPhone11,2,iPhone11,4,iPhone11,6_18.0_22A5350a_Restore.ipsw

xdayong, 2024-09-10:
> I can't really see what might be wrong... we'd need a reference TSS request sent out by iTunes or another tool. Here is a patch but I don't believe it has any effect on the current problem:
> 
> ```diff
> diff --git a/src/restore.c b/src/restore.c
> index d3828f9..d836470 100644
> --- a/src/restore.c
> +++ b/src/restore.c
> @@ -698,16 +698,22 @@ static int _restore_service_send(restore_service_client_t service, plist_t plist
>  	return -1;
>  }
>  
> -static int _restore_service_recv(restore_service_client_t service, plist_t *plist)
> +static int _restore_service_recv_timeout(restore_service_client_t service, plist_t *plist, unsigned int timeout)
>  {
> +	struct restored_client_private {
> +		property_list_service_client_t parent;
> +		char *udid;
> +		char *label;
> +		plist_t info;
> +	};
>  	if (!service) {
>  		return -1;
>  	}
>  	switch (service->type) {
>  		case SERVICE_TYPE_RESTORED:
> -			return restored_receive((restored_client_t)service->client, plist);
> +			return property_list_service_receive_plist_with_timeout(((struct restored_client_private*)service->client)->parent, plist, timeout);
>  		case SERVICE_TYPE_PLIST:
> -			return property_list_service_receive_plist((property_list_service_client_t)service->client, plist);
> +			return property_list_service_receive_plist_with_timeout((property_list_service_client_t)service->client, plist, timeout);
>  		default:
>  			break;
>  	}
> @@ -4090,8 +4096,12 @@ static int _restore_send_file_data(struct _restore_send_file_data_ctx* rctx, voi
>  	if (done == 0 && (memcmp(data, "AEA1", 4) == 0)) {
>  		info("Encountered First Chunk in AEA image\n");
>  		plist_t message = NULL;
> -		_restore_service_recv(rctx->service, &message);
> -		restore_send_url_asset(rctx->client, message);
> +		property_list_service_error_t err = _restore_service_recv_timeout(rctx->service, &message, 3000);
> +		if (err == PROPERTY_LIST_SERVICE_E_RECEIVE_TIMEOUT) {
> +			info("NOTE: No URLAsset requested, assuming it is not necessary.");
> +		} else if (err == PROPERTY_LIST_SERVICE_E_SUCCESS) {
> +			restore_send_url_asset(rctx->client, message);
> +		}
>  	}
>  
>  	if (total_size > 0x1000000) {
> ```

Hello nikias,  I just tested this patch and unfortunately,  It's the same issue.

nikias, 2024-09-11:
I can't find what is actually wrong with the TSS... again I don't believe this would fix it, but I have a small patch for libtatsu:
```diff
diff --git a/src/tss.c b/src/tss.c
index fcb683c..3ffba0f 100644
--- a/src/tss.c
+++ b/src/tss.c
@@ -37,7 +37,7 @@

 #define USER_AGENT_STRING "InetURL/1.0"

-#define AUTH_VERSION "973.40.2"
+#define AUTH_VERSION "1033.0.2"

 #ifdef WIN32
 #define TSS_CLIENT_VERSION_STRING "libauthinstall_Win-"AUTH_VERSION""
```

xdayong, 2024-09-13:
> 我找不到 TSS 到底出了什么问题......同样，我不相信这会解决这个问题，但我有一个针对 libtatsu 的小补丁：
> 
> ```diff
> diff --git a/src/tss.c b/src/tss.c
> index fcb683c..3ffba0f 100644
> --- a/src/tss.c
> +++ b/src/tss.c
> @@ -37,7 +37,7 @@
> 
>  #define USER_AGENT_STRING "InetURL/1.0"
> 
> -#define AUTH_VERSION "973.40.2"
> +#define AUTH_VERSION "1033.0.2"
> 
>  #ifdef WIN32
>  #define TSS_CLIENT_VERSION_STRING "libauthinstall_Win-"AUTH_VERSION""
> ```

I have updated this patch, but the problem still persists

skortela, 2024-09-17:
Reproduced with iPhone XS, iOS 18.
[idevicerestore.log](https://github.com/user-attachments/files/17026546/idevicerestore.log)

Or is this same problem? @nikias

nikias, 2024-09-18:
Should be fixed with commit 3faf2926aa03f5d9fd9e394fda9f88e22ae43a5b.

xdayong, 2024-09-19:
> Should be fixed with commit [3faf292](https://github.com/libimobiledevice/idevicerestore/commit/3faf2926aa03f5d9fd9e394fda9f88e22ae43a5b).

Perfect, iOS 18 can now be updated