libimobiledevice/idevicerestore #662: Cannot flash macOS sequoia IPSW

ey3ball, 2024-08-02:
Using the following invocation:
```
/opt/idevice/bin/idevicerestore -P -d -ey /images/UniversalMac_15.0_24A5298h_Restore.ipsw
```

The restore process fails during the filesystem upload with the following sequence:
```
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
ERROR: Unable to restore device
```

This is with :
* a mac mini M1, the same seems to happen with a mac mini M2 pro & the same ipsw
* freshly compiled idevicerestore from master

Full logs (idevicerestore in debug mode & usbmuxd) attached below :
[macos15.beta.log](https://github.com/user-attachments/files/16470438/macos15.beta.log)
[usbmuxd.log](https://github.com/user-attachments/files/16470440/usbmuxd.log)

julienkrijnen8, 2024-08-02:
---------------------------------------------------------------------
Met vriendelijke groet,ok
---------------------------------------------------------------------
Julien Krijnen
---------------------------------------------------------------------
***@***.***
---------------------------------------------------------------------------------

fvallee-scw, 2024-09-10:
FWIW the latest RC images have the same behavior, I plan on digging more this week

U-siro, 2024-09-18:
Same problem on my M1 Macbook Air.

fvallee-scw, 2024-09-20:
According to https://theapplewiki.com/wiki/Apple_Encrypted_Archive :

`Since iOS 18 / macOS 15, the root filesystem in IPSWs is encrypted into .dmg.aea files, using AEA in profile 1 (symmetric encryption but no signing). The auth data in these files contains key-value pairs in the standard AEAAuthData format, which provide the [WKMS](https://theapplewiki.com/index.php?title=WKMS&action=edit&redlink=1) parameters needed to derive the decryption key.`

fvallee-scw, 2024-09-23:
With the latest commits from this weekend, restore now seems to progress past the ASR upload

closed 2024-09-23 by ey3ball