libimobiledevice/idevicerestore #666: iOS 18.1 - Unable to send iBootData component

andrewbaker-uk, 2024-08-20:
Attempting to restore my iPad Pro (4th Generation) back to iOS 17.6.1. 

Everytime I attempt I get the error: Unable to send iBootData to device and then the device gets stuck in recovery mode until I use 3utools to exit recovery mode. 

```
andrewbaker@fedora-andrew-pc:~$ sudo idevicerestore -l -dd
idevicerestore 1.0.0^20230523git609f7f0
irecv_event_cb: device 001a501e0178201e (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
found device with ECID 001a501e0178201e
Setting to configuration 1
Setting to interface 0:0
ECID: 7406439198367774
Identified device as j617ap, iPad14,3
* Host api.ipsw.me:443 was resolved.
* IPv6: 2606:4700:20::681a:89c, 2606:4700:20::ac43:47bb, 2606:4700:20::681a:99c
* IPv4: 104.26.9.156, 104.26.8.156, 172.67.71.187
*   Trying [2606:4700:20::681a:89c]:443...
* Connected to api.ipsw.me (2606:4700:20::681a:89c) port 443
* ALPN: curl offers h2,http/1.1
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384 / x25519 / id-ecPublicKey
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=ipsw.me
*  start date: Jun 30 00:35:28 2024 GMT
*  expire date: Sep 28 00:35:27 2024 GMT
*  subjectAltName: host "api.ipsw.me" matched cert's "*.ipsw.me"
*  issuer: C=US; O=Google Trust Services; CN=WE1
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
*   Certificate level 0: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA256
*   Certificate level 1: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA384
*   Certificate level 2: Public key type EC/secp384r1 (384/192 Bits/secBits), signed using sha256WithRSAEncryption
* using HTTP/2
* [HTTP/2] [1] OPENED stream for https://api.ipsw.me/v4/device/iPad14,3
* [HTTP/2] [1] [:method: GET]
* [HTTP/2] [1] [:scheme: https]
* [HTTP/2] [1] [:authority: api.ipsw.me]
* [HTTP/2] [1] [:path: /v4/device/iPad14,3]
* [HTTP/2] [1] [user-agent: InetURL/1.0 idevicerestore/1.0.0^20230523git609f7f0]
* [HTTP/2] [1] [accept: */*]
> GET /v4/device/iPad14,3 HTTP/2
Host: api.ipsw.me
User-Agent: InetURL/1.0 idevicerestore/1.0.0^20230523git609f7f0
Accept: */*

* old SSL session ID is stale, removing
< HTTP/2 200 
< date: Tue, 20 Aug 2024 10:07:41 GMT
< content-type: application/json
< vary: Accept-Encoding
< vary: Origin
< access-control-allow-methods: GET, OPTIONS
< access-control-allow-origin: *
< cache-control: public, max-age=14400
< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< expires: Tue, 20 Aug 2024 09:26:34 GMT
< last-modified: Mon, 19 Aug 2024 17:59:49 GMT
< x-content-type-options: nosniff
< x-data-location: cache
< x-frame-options: DENY
< x-frame-options: DENY
< x-xss-protection: 1; mode=block
< strict-transport-security: max-age=15552000; preload
< cf-cache-status: HIT
< age: 2511
< report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=tWWl%2Bctnwr4q1tr5GWE8OoPe8U0Xzxw7XWQc0bELUNXtZbKriWR6iz4sS%2FM%2BlL8svNk%2FOjvPf65mduOQUx2h33KNs94IJb4D5NzChXL33iV5l45EHXJqvzyquE%2B70GH3RKagt9UV7Wq6"}],"group":"cf-nel","max_age":604800}
< nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
< server: cloudflare
< cf-ray: 8b6193e96df99545-LHR
[... 1353 lines cut ...]
ERROR: Unable to send component 'RestoreTrustCache' to device.
DEBUG: SIO is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'SIO' in TSS response
NOTE: No path for component SIO in TSS, will fetch from build_identity
Extracting SmartIOFirmware_ASCv5.im4p (Firmware/SmartIOFirmware_ASCv5.im4p)...
Personalizing IMG4 component SIO...
Tag found
Sending SIO (1163354 bytes)...
ERROR: Unable to send SIO component: Invalid input
ERROR: Unable to send SIO to device.
ERROR: Unable to send component 'SIO' to device.
DEBUG: iBootData is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'iBootData' in TSS response
NOTE: No path for component iBootData in TSS, will fetch from build_identity
Extracting iBootData.j617.RELEASE.im4p (Firmware/all_flash/iBootData.j617.RELEASE.im4p)...
Personalizing IMG4 component iBootData...
Tag found
Sending iBootData (10041 bytes)...
ERROR: Unable to send iBootData component: Invalid input
ERROR: Unable to send iBootData to device.
ERROR: Unable to send component 'iBootData' to device.
ERROR: Unable to send components supposed to be loaded by iBoot
ERROR: Unable to place device into restore mode

```

nikias, 2024-08-21:
Your output suggests that the build is more than a year old. Please rebuild from git and try again.

andrewbaker-uk, 2024-08-24:
Will do

closed 2024-08-24 by andrewbaker-uk