libimobiledevice/idevicerestore #668: iOS 18 beta6  restore failed

irainbw, 2024-08-30:
Hello 
I'm trying to update to iOS 18b6 with the latest build of idevicerestore. (iPhone 14)
A loop problem occurred.
last message： 
[iPhone14_iOS18b6.txt](https://github.com/user-attachments/files/16813741/iPhone14_iOS18b6.txt)

irainbw, 2024-09-02:
> Hello I'm trying to update to iOS 18b6 with the latest build of idevicerestore. (iPhone 14) A loop problem occurred. last message： [iPhone14_iOS18b6.txt](https://github.com/user-attachments/files/16813741/iPhone14_iOS18b6.txt)
When I debugged the code I found ：
[image]
DataRequestMsg Continuously circulating

nikias, 2024-09-02:
Are you using latest libtatsu (1.0.3)?

irainbw, 2024-09-02:
> Are you using latest libtatsu (1.0.3)?

Yes, I compiled libtatsu from the master branch （windows）

nikias, 2024-09-04:
Ok, just making sure. Can you run idevicerestore with `-d -d` so we can see the TSS request (and the reply)?

nikias, 2024-09-09:
You can try with this patch, but I think it won't have any effect on the current problem...

```diff
diff --git a/src/restore.c b/src/restore.c
index d3828f9..d836470 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -698,16 +698,22 @@ static int _restore_service_send(restore_service_client_t service, plist_t plist
        return -1;
 }

-static int _restore_service_recv(restore_service_client_t service, plist_t *plist)
+static int _restore_service_recv_timeout(restore_service_client_t service, plist_t *plist, unsigned int timeout)
 {
+       struct restored_client_private {
+               property_list_service_client_t parent;
+               char *udid;
+               char *label;
+               plist_t info;
+       };
        if (!service) {
                return -1;
        }
        switch (service->type) {
                case SERVICE_TYPE_RESTORED:
-                       return restored_receive((restored_client_t)service->client, plist);
+                       return property_list_service_receive_plist_with_timeout(((struct restored_client_private*)service->client)->parent, plist, timeout);
                case SERVICE_TYPE_PLIST:
-                       return property_list_service_receive_plist((property_list_service_client_t)service->client, plist);
+                       return property_list_service_receive_plist_with_timeout((property_list_service_client_t)service->client, plist, timeout);
                default:
                        break;
        }
@@ -4090,8 +4096,12 @@ static int _restore_send_file_data(struct _restore_send_file_data_ctx* rctx, voi
        if (done == 0 && (memcmp(data, "AEA1", 4) == 0)) {
                info("Encountered First Chunk in AEA image\n");
                plist_t message = NULL;
-               _restore_service_recv(rctx->service, &message);
-               restore_send_url_asset(rctx->client, message);
+               property_list_service_error_t err = _restore_service_recv_timeout(rctx->service, &message, 3000);
+               if (err == PROPERTY_LIST_SERVICE_E_RECEIVE_TIMEOUT) {
+                       info("NOTE: No URLAsset requested, assuming it is not necessary.");
+               } else if (err == PROPERTY_LIST_SERVICE_E_SUCCESS) {
+                       restore_send_url_asset(rctx->client, message);
+               }
        }

        if (total_size > 0x1000000) {
```

irainbw, 2024-09-10:
> You can try with this patch, but I think it won't have any effect on the current problem...
>
> ```diff
> diff --git a/src/restore.c b/src/restore.c
> index d3828f9..d836470 100644
> --- a/src/restore.c
> +++ [...]

I tried the patch but it had no effect on the results

irainbw, 2024-09-10:
> Ok, just making sure. Can you run idevicerestore with `-d -d` so we can see the TSS request (and the reply)?

NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: restore_get_yonkers_firmware_data: using Yonkers,SysTopPatch1
Sending Yonkers TSS request...
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>@HostPlatformInfo</key>
        <string>windows</string>
        <key>@VersionInfo</key>
        <string>libauthinstall_Win-973.40.2</string>
        <key>@UUID</key>
        <string>7EA19967-C50F-E902-B189-4C88F5FDF539</string>
        <key>@BBTicket</key>
        <true/>
        <key>@Yonkers,Ticket</key>
        <true/>
        <key>SEP</key>
        <dict>
                <key>Digest</key>
                <data>
                Pl11/+lmV9clOP9sPvEi7nazcTd6El7Eg08iPPbjhZuYgnKKbBTqQrHfaG6a
                UBgt
                </data>
        </dict>
        <key>Yonkers,AllowOfflineBoot</key>
        <true/>
        <key>Yonkers,BoardID</key>
        <integer>0</integer>
        <key>Yonkers,ChipID</key>
        <integer>1300</integer>
        <key>Yonkers,ECID</key>
        <data>
        X9uJxOOLH9K5X9N0RCI+7A==
        </data>
        <key>Yonkers,Nonce</key>
        <data>
        iP/9vXMp9t26OLZWkF9DRQ==
        </data>
        <key>Yonkers,PatchEpoch</key>
        <integer>0</integer>
        <key>Yonkers,ProductionMode</key>
        <true/>
        <key>Yonkers,ReadECKey</key>
        <true/>
        <key>Yonkers,ReadFWKey</key>
        <true/>
        <key>Yonkers,SysTopPatch1</key>
        <dict>
                <key>Digest</key>
                <data>
                Gl4AYM/R9NiXFwxCy+oum8F5gzVkWtPC+ZKmc4Cu6PI=
                </data>
                <key>EPRO</key>
                <true/>
                <key>FabRevision</key>
                <integer>5377</integer>
        </dict>
</dict>
</plist>
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... *   Trying 17.137.162.1:443...
* Connected to gs.apple.com (17.137.162.1) port 443
* schannel: disabled automatic use of client certificate
* ALPN: curl offers http/1.1
* schannel: connection hostname (gs.apple.com) did not match against certificate name (gs-mdn-nx.apple.com)
[... 228 lines cut ...]
                                <key>EPRO</key>
                                <true/>
                                <key>FabRevision</key>
                                <integer>5377</integer>
                        </dict>
                        <key>Yonkers,ReadFWKey</key>
                        <true/>
                        <key>Yonkers,AllowOfflineBoot</key>
                        <true/>
                        <key>Yonkers,ReadECKey</key>
                        <true/>
                        <key>Yonkers,Nonce</key>
                        <data>
                        iP/9vXMp9t26OLZWkF9DRQ==
                        </data>
                        <key>Yonkers,ECID</key>
                        <data>
                        X9uJxOOLH9K5X9N0RCI+7A==
                        </data>
                </dict>
                <key>MessageArgUpdaterName</key>
                <string>Savage</string>
        </dict>
</dict>
</plist>

nikias, 2024-09-11:
I can't find what is actually wrong with the TSS... again I don't believe this would fix it, but I have a small patch for libtatsu:
```diff
diff --git a/src/tss.c b/src/tss.c
index fcb683c..3ffba0f 100644
--- a/src/tss.c
+++ b/src/tss.c
@@ -37,7 +37,7 @@

 #define USER_AGENT_STRING "InetURL/1.0"

-#define AUTH_VERSION "973.40.2"
+#define AUTH_VERSION "1033.0.2"

 #ifdef WIN32
 #define TSS_CLIENT_VERSION_STRING "libauthinstall_Win-"AUTH_VERSION""
```
It seems you are running this on Windows, not sure if that might be causing some issue.

irainbw, 2024-09-13:
> ```diff
> 1033.0.2
> ```

also failed

irainbw, 2024-09-13:
> I can't find what is actually wrong with the TSS... again I don't believe this would fix it, but I have a small patch for libtatsu:
>
> ```diff
> diff --git a/src/tss.c b/src/tss.c
> index fcb683c..3ffba0f [...]

also failed

rekyagis, 2024-09-17:
Hello.

I'm facing the same issue.
I'm trying to restore iPhone 15 with iOS 18 (released version) on Linux by the latest idevicerestore and libs.
The process shown in the attachment log loops.
[idevicerestore_loop.log.txt](https://github.com/user-attachments/files/17031654/idevicerestore_loop.log.txt)

It receives following 4 requests repeatedly.
 1. SourceBootObjectV4 of SEP
 2. SourceBootObjectV4 of Yonkers,SysTopPatch9
 3. SourceBootObjectV4 of Yonkers,SysTopPatch1
 4. FirmwareUpdaterData of Yonkers 

The progress-bar on iPhone shows about 95% or more.
No error nor suspicious message are found on debug log.
The TSS request is almost same as what iTunes does, no meaningful difference can be found.
idevicerestore exits with "DONE" message if I send SYGINT (hit Ctrl-C).

All log is attached.
[idevicerestore.log](https://github.com/user-attachments/files/17031674/idevicerestore.log)

Thanks.

nikias, 2024-09-17:
I see, because the TSS stuff can be quite tricky even with small differences, would you be able to post an iTunes created TSS request vs. the idevicerestore one?

rekyagis, 2024-09-17:
Hi,

This is what iTunes on Windows does:
[tss04.txt](https://github.com/user-attachments/files/17036348/tss04.txt)

This is what idevicerestore/libtatsu (latest version) does:
[idevicerestore_tss_1.txt](https://github.com/user-attachments/files/17036447/idevicerestore_tss_1.txt)

I've modified libtatsu to send exactly same request as iTunes, modifying VersionInfo to 1033.0.6 (but not "win"), adding Locality, ApBoardID, ApChipID, ApProductionMode and ApSecurityMode keys, removing SEP key. But it still failed, looping.
[idevicerestore_tss_2.txt](https://github.com/user-attachments/files/17036468/idevicerestore_tss_2.txt)

Thanks.

nikias, 2024-09-18:
Thanks for your help. Still investigating...

nikias, 2024-09-18:
@rekyagis What method are you using to get the debug log on Windows?

rekyagis, 2024-09-18:
> @rekyagis What method are you using to get the debug log on Windows?

I'm running idevicerestore on Linux and macOS. I get its debug log with -d or -dd option.

I also have a Windows PC.
I can get the log of iTunes on Windows, using mitmproxy for TSS requests, Wireshark/USBPcap for USB packets.

Now, I'm investigating its USB packets, but it's harder to understand than before.
It seems there is some change recently, using asynchronous/parallel communicating?

Please let me know if there's anything I can help.
Thanks.

nikias, 2024-09-18:
Yes, now some stuff is happening in parallel...
Early beta restores have been reported to be working though, it seemed to have started with beta 3.

nikias, 2024-09-18:
@rekyagis if you manage to make a USB dump, we are currently looking into the `FirmwareResponseData` response that is sent to the device after the Savage TSS request. There seems to be a change there.

rekyagis, 2024-09-18:
> @rekyagis if you manage to make a USB dump, we are currently looking into the `FirmwareResponseData` response that is sent to the device after the Savage TSS request. There seems to be a change there.

Thank you. Good advice.

This is what idevicerestore sends for FirmwareResponseData response:
[log_idevicerestore.txt](https://github.com/user-attachments/files/17038393/log_idevicerestore.txt)

This is the USB packet damp by Wireshark of iTunes:
[51396.txt](https://github.com/user-attachments/files/17038400/51396.txt)
And decoded plist:
[itunes_firmwareresponsedata_xml.txt](https://github.com/user-attachments/files/17038430/itunes_firmwareresponsedata_xml.txt)

You can find that there is no "FirmwareData" in response of iTunes.

The restore is success with idevicerestore when I remove FirmwareData from the FirmwareResponseData of Yonkers.
(just comment out "plist_dict_set_item(response, "FirmwareData", firmware_data);" at the bottom of restore_get_yonkers_firmware_data)

I think FirmwareData is not necessary in FirmwareResponseData since it is sent as response of SourceBootObjectV4 of Yonkers,SysTopPatch9.
iTunes seems not to send FirmwareData in any FirmwareResponseData (not only in Yonkers) since I can not find "FirmwareData" word in iTunes's USB packet dump.

Thanks.

rekyagis, 2024-09-18:
> iTunes seems not to send FirmwareData in any FirmwareResponseData (not only in Yonkers) since I can not find "FirmwareData" word in iTunes's USB packet dump.

I removed all "FirmwareData" from FirmwareResponseData (just comment out all plist_dict_set_item for "FirmwareData" in restore.c, I found 7 lines).
And got success for restoring iPhone 15 with iOS 18.0.
The success log is attached:
[idevicerestore.log](https://github.com/user-attachments/files/17038797/idevicerestore.log)

I'm not sure we can remove all "FirmwareData" for another model of iPhone or another version of iOS. But it went well with my case.

Thanks.

tudorcraciun, 2024-09-18:
> I removed all "FirmwareData" from FirmwareResponseData (just comment out all plist_dict_set_item for "FirmwareData" in restore.c, I found 7 lines). And got success for restoring iPhone 15 with iOS [...]

Hello,

I have also found these lines of code in `restore.c`:

```
/* skip FirmwareData for newer versions */
if (client->build_major >= 20) {
	debug("DEBUG: Not adding FirmwareData.\n");
	return response;
}
```
	
Maybe this is a clue as to which versions are compatible with this fix?

nikias, 2024-09-18:
I figured that if FirmwareData is not in ResponseTags, it is not required. I pushed commit 3faf292 that should fix it.

tudorcraciun, 2024-09-19:
The fix seems to work (tested on XS and 15), but something happened between the commit on Jul 9 (the previous build I had tested with) and today, as it seems the restoration process takes 3-4 times longer.

irainbw, 2024-09-20:
> I figured that if FirmwareData is not in ResponseTags, it is not required. I pushed commit [3faf292](https://github.com/libimobiledevice/idevicerestore/commit/3faf2926aa03f5d9fd9e394fda9f88e22ae43a5b) that should fix it.

There seems to be a difference in my performance, and now the error appears at the beginning on iphone14
[iphone14_b6.txt](https://github.com/user-attachments/files/17068561/iphone14_b6.txt)

nikias, 2024-09-20:
I can't see why the recent changes would result in slower performance. Any idea what commit you where using when it was still fine? Remember also that newer iOS versions are larger.

tudorcraciun, 2024-09-20:
Initially, I built `idevicerestore` from [f7e24ce](https://github.com/libimobiledevice/idevicerestore/commit/f7e24ce6e56d67c9889744bb270ad6a98fe653f5). The difference in performance I was referring to was between that build and [ad46e14](https://github.com/libimobiledevice/idevicerestore/commit/ad46e149702ece0af81932cc48f8603362102b96).

That was yesterday. It is weird that today the slowness disappeared, even though the hardware is the same (same number of devices in parallel, same workstation). Maybe I am missing something.

The slowness was observed when sending the filesystem (after the log 'Sending filesystem now').

irainbw, 2024-09-26:
> I figured that if FirmwareData is not in ResponseTags, it is not required. I pushed commit [3faf292](https://github.com/libimobiledevice/idevicerestore/commit/3faf2926aa03f5d9fd9e394fda9f88e22ae43a5b) that should fix it.

Now works in recovery mode, but fails in normal state(iphone14,ios18b1)

Failure returned 168 error

[iphone14_18b1.txt](https://github.com/user-attachments/files/17140371/iphone14_18b1.txt)

nikias, 2024-09-28:
@irainbw does it work with 18.0 ?

irainbw, 2024-09-29:
> @irainbw does it work with 18.0 ?

It works in recovery mode, but not in normal mode.

nikias, 2024-09-29:
Can you capture a tss request from Windows/macOS with the device in normal mode? Since we are facing a TSS issue it must be with the request. In recovery mode it does not send the eUICC stuff since it's not available in that mode...