libimobiledevice/idevicerestore #670: iPad Pro 11-inch (M4, Cellular)  failed to enter restore mode

Neil-0609, 2024-09-11:
I compiled the latest version（windows） and tested other devices, such as iPhone12, iPhone13, iPad Air 11-inch (M2), etc. Currently found iPad Pro 11-inch (M4, Cellular)  failed to enter restore mode
log：
ipsw:{C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw}
irecv_event_cb: device 001264421e99001c (udid: N/A) connected in Recovery mode
Found device in Recovery mode
ECID: 5176784724754460
Found device in iPad Pro 11-inch (M4, Cellular)
Identified device as j718ap, iPad16,4
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 17.6.1
IPSW Product Build: 21G101 Major: 21
INFO: device serial number is JT979WTN7T
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
ipsw path:{C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw}
Getting ApNonce in recovery mode... 4a 90 31 64 f8 92 14 55 91 cc 31 51 6e 79 3c 57 1d af 82 df ec ec ca fd e9 ab 0c 83 38 3e e9 70
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... b4 87 35 d1 53 33 c5 0f a5 3c 73 ef 5b 6d 8c 81 c0 10 5a fb
Received SHSH blobs
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.j718.RELEASE.im4p (Firmware/dfu/iBEC.j718.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (1820450 bytes)...
Waiting for device to disconnect...
irecv_event_cb: device 001264421e99001c (udid: N/A) disconnected
Waiting for device to reconnect in recovery mode...
irecv_event_cb: device 001264421e99001c (udid: N/A) connected in Recovery mode
Recovery Mode Environment:
iBoot build-version=iBoot-10151.140.19
iBoot build-style=RELEASE
Sending RestoreLogo...
NOTE: No path for component RestoreLogo in TSS, will fetch from build_identity
Extracting applelogo@2420~ipad-USBc-woven.im4p (Firmware/all_flash/applelogo@2420~ipad-USBc-woven.im4p)...
Personalizing IMG4 component RestoreLogo...
Tag found
Sending RestoreLogo (38835 bytes)...
DEBUG: ANE is loaded by iBoot.
NOTE: No path for component ANE in TSS, will fetch from build_identity
Extracting h16_ane_fw_leto_j7x.im4p (Firmware/ane/h16_ane_fw_leto_j7x.im4p)...
Personalizing IMG4 component ANE...
Tag found
Sending ANE (4992753 bytes)...
DEBUG: AOP is loaded by iBoot.
NOTE: No path for component AOP in TSS, will fetch from build_identity
Extracting aopfw-ipad16aop.RELEASE.im4p (Firmware/AOP/aopfw-ipad16aop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Tag found
Sending AOP (3362545 bytes)...
DEBUG: AVE is loaded by iBoot.
NOTE: No path for component AVE in TSS, will fetch from build_identity
Extracting AppleAVE2FW_H16G.im4p (Firmware/ave/AppleAVE2FW_H16G.im4p)...
Personalizing IMG4 component AVE...
Tag found
Sending AVE (1600809 bytes)...
DEBUG: Ap,RestoreCIO is loaded by iBoot.
NOTE: No path for component Ap,RestoreCIO in TSS, will fetch from build_identity
Extracting t8132ciofw.im4p (Firmware/t8132ciofw.im4p)...
Personalizing IMG4 component Ap,RestoreCIO...
Tag found
Sending Ap,RestoreCIO (190161 bytes)...
DEBUG: Ap,RestoreSecureM3Firmware is loaded by iBoot.
NOTE: No path for component Ap,RestoreSecureM3Firmware in TSS, will fetch from build_identity
Extracting securem3fw-j7xx.im4p (Firmware/M3/bin/securem3fw-j7xx.im4p)...
Personalizing IMG4 component Ap,RestoreSecureM3Firmware...
Tag found
Sending Ap,RestoreSecureM3Firmware (85305 bytes)...
DEBUG: Ap,RestoreSecurePageTableMonitor is loaded by iBoot.
NOTE: No path for component Ap,RestoreSecurePageTableMonitor in TSS, will fetch from build_identity
Extracting sptm.t8132.release.im4p (Firmware/sptm.t8132.release.im4p)...
Personalizing IMG4 component Ap,RestoreSecurePageTableMonitor...
Tag found
Sending Ap,RestoreSecurePageTableMonitor (124065 bytes)...
DEBUG: Ap,RestoreTMU is loaded by iBoot.
NOTE: No path for component Ap,RestoreTMU in TSS, will fetch from build_identity
Extracting t8132tmufw.im4p (Firmware/t8132tmufw.im4p)...
Personalizing IMG4 component Ap,RestoreTMU...
Tag found
Sending Ap,RestoreTMU (26622 bytes)...
DEBUG: Ap,RestoreTrustedExecutionMonitor is loaded by iBoot.
NOTE: No path for component Ap,RestoreTrustedExecutionMonitor in TSS, will fetch from build_identity
Extracting txm.iphoneos.release.im4p (Firmware/txm.iphoneos.release.im4p)...
Personalizing IMG4 component Ap,RestoreTrustedExecutionMonitor...
Tag found
Sending Ap,RestoreTrustedExecutionMonitor (146219 bytes)...
DEBUG: Ap,RestorecL4 is loaded by iBoot.
NOTE: No path for component Ap,RestorecL4 in TSS, will fetch from build_identity
Extracting exclavecore_bundle.t8132.RELEASE.restore.im4p (Firmware/image4/exclavecore_bundle.t8132.RELEASE.restore.im4p)...
Personalizing IMG4 component Ap,RestorecL4...
Tag found
Sending Ap,RestorecL4 (31862723 bytes)...
DEBUG: GFX is loaded by iBoot.
NOTE: No path for component GFX in TSS, will fetch from build_identity
Extracting armfw_g16g.im4p (Firmware/agx/armfw_g16g.im4p)...
Personalizing IMG4 component GFX...
Tag found
Sending GFX (2420273 bytes)...
DEBUG: ISP is loaded by iBoot.
NOTE: No path for component ISP in TSS, will fetch from build_identity
Extracting adc-eris-j7xx.im4p (Firmware/isp_bni/adc-eris-j7xx.im4p)...
Personalizing IMG4 component ISP...
Tag found
Sending ISP (22245108 bytes)...
DEBUG: MtpFirmware is loaded by iBoot.
NOTE: No path for component MtpFirmware in TSS, will fetch from build_identity
Extracting J718_MtpFirmware.im4p (Firmware/J718_MtpFirmware.im4p)...
Personalizing IMG4 component MtpFirmware...
Tag found
Sending MtpFirmware (1119291 bytes)...
DEBUG: PMP is loaded by iBoot.
NOTE: No path for component PMP in TSS, will fetch from build_identity
Extracting t8132pmp.im4p (Firmware/pmp/t8132pmp.im4p)...
Personalizing IMG4 component PMP...
Tag found
Sending PMP (630513 bytes)...
DEBUG: RestoreANS is loaded by iBoot.
NOTE: No path for component RestoreANS in TSS, will fetch from build_identity
Extracting rans.t8132_ASP3.release.im4p (Firmware/rans.t8132_ASP3.release.im4p)...
Personalizing IMG4 component RestoreANS...
Tag found
Sending RestoreANS (1202801 bytes)...
DEBUG: RestoreDCP is loaded by iBoot.
NOTE: No path for component RestoreDCP in TSS, will fetch from build_identity
Extracting t8132dcp_restore.im4p (Firmware/dcp/t8132dcp_restore.im4p)...
Personalizing IMG4 component RestoreDCP...
Tag found
Sending RestoreDCP (16211697 bytes)...
DEBUG: RestoreTrustCache is loaded by iBoot.
NOTE: No path for component RestoreTrustCache in TSS, will fetch from build_identity
Extracting 090-91937-104.dmg.trustcache (Firmware/090-91937-104.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Tag found
Sending RestoreTrustCache (18154 bytes)...
DEBUG: SIO is loaded by iBoot.
NOTE: No path for component SIO in TSS, will fetch from build_identity
Extracting SmartIOFirmware_ASCv7.im4p (Firmware/SmartIOFirmware_ASCv7.im4p)...
Personalizing IMG4 component SIO...
Tag found
Sending SIO (1165381 bytes)...
DEBUG: iBootData is loaded by iBoot.
NOTE: No path for component iBootData in TSS, will fetch from build_identity
Extracting iBootData.j718.RELEASE.im4p (Firmware/all_flash/iBootData.j718.RELEASE.im4p)...
Personalizing IMG4 component iBootData...
Tag found
Sending iBootData (12252 bytes)...
ramdisk-size=0x20000000
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 090-91937-104.dmg (090-91937-104.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (201338612 bytes)...
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.j718ap.im4p (Firmware/all_flash/DeviceTree.j718ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (69468 bytes)...
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j718.RELEASE.im4p (Firmware/all_flash/sep-firmware.j718.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (5258123 bytes)...
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad16 (kernelcache.release.ipad16)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (19207478 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 001264421e99001c (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 001264421e99001c (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.

nikias, 2024-09-11:
The device reconnects in recovery mode, so something is wrong with the image personalization. However it also seems you are not running a fresh build of idevicerestore (it would print the version as the first line of output).

Neil-0609, 2024-09-12:
> personalization

I commented out the line of code that prints the version（info("%s %s\n", PACKAGE_NAME, PACKAGE_VERSION);）。

How should I determine where the problem lies now?

nikias, 2024-09-12:
Why would you do that, if it's because of a compilation issue then chances are that other things might be broken as well.
In any case, I assume that either the TSS request is missing something (did you update to use latest libtatsu?) or maybe something else is broken in the build. You can run idevicerestore with `-d -d` to get a verbose debug output, including the TSS request.

Neil-0609, 2024-09-13:
PS C:\Users\MDW_TKC_03\Desktop\Release> .\idevicerestore.exe -i 0x000239DA3AB9001C -e -d -d  C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw
ipsw:{C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw}
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&29af2330&0&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:000239da3ab9001c_ibfl:3d_sika:00_srnm:[l3412n76qj]#mux1, location: 892483632
irecv_event_cb: device 000239da3ab9001c (udid: N/A) connected in Recovery mode
Found device in Recovery mode
found device with ECID 000239da3ab9001c
Setting to interface 0:0
ECID: 626559404277788
Identified device as j718ap, iPad16,4
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 17.6.1
IPSW Product Build: 21G101 Major: 21
found device with ECID 000239da3ab9001c
Setting to interface 0:0
INFO: device serial number is L3412N76QJ
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... 62 ea dc 95 7b 0a c8 5e c8 0f 0b 5c ce f1 6f 52 cc fc be 8c c6 44 98 98 38 69 30 5f e4 e2 0a 5d
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 8b 57 b4 39 7c 51 02 ff f4 25 63 99 8c 21 1f 1f 5d c9 96 ba
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry ANS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,CIO
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSIntegrityCatalog
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSTrustCache
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSVolume
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,LowPowerMode
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreCIO
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreSecureM3Firmware
DEBUG: Adding EPRO=true to TSS entry
[... 1913 lines cut ...]
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.j718ap.im4p (Firmware/all_flash/DeviceTree.j718ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (69468 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j718.RELEASE.im4p (Firmware/all_flash/sep-firmware.j718.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (5258123 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad16 (kernelcache.release.ipad16)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (19207478 bytes)...
DEBUG: Waiting for device to disconnect...
_irecv_event_handler: removed ecid: 000239da3ab9001c, location: 926038064
irecv_event_cb: device 000239da3ab9001c (udid: N/A) disconnected
Waiting for device to enter restore mode...
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&29af2330&0&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:000239da3ab9001c_ibfl:3d_sika:00_srnm:[l3412n76qj]#mux1, location: 892483632
irecv_event_cb: device 000239da3ab9001c (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.

Neil-0609, 2024-09-13:
`PS C:\Users\MDW_TKC_03\Desktop\Release> .\idevicerestore.exe -i 0x000239DA3AB9001C -e -d -d  C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw
ipsw:{C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw}
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&29af2330&0&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:000239da3ab9001c_ibfl:3d_sika:00_srnm:[l3412n76qj]#mux1, location: 892483632
irecv_event_cb: device 000239da3ab9001c (udid: N/A) connected in Recovery mode
Found device in Recovery mode
found device with ECID 000239da3ab9001c
Setting to interface 0:0
ECID: 626559404277788
Identified device as j718ap, iPad16,4
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 17.6.1
IPSW Product Build: 21G101 Major: 21
found device with ECID 000239da3ab9001c
Setting to interface 0:0
INFO: device serial number is L3412N76QJ
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... 62 ea dc 95 7b 0a c8 5e c8 0f 0b 5c ce f1 6f 52 cc fc be 8c c6 44 98 98 38 69 30 5f e4 e2 0a 5d
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 8b 57 b4 39 7c 51 02 ff f4 25 63 99 8c 21 1f 1f 5d c9 96 ba
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry ANS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,CIO
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSIntegrityCatalog
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSTrustCache
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSVolume
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,LowPowerMode
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreCIO
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreSecureM3Firmware
DEBUG: Adding EPRO=true to TSS entry
[... 1914 lines cut ...]
Extracting DeviceTree.j718ap.im4p (Firmware/all_flash/DeviceTree.j718ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (69468 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j718.RELEASE.im4p (Firmware/all_flash/sep-firmware.j718.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (5258123 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad16 (kernelcache.release.ipad16)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (19207478 bytes)...
DEBUG: Waiting for device to disconnect...
_irecv_event_handler: removed ecid: 000239da3ab9001c, location: 926038064
irecv_event_cb: device 000239da3ab9001c (udid: N/A) disconnected
Waiting for device to enter restore mode...
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&29af2330&0&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:000239da3ab9001c_ibfl:3d_sika:00_srnm:[l3412n76qj]#mux1, location: 892483632
irecv_event_cb: device 000239da3ab9001c (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.
`

Neil-0609, 2024-09-13:
[log.txt](https://github.com/user-attachments/files/16987555/log.txt)

Neil-0609, 2024-09-19:
> Why would you do that, if it's because of a compilation issue then chances are that other things might be broken as well. In any case, I assume that either the TSS request is missing something (did [...]

```c
ipsw:{C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw}
libimobiledevice-win32 1.0.0 (libtatsu (null))
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&29af2330&0&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:000239da3ab9001c_ibfl:3d_sika:00_srnm:[l3412n76qj]#mux1, location: 892483632
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&9c5d003&2&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:001264421e99001c_ibfl:3d_sika:00_srnm:[jt979wtn7t]#mux1, location: 842149936
irecv_event_cb: device 001264421e99001c (udid: N/A) connected in Recovery mode
Found device in Recovery mode
found device with ECID 001264421e99001c
Setting to interface 0:0
ECID: 5176784724754460
Found device in iPad Pro 11-inch (M4, Cellular)
Identified device as j718ap, iPad16,4
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 17.6.1
IPSW Product Build: 21G101 Major: 21
found device with ECID 001264421e99001c
Setting to interface 0:0
INFO: device serial number is JT979WTN7T
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
ipsw path:{C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw}
Getting ApNonce in recovery mode... b1 55 ca c6 ce 22 41 ee f8 dd 97 7e c8 a4 c2 4d 86 64 8e f2 d4 50 6f 3f e2 cf 82 3b a6 ec cc fd
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 30 4c 6a 15 43 fa f7 b2 bc 4f 38 57 25 17 ce b8 52 17 5a da
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry ANS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,CIO
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSIntegrityCatalog
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSTrustCache
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSVolume
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,LowPowerMode
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreCIO
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreSecureM3Firmware
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
[... 4022 lines cut ...]
Extracting DeviceTree.j718ap.im4p (Firmware/all_flash/DeviceTree.j718ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (69468 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j718.RELEASE.im4p (Firmware/all_flash/sep-firmware.j718.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (5258123 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad16 (kernelcache.release.ipad16)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (19207478 bytes)...
DEBUG: Waiting for device to disconnect...
_irecv_event_handler: removed ecid: 001264421e99001c, location: 825374768
irecv_event_cb: device 001264421e99001c (udid: N/A) disconnected
Waiting for device to enter restore mode...
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&9c5d003&2&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:001264421e99001c_ibfl:3d_sika:00_srnm:[jt979wtn7t]#mux1, location: 842149936
irecv_event_cb: device 001264421e99001c (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.
```

Neil-0609, 2024-09-19:
> Why would you do that, if it's because of a compilation issue then chances are that other things might be broken as well. In any case, I assume that either the TSS request is missing something (did [...]

@nikias  
```c
ipsw:{C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw}
libimobiledevice-win32 1.0.0 (libtatsu (null))
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&29af2330&0&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:000239da3ab9001c_ibfl:3d_sika:00_srnm:[l3412n76qj]#mux1, location: 892483632
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&9c5d003&2&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:001264421e99001c_ibfl:3d_sika:00_srnm:[jt979wtn7t]#mux1, location: 842149936
irecv_event_cb: device 001264421e99001c (udid: N/A) connected in Recovery mode
Found device in Recovery mode
found device with ECID 001264421e99001c
Setting to interface 0:0
ECID: 5176784724754460
Found device in iPad Pro 11-inch (M4, Cellular)
Identified device as j718ap, iPad16,4
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 17.6.1
IPSW Product Build: 21G101 Major: 21
found device with ECID 001264421e99001c
Setting to interface 0:0
INFO: device serial number is JT979WTN7T
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
ipsw path:{C:\Users\MDW_TKC_03\Downloads\iPad_Pro_M4_17.6.1_21G101_Restore.ipsw}
Getting ApNonce in recovery mode... b1 55 ca c6 ce 22 41 ee f8 dd 97 7e c8 a4 c2 4d 86 64 8e f2 d4 50 6f 3f e2 cf 82 3b a6 ec cc fd
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 30 4c 6a 15 43 fa f7 b2 bc 4f 38 57 25 17 ce b8 52 17 5a da
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry ANS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,CIO
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSIntegrityCatalog
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSTrustCache
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,ExclaveOSVolume
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,LowPowerMode
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreCIO
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreSecureM3Firmware
DEBUG: Adding EPRO=true to TSS entry
[... 1914 lines cut ...]
Extracting DeviceTree.j718ap.im4p (Firmware/all_flash/DeviceTree.j718ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (69468 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j718.RELEASE.im4p (Firmware/all_flash/sep-firmware.j718.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (5258123 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad16 (kernelcache.release.ipad16)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (19207478 bytes)...
DEBUG: Waiting for device to disconnect...
_irecv_event_handler: removed ecid: 001264421e99001c, location: 825374768
irecv_event_cb: device 001264421e99001c (udid: N/A) disconnected
Waiting for device to enter restore mode...
_irecv_event_handler: found new: \\?\usb#vid_05ac&pid_1281&mi_00#8&9c5d003&2&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#sdom:01_cpid:8132_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:001264421e99001c_ibfl:3d_sika:00_srnm:[jt979wtn7t]#mux1, location: 842149936
irecv_event_cb: device 001264421e99001c (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.
```