libimobiledevice/idevicerestore #671: hang on restore - FDR 0x559117bfef10 waiting for message... No data to read (timeout)

ds2k5, 2024-09-11:
Have this issue that restore did not work

iPhone 7 + iPhone_4.7_P3_15.8.3_19H386_Restore.ipsw

**$ idevicerestore --version
idevicerestore 1.0.0**

**$ idevicerestore -e -R -l -y -d 2>&1 | tee iphone7.log**

[iphone7.log](https://github.com/user-attachments/files/16969751/iphone7.log)

nikias, 2024-09-12:
Please build idevicerestore from source and try again.

ds2k5, 2024-09-12:
Thanks...

have build.... but got this error when run:  /usr/local/bin/idevicerestore -e -R -l -y

ProductionMode: true
Starting Reverse Proxy
/usr/local/bin/idevicerestore: symbol lookup error: /usr/local/bin/idevicerestore: undefined symbol: reverse_proxy_client_create_with_port

nikias, 2024-09-12:
Just rebuild everything from latest source code.

ds2k5, 2024-09-12:
>Just rebuild everything from latest source code.

ok I will try - thanks