libimobiledevice/idevicerestore #673: iPhone14,6 (iPhone SE 3rd Gen) 17.6.1 (build 21G101) restore failed

T0biii, 2024-09-15:
getting the following error when i try to recovery the iPhone SE thats are stuck in the Recovery Screen after an IOS Update

```bash
Sending SystemImageCanonicalMetadata now...
Done sending SystemImageCanonicalMetadata
ERROR: Could not read data (-256). Aborting.
ReverseProxy[Ctrl]: Connection closed
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device
idevice_event_cb: device 000158d23a0b801e (udid: 00008110-000158D23A0B801E) disconnected
```

<details>
<summary>idevicerestore --version</summary>

```bash
idevicerestore 1.0.0-203-gf7e24ce
```
</details>

<details>
<summary>ldd /usr/local/bin/idevicerestore</summary>

```bash
linux-vdso.so.1 (0x00007fffbb14f000)
libirecovery-1.0.so.5 => /usr/local/lib/libirecovery-1.0.so.5 (0x000076dc57496000)
libimobiledevice-1.0.so.6 => /usr/local/lib/libimobiledevice-1.0.so.6 (0x000076dc5744f000)
libusbmuxd-2.0.so.7 => /usr/local/lib/libusbmuxd-2.0.so.7 (0x000076dc57444000)
libimobiledevice-glue-1.0.so.0 => /usr/local/lib/libimobiledevice-glue-1.0.so.0 (0x000076dc5742e000)
libtatsu.so.0 => /usr/local/lib/libtatsu.so.0 (0x000076dc57421000)
libplist-2.0.so.4 => /usr/local/lib/libplist-2.0.so.4 (0x000076dc57402000)
libzip.so.4 => /lib/x86_64-linux-gnu/libzip.so.4 (0x000076dc573e5000)
libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x000076dc573c9000)
libcurl.so.4 => /lib/x86_64-linux-gnu/libcurl.so.4 (0x000076dc57308000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x000076dc57000000)
libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x000076dc572ea000)
libssl.so.3 => /lib/x86_64-linux-gnu/libssl.so.3 (0x000076dc5723e000)
libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x000076dc56a00000)
libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x000076dc56f17000)
libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x000076dc5722a000)
libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x000076dc569d5000)
libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x000076dc569b3000)
librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x000076dc56995000)
libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x000076dc56924000)
libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x000076dc57214000)
libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x000076dc568d0000)
libldap.so.2 => /lib/x86_64-linux-gnu/libldap.so.2 (0x000076dc56873000)
liblber.so.2 => /lib/x86_64-linux-gnu/liblber.so.2 (0x000076dc56863000)
libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x000076dc567a9000)
libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x000076dc5679b000)
/lib64/ld-linux-x86-64.so.2 (0x000076dc574fe000)
libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x000076dc56768000)
libunistring.so.5 => /lib/x86_64-linux-gnu/libunistring.so.5 (0x000076dc565bb000)
libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x000076dc563c1000)
libhogweed.so.6 => /lib/x86_64-linux-gnu/libhogweed.so.6 (0x000076dc56379000)
libnettle.so.8 => /lib/x86_64-linux-gnu/libnettle.so.8 (0x000076dc56324000)
libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x000076dc562a0000)
libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x000076dc561d7000)
libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x000076dc561ab000)
libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x000076dc561a5000)
libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x000076dc56198000)
libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x000076dc5617e000)
libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x000076dc5615b000)
libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x000076dc5614c000)
libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x000076dc55fa8000)
libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x000076dc55f92000)
libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x000076dc55f8b000)
libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x000076dc55f78000)
libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x000076dc55f6a000)
```
</details>

idevicerestore -l -R --debug
https://gist.github.com/T0biii/8f9912a813b0aa238ddf2ba42685e783

nikias, 2024-09-15:
I have not seen this error before. It doesn't say broken though, but it says "busy":
```
dev_init:1132: open /dev/disk2 hit EBUSY, attempts remaining: 4 dev_name = disk2
dev_init:1132: open /dev/disk2 hit EBUSY, attempts remaining: 3 dev_name = disk2
dev_init:1132: open /dev/disk2 hit EBUSY, attempts remaining: 2 dev_name = disk2
dev_init:1132: open /dev/disk2 hit EBUSY, attempts remaining: 1 dev_name = disk2
nx_dev_init:717: device initialization failed: 16
nx_mount:1094: device initialization failed: 16
```
Previous access to the device works fine, according to the log output...

T0biii, 2024-09-16:
its very strange, tried it via iTunes but it Always stuck at the end of the Progress bar.
I can try another cable but i dont think this ist going to help 

i am running a Ubuntu 24.04 over VMware Workstation in Win11 and connect the iPhone to the vm. 

got no more ideas its stuck on the Recovery Screen

nikias, 2024-09-16:
If you don't care about the data on the device, you can try an erase restore with `-e`. But if iTunes cannot continue either then it's definitely not a bug in idevicerestore.

T0biii, 2024-09-16:
i tried `-e` already.
so maybe i need to give it to Apple :/

closed 2024-09-16 by T0biii