libimobiledevice/idevicerestore #675: iOS 18 restore failure

TPJoseph, 2024-09-17:
idevicerestore /Users/ctdi/Thunder/Firmware/iPhone15\,2_18.0_22A3354_Restore
idevicerestore 1.0.0-git-4442c3a
Found device in Recovery mode
ECID: 4895945474654238
Identified device as d73ap, iPhone15,2
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 18.0
IPSW Product Build: 22A3354 Major: 22
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... 2c d1 55 ee 9a 76 d0 2f 68 07 b2 33 34 33 b1 c0 8a e6 84 21 21 c0 a0 1f 80 74 d6 00 79 64 d0 34 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... f1 3d 68 cc 1b 46 aa 65 17 0b bf 17 a7 f8 e8 7b b7 ad 80 0d 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d73.RELEASE.im4p (Firmware/dfu/iBEC.d73.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (1410316 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-11881.2.10
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2556~iphone.im4p (Firmware/all_flash/applelogo@2556~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (29999 bytes)...
Extracting h15_ane_fw_themis_d7x.im4p (Firmware/ane/h15_ane_fw_themis_d7x.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (1550928 bytes)...
Extracting aopfw-iphone15aop.RELEASE.im4p (Firmware/AOP/aopfw-iphone15aop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (3062352 bytes)...
Extracting AppleAVE2FW_H15.im4p (Firmware/ave/AppleAVE2FW_H15.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1547376 bytes)...
Extracting sptm.t8120.release.im4p (Firmware/sptm.t8120.release.im4p)...
Personalizing IMG4 component Ap,RestoreSecurePageTableMonitor...
Sending Ap,RestoreSecurePageTableMonitor (160691 bytes)...
Extracting txm.iphoneos.release.im4p (Firmware/txm.iphoneos.release.im4p)...
Personalizing IMG4 component Ap,RestoreTrustedExecutionMonitor...
Sending Ap,RestoreTrustedExecutionMonitor (156738 bytes)...
Extracting armfw_g15p.im4p (Firmware/agx/armfw_g15p.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (1296768 bytes)...
Extracting adc-kronos-d7x.im4p (Firmware/isp_bni/adc-kronos-d7x.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (21932627 bytes)...
Extracting D73_MtpFirmware.im4p (Firmware/D73_MtpFirmware.im4p)...
Personalizing IMG4 component MtpFirmware...
Sending MtpFirmware (4568216 bytes)...
Extracting t8120pmp.im4p (Firmware/pmp/t8120pmp.im4p)...
Personalizing IMG4 component PMP...
Sending PMP (535120 bytes)...
Extracting rans.t8120.release.im4p (Firmware/rans.t8120.release.im4p)...
Personalizing IMG4 component RestoreANS...
Sending RestoreANS (1035574 bytes)...
Extracting iphone15dcp_restore.im4p (Firmware/dcp/iphone15dcp_restore.im4p)...
Personalizing IMG4 component RestoreDCP...
Sending RestoreDCP (1510393 bytes)...
Extracting 044-36986-002.dmg.trustcache (Firmware/044-36986-002.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (18333 bytes)...
Extracting SmartIOFirmware_ASCv6.im4p (Firmware/SmartIOFirmware_ASCv6.im4p)...
Personalizing IMG4 component SIO...
Sending SIO (1180492 bytes)...
[... 718 lines cut ...]
restore-step-user-progress = 0
Considering saving tolerated failures: false (null) false
[17:35:08.0766-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x11030655:16;0x110306C2:20}
restore-step-names = {0x11030655:is_host_compatible;0x110306C2:cleanup_send_crash_logs}
restore-step-uptime = 9
restore-step-user-progress = 0
send_crash_logs is not supported by the host.
[17:35:08.0767-GMT]{3>6} CHECKPOINT END: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x11030655:16}
restore-step-names = {0x11030655:is_host_compatible}
restore-step-uptime = 9
restore-step-user-progress = 0
Considering saving tolerated failures: false (null) false
[17:35:08.0767-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030655:16;0x11030648:21}
restore-step-names = {0x11030655:is_host_compatible;0x11030648:cleanup_send_final_status}
restore-step-uptime = 9
restore-step-user-progress = 0

ERROR: Unable to successfully restore device
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(This host version is unsupported. You may need to update your host tools.)
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device

nikias, 2024-09-18:
You need to build idevicerestore from latest source, the above commit mentioned in the first line doesn't even exist in the official repo. But the code is too old (the device even says it: `This host version is unsupported. You may need to update your host tools.`)
However note that there is currently an issue with some devices failing restore even with the latest code.

AKT04, 2024-09-18:
I am using the latest version of idevicerestore, and the restoration process has been successful for iPhone 13. However, I am encountering issues with restoring iPhone 14 and iPhone 15; the process fails for these models.

AKT04, 2024-09-18:
fails sometime idevicerestore 

idevicerestore 1.0.0-203-gf7e24ce
progress: 0 0.000000
progress: 0 0.100000
Found device in Recovery mode
ECID: 7155901184311342
progress: 0 0.200000
Identified device as d79ap, iPhone12,8
Device Product Version: N/A
Device Product Build: N/A
progress: 0 0.600000
Extracting BuildManifest from IPSW
BitRaserProg:1
progress: 0 0.800000
BitRaserProgk:1
BitRaserProgk:2
IPSW Product Version: 18.0
IPSW Product Build: 22A3354 Major: 22
BitRaserProgk:3
INFO: device serial number is FFXCL4XDPLJQ
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
BitRaserProg:4
progress: 1 0.000000
Checking IPSW for required components...
All required components found in IPSW
progress: 1 0.200000
Getting ApNonce in recovery mode... 2a 4f 69 fe 82 5f 8b 52 91 40 41 fa 65 2d a0 5e 97 83 6a b0 f3 b3 71 fa 0b d9 77 cb 5c 6a cc 06 
BitRaserProg:6
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 9e c4 3d 93 1d bf cd c2 30 bd 8d 1c 65 d9 b6 f5 02 61 27 18 
Received SHSH blobs
BitRaserProg:7
progress: 1 0.250000
BitRaserProg:9
progress: 1 0.300000
Extracting iBEC.d79.RELEASE.im4p (Firmware/dfu/iBEC.d79.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (1542667 bytes)...
BitRaserProg:11
BitRaserProg:12
BitRaserProg:13
progress: 1 0.500000
BitRaserProg:14
progress: 1 0.700000
Recovery Mode Environment:
iBoot build-version=iBoot-11881.2.10
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p (Firmware/all_flash/applelogo@2x~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (14840 bytes)...
Extracting h12_ane_fw_metis.im4p (Firmware/ane/h12_ane_fw_metis.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (4759699 bytes)...
Extracting aopfw-iphone12caop.RELEASE.im4p (Firmware/AOP/aopfw-iphone12caop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (1736851 bytes)...
Extracting AppleAVE2FW_H12.im4p (Firmware/ave/AppleAVE2FW_H12.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1421955 bytes)...
Extracting armfw_g12p.im4p (Firmware/agx/armfw_g12p.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (1228643 bytes)...
Extracting adc-zelus-d79.im4p (Firmware/isp_bni/adc-zelus-d79.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (14983315 bytes)...
Extracting t8030pmp.im4p (Firmware/pmp/t8030pmp.im4p)...
[... 1990 lines cut ...]
progress: 2 0.790017
progress: 2 0.800004
progress: 2 0.810014
progress: 2 0.820002
progress: 2 0.830012
progress: 2 0.840021
progress: 2 0.850009
progress: 2 0.860019
progress: 2 0.870006
progress: 2 0.880016
progress: 2 0.890004
progress: 2 0.900013
progress: 2 0.910001
progress: 2 0.920011
progress: 2 0.930020
progress: 2 0.940008
progress: 2 0.950018
progress: 2 0.960005
progress: 2 0.970015
progress: 2 0.980003
progress: 2 0.990012
progress: 2 1.000000
Done sending filesystem
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated

khang30, 2024-09-18:
Unable to restore my device; it is now stuck on the Apple logo screen.
How to put the device into normal mode? Thanks

nikias, 2024-09-18:
@AKT04 I pushed a commit that should fix it.

AKT04, 2024-09-19:
> @AKT04 I pushed a commit that should fix it.

Thanks it's working.

khang30, 2024-09-19:
@nikias 
Unable to restore device - iOS 17.6.1

nikias, 2024-09-19:
@khang30 can you provide a debug log?

TPJoseph, 2024-09-19:
@nikias i was trying to restore iPad16,6 to iOS 18

but failed for restore

idevicerestore -e -y iPad_Pro_M4_18.0_22A3354_Restore.ipsw -d
idevicerestore 1.0.0-git-3faf292
idevice_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) connected in normal mode
Found device in Normal mode
idevice_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) connected in normal mode
ECID: 1770016667664412
Identified device as j721ap, iPad16,6
Device Product Version: 17.6
Device Product Build: 21G80
Extracting BuildManifest from IPSW
IPSW Product Version: 18.0
IPSW Product Build: 22A3354 Major: 22
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in normal mode... 84 c0 8e a2 80 23 d4 25 47 9b 7e 22 0a 8d 8e ed 3d ad 8c cf cb fd 3a c3 cf ba 4a 11 3e d4 e2 2f 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... ab 20 dd fb 32 ed cb 12 e4 5c 46 f3 71 78 0d 91 6d bf ee 94 
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

TPJoseph, 2024-09-19:
I also tried it with 17.6.1 and is still a failure

idevicerestore -e -y iPad_Pro_M4_17.6.1_21G101_Restore.ipsw -d
idevicerestore 1.0.0-git-3faf292
idevice_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) connected in normal mode
Found device in Normal mode
idevice_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) connected in normal mode
ECID: 1770016667664412
Identified device as j721ap, iPad16,6
Device Product Version: 17.6
Device Product Build: 21G80
Extracting BuildManifest from IPSW
IPSW Product Version: 17.6.1
IPSW Product Build: 21G101 Major: 21
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in normal mode... 84 c0 8e a2 80 23 d4 25 47 9b 7e 22 0a 8d 8e ed 3d ad 8c cf cb fd 3a c3 cf ba 4a 11 3e d4 e2 2f 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... ba db 6e 3d db 84 db 7b ab 98 d7 86 22 01 7e e2 88 0e 3b fb 
Received SHSH blobs
Entering recovery mode...
DEBUG: Waiting for device to disconnect...
idevice_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) disconnected
DEBUG: Waiting for device to connect in recovery mode...
irecv_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) connected in Recovery mode
INFO: device serial number is CC9XXKXJRW
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.j721.RELEASE.im4p (Firmware/dfu/iBEC.j721.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (1820786 bytes)...
Waiting for device to disconnect...
irecv_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) disconnected
Waiting for device to reconnect in recovery mode...
irecv_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) connected in Recovery mode
Recovery Mode Environment:
iBoot build-version=iBoot-10151.140.19
iBoot build-style=RELEASE
Sending RestoreLogo...
NOTE: No path for component RestoreLogo in TSS, will fetch from build_identity
Extracting applelogo@2752~ipad-USBc-woven.im4p (Firmware/all_flash/applelogo@2752~ipad-USBc-woven.im4p)...
Personalizing IMG4 component RestoreLogo...
Tag found
Sending RestoreLogo (45136 bytes)...
DEBUG: ANE is loaded by iBoot.
NOTE: No path for component ANE in TSS, will fetch from build_identity
Extracting h16_ane_fw_leto_j7x.im4p (Firmware/ane/h16_ane_fw_leto_j7x.im4p)...
Personalizing IMG4 component ANE...
Tag found
Sending ANE (4992753 bytes)...
DEBUG: AOP is loaded by iBoot.
NOTE: No path for component AOP in TSS, will fetch from build_identity
Extracting aopfw-ipad16aop.RELEASE.im4p (Firmware/AOP/aopfw-ipad16aop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Tag found
Sending AOP (3362545 bytes)...
DEBUG: AVE is loaded by iBoot.
NOTE: No path for component AVE in TSS, will fetch from build_identity
Extracting AppleAVE2FW_H16G.im4p (Firmware/ave/AppleAVE2FW_H16G.im4p)...
Personalizing IMG4 component AVE...
Tag found
Sending AVE (1600809 bytes)...
DEBUG: Ap,RestoreCIO is loaded by iBoot.
NOTE: No path for component Ap,RestoreCIO in TSS, will fetch from build_identity
Extracting t8132ciofw.im4p (Firmware/t8132ciofw.im4p)...
Personalizing IMG4 component Ap,RestoreCIO...
Tag found
[... 87 lines cut ...]
Extracting 090-91937-104.dmg (090-91937-104.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (201338612 bytes)...
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.j721ap.im4p (Firmware/all_flash/DeviceTree.j721ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (68251 bytes)...
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j721.RELEASE.im4p (Firmware/all_flash/sep-firmware.j721.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (5258123 bytes)...
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad16 (kernelcache.release.ipad16)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (19207478 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 000649d21eb9001c (udid: 00008132-000649D21EB9001C) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.

nikias, 2024-09-19:
There must be something wrong with some dependency. Did you compile libplist yourself? If so, have you tried to run `make check` to see if it passes all tests correctly? The 18.0 restore fails because of a TSS request error. It could also be something with libtatsu but I would check libplist first.

TPJoseph, 2024-09-19:
make check
Making check in libcnary
make[1]: Nothing to be done for `check'.
Making check in src
make[1]: Nothing to be done for `check'.
Making check in include
make[1]: Nothing to be done for `check'.
Making check in tools
make[1]: Nothing to be done for `check'.
Making check in docs
make[1]: Nothing to be done for `check'.
Making check in test
/Library/Developer/CommandLineTools/usr/bin/make  check-TESTS
PASS: empty.test
PASS: small.test
PASS: medium.test
PASS: large.test
PASS: huge.test
PASS: bigarray.test
PASS: empty++.test
PASS: small++.test
PASS: medium++.test
PASS: large++.test
PASS: huge++.test
PASS: bigarray++.test
PASS: dates.test
PASS: timezone1.test
PASS: timezone2.test
PASS: signedunsigned1.test
PASS: signedunsigned2.test
PASS: signedunsigned3.test
PASS: hex.test
PASS: order.test
PASS: recursion.test
PASS: entities.test
PASS: empty_keys.test
PASS: amp.test
PASS: invalid_tag.test
PASS: cdata.test
PASS: offsetsize.test
PASS: refsize.test
PASS: malformed_dict.test
PASS: uid.test
PASS: integer_set.test
PASS: json1.test
PASS: json2.test
PASS: json3.test
PASS: json-invalid-types.test
PASS: json-int64-min-max.test
PASS: ostep1.test
PASS: ostep2.test
PASS: ostep-strings.test
PASS: ostep-comments.test
PASS: ostep-invalid-types.test
============================================================================
Testsuite summary for libplist 2.6.0
============================================================================
# TOTAL: 41
# PASS:  41
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================

TPJoseph, 2024-09-19:
Is this what you were looking for

nikias, 2024-09-19:
Ok great so libplist is fine. Could you attempt 18.0 again but add `-d -d` to the command line?

Thomas639, 2024-09-19:
idevicerestore -e -y iPad_Pro_M4_18.0_22A3354_Restore.ipsw -d

I had given the output for the above command 

Do you want me to add -d twice

nikias, 2024-09-19:
Yes, twice please

TPJoseph, 2024-09-20:
[iPad_16,6_17.6.1.txt](https://github.com/user-attachments/files/17075453/iPad_16.6_17.6.1.txt)
[iPad_16,6_18.0.txt](https://github.com/user-attachments/files/17075454/iPad_16.6_18.0.txt)

TPJoseph, 2024-09-23:
@nikias Did you get a chance of looking into this please?

Neil-0609, 2024-09-24:
#670 
encountered the same problem

khang30, 2024-09-25:
`idevicerestore.exe -e -y download/ios_firmwares/iPhone14,3_18.0_22A3354_Restore.ipsw -i 1777412116463646 -d
idevicerestore 1.0.0-216-g93f5e31 (libirecovery 1.2.0, libtatsu 1.0.3-3-g1e161fc)
irecv_event_cb: device 0006508c01d2401e (udid: N/A) connected in Recovery mode
Found device in Recovery mode
ECID: 1777412116463646
Identified device as d64ap, iPhone14,3
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 18.0
IPSW Product Build: 22A3354 Major: 22
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... bd 1d cf b1 10 2b 87 89 59 ca b1 9b 0b 11 83 68 03 1f 3a 43 6c 6e d1 00 a1 e0 39 c3 7c 5c cc 27
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 7b e0 d6 d1 52 d9 41 56 b3 c8 f0 91 21 3c c5 3a 74 0f 11 8e
Received SHSH blobs
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.d64.RELEASE.im4p (Firmware/dfu/iBEC.d64.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (1054413 bytes)...
Waiting for device to disconnect...
irecv_event_cb: device 0006508c01d2401e (udid: N/A) disconnected
Waiting for device to reconnect in recovery mode...
irecv_event_cb: device 0006508c01d2401e (udid: N/A) connected in Recovery mode
Recovery Mode Environment:
iBoot build-version=iBoot-11881.2.10
iBoot build-style=RELEASE
Sending RestoreLogo...
NOTE: No path for component RestoreLogo in TSS, will fetch from build_identity
Extracting applelogo@2778~iphone.im4p (Firmware/all_flash/applelogo@2778~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Tag found
Sending RestoreLogo (29759 bytes)...
DEBUG: ANE is loaded by iBoot.
NOTE: No path for component ANE in TSS, will fetch from build_identity
Extracting h14_ane_fw_bia_d6x.im4p (Firmware/ane/h14_ane_fw_bia_d6x.im4p)...
Personalizing IMG4 component ANE...
Tag found
Sending ANE (4925792 bytes)...
DEBUG: AOP is loaded by iBoot.
NOTE: No path for component AOP in TSS, will fetch from build_identity
Extracting aopfw-iphone14aop.RELEASE.im4p (Firmware/AOP/aopfw-iphone14aop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Tag found
Sending AOP (2705760 bytes)...
DEBUG: AVE is loaded by iBoot.
NOTE: No path for component AVE in TSS, will fetch from build_identity
Extracting AppleAVE2FW_H14.im4p (Firmware/ave/AppleAVE2FW_H14.im4p)...
Personalizing IMG4 component AVE...
Tag found
Sending AVE (1476616 bytes)...
DEBUG: Ap,RestoreSecurePageTableMonitor is loaded by iBoot.
NOTE: No path for component Ap,RestoreSecurePageTableMonitor in TSS, will fetch from build_identity
Extracting sptm.t8110.release.im4p (Firmware/sptm.t8110.release.im4p)...
Personalizing IMG4 component Ap,RestoreSecurePageTableMonitor...
Tag found
Sending Ap,RestoreSecurePageTableMonitor (160218 bytes)...
DEBUG: Ap,RestoreTrustedExecutionMonitor is loaded by iBoot.
NOTE: No path for component Ap,RestoreTrustedExecutionMonitor in TSS, will fetch from build_identity
Extracting txm.iphoneos.release.im4p (Firmware/txm.iphoneos.release.im4p)...
Personalizing IMG4 component Ap,RestoreTrustedExecutionMonitor...
Tag found
Sending Ap,RestoreTrustedExecutionMonitor (156498 bytes)...
DEBUG: GFX is loaded by iBoot.
NOTE: No path for component GFX in TSS, will fetch from build_identity
[... 332 lines cut ...]
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
Mounting filesystems (16)
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes. Error code: -2
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
Checkpoint completed id: 0x626 (mount_filesystem) result=0
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device`

nikias, 2024-09-25:
This is the error:
```* Could not resolve host: wkms.sd.apple.com```
You need to make sure you can reach this host

AKT04, 2024-09-27:
Restoration fails when I connect multiple devices. I have connected 32 devices (18 iPhone 13 and multiple others, including iPhone 7, 8, 14, and 15). Some iPhone 13 devices completed the restoration phase, but others are stuck. The same issue occurs with the other devices. i am using latest library.

nikias, 2024-09-27:
> Restoration fails when I connect multiple devices. I have connected 32 devices (18 iPhone 13 and multiple others, including iPhone 7, 8, 14, and 15). Some iPhone 13 devices completed the restoration [...]

Please open a separate issue, and provide more helpful information, like what platform and hardware you are using, add some log output of the failures etc.

nikias, 2024-09-27:
> @nikias Did you get a chance of looking into this please?

I still didn't figure out what is wrong with the requests...

nikias, 2024-09-27:
Actually, regarding 18.0 on that particular device, iOS 18.0 is currently not being signed by Apple:
[image]

nikias, 2024-10-09:
FYI @TPJoseph 18.0.1 is out and that is also signed for the iPad in question. So maybe that does work now?

yourchanges, 2024-11-08:
ipad 13,1 to 18.1 same problem

``` 
build\windows\idevicerestore.exe -e -y -R D:\\JiuXunCleanV202409\\iPad_Fall_2020_18.1_22B83_Restore.ipsw
idevicerestore 1.0.0-171-g8a5abb9
Found device in Recovery mode
ECID: 7091576166023198
Identified device as j307ap, iPad13,1
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 18.1
IPSW Product Build: 22B83 Major: 22
INFO: device serial number is GG7GK30VQ16T
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... ee 10 49 b2 f5 6d 6c 70 e6 47 c6 2c ff ba 2e 87 23 70 98 78 52 49 cf 36 17 2f 94 57 8a 63 51 b5
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... ea 55 55 20 11 74 3e 50 93 09 c1 eb ed 3a ab 3d 7f 25 5f 24
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.j307.RELEASE.im4p (Firmware/dfu/iBEC.j307.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (868772 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-11881.40.163
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2360~ipad-USBc.im4p (Firmware/all_flash/applelogo@2360~ipad-USBc.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (30540 bytes)...
Extracting h13_ane_fw_styx_j30x.im4p (Firmware/ane/h13_ane_fw_styx_j30x.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (4989717 bytes)...
Extracting aopfw-ipad13paop.RELEASE.im4p (Firmware/AOP/aopfw-ipad13paop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (1934101 bytes)...
Extracting AppleAVE2FW_H13.im4p (Firmware/ave/AppleAVE2FW_H13.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1455517 bytes)...
Extracting armfw_g13p.im4p (Firmware/agx/armfw_g13p.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (3572213 bytes)...
Extracting adc-pallas-j30x.im4p (Firmware/isp_bni/adc-pallas-j30x.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (20751128 bytes)...
Extracting t8101pmp.im4p (Firmware/pmp/t8101pmp.im4p)...
Personalizing IMG4 component PMP...
Sending PMP (434965 bytes)...
Extracting rans.t8101.release.im4p (Firmware/rans.t8101.release.im4p)...
Personalizing IMG4 component RestoreANS...
Sending RestoreANS (964941 bytes)...
Extracting ipad13pdcp_restore.im4p (Firmware/dcp/ipad13pdcp_restore.im4p)...
Personalizing IMG4 component RestoreDCP...
Sending RestoreDCP (1972305 bytes)...
Extracting 044-05174-134.dmg.trustcache (Firmware/044-05174-134.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (16328 bytes)...
Extracting SmartIOFirmware_ASCv4.im4p (Firmware/SmartIOFirmware_ASCv4.im4p)...
Personalizing IMG4 component SIO...
Sending SIO (1162161 bytes)...
Extracting iBootData.j307.RELEASE.im4p (Firmware/all_flash/iBootData.j307.RELEASE.im4p)...
Personalizing IMG4 component iBootData...
Sending iBootData (15908 bytes)...
ramdisk-size=0x20000000
Extracting 044-05174-134.dmg (044-05174-134.dmg)...
[... 696 lines cut ...]
[09:38:57.0982-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x11030655:16;0x110306C2:20}
restore-step-names = {0x11030655:is_host_compatible;0x110306C2:cleanup_send_crash_logs}
restore-step-uptime = 5
restore-step-user-progress = 0
send_crash_logs is not supported by the host.
[09:38:57.0982-GMT]{3>6} CHECKPOINT END: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x11030655:16}
restore-step-names = {0x11030655:is_host_compatible}
restore-step-uptime = 5
restore-step-user-progress = 0
Considering saving tolerated failures: false (null) false
[09:38:57.0983-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030655:16;0x11030648:21}
restore-step-names = {0x11030655:is_host_compatible;0x11030648:cleanup_send_final_status}
restore-step-uptime = 5
restore-step-user-progress = 0

ERROR: Unable to successfully restore device
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(This host version is unsupported. You may need to update your host tools.)
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device

```

nikias, 2024-11-08:
You are not using the latest code. Recompile and try again....