libimobiledevice/idevicerestore #679: iBridge 9.0 Restore - Unknown Command ASR

P5-2005, 2024-09-20:
Hello,
using last commit,
trying to restore ibridge 9.0,
first try get :
```
Received 320 bytes:
Checkpoint completed id: 0x675 (init_gigalocker) result=0
Checkpoint started   id: 0x68C (init_permanent_update_volume)
printing 320 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Checksum Chunks</key>
        <true/>
        <key>Checksum Type</key>
        <integer>1</integer>
        <key>Command</key>
        <string>Initiate</string>
</dict>
</plist>
ERROR: Unknown command received from ASR: 'Initiate'
ERROR: ASR was unable to validate the filesystem
ERROR: Unable to send filesystem
Checkpoint completed id: 0x68C (init_permanent_update_volume) result=0
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device
```
after edit on asr.c :
```
else if (!strcmp(command, "Initiate")) {
    
		printf("Handling 'Initiate' command.\n");
		plist_free(packet);
		break;
		
		}
```
restore passed this step,
and get weird issue on this state after :

```
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[20:17:19.0486-GMT]{3>5} CHECKPOINT BEGIN: (null):[0x06B4] await_system_image_invert_retry
restore-step-ids = {0x1103067B:62;0x110306B4:101}
restore-step-names = {0x1103067B:perform_restore_installing;0x110306B4:await_system_image_invert_retry}
restore-step-uptime = 79
restore-step-user-progress = 31
waiting for await_system_image_invert_retry.
[AsyncCP][WaitStart] Start to Wait checkpoint restore_system_image.
[AsyncCP][WaitEnd] Wait checkpoint restore_system_image success.
entering mount_filesystems
Not counting data volume as required.
LPMedia *copy_media_for_partition_id(partition_id_t): partition id (3) device_node is set to NULL. Returning nil
Skipping mount of update partition
restore_apfs_image_async : started restoring an APFS image
entering mount_partition
System already mounted read-write (mount ignored)
OS digest = b805dd20bdf9f4853154a883bff41fd5df2f10f1224afc9e4cbde3d9583a53635599dd0d0538ddc03408dac9eb5a0d51
sock  14: set SO_NOSIGPIPE=1
sock  14: listening on [::]:12345
requesting sync restore (SystemImageData) on port 12345: {
    MessageUseStreamedImageFile = 0;
}
sock  15: accepted [::ffff:127.0.0.1]:12345 <- [::ffff:127.0.0.1]:49160 (from 14)
receiving sync restore (SystemImageData) on socket 15
restore_image : Restoring via ASR
Restoring to APFS Inverter Image = /mnt1/apfs_invert_asr_img
entering ramrod_execute_command_with_config: /usr/sbin/asr
[... 103 lines cut ...]
restore-step-user-progress = 31
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[20:17:41.0908-GMT]{3>5} CHECKPOINT BEGIN: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x1103067B:62;0x110306B4:101;0x110306C2:105}
restore-step-names = {0x1103067B:perform_restore_installing;0x110306B4:await_system_image_invert_retry;0x110306C2:cleanup_send_crash_logs}
restore-step-uptime = 101
restore-step-user-progress = 31
send_crash_logs is not supported by the host.
[20:17:41.0908-GMT]{3>5} CHECKPOINT END: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x1103067B:62;0x110306B4:101}
restore-step-names = {0x1103067B:perform_restore_installing;0x110306B4:await_system_image_invert_retry}
restore-step-uptime = 101
restore-step-user-progress = 31
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[20:17:41.0908-GMT]{3>5} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103067B:62;0x110306B4:101;0x11030648:106}
restore-step-names = {0x1103067B:perform_restore_installing;0x110306B4:await_system_image_invert_retry;0x11030648:cleanup_send_final_status}
restore-step-uptime = 101
restore-step-user-progress = 31

WARNING: Attempting to continue after critical error, restore might fail...
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(failed to restore APFS image)
Checkpoint started   id: 0x649 (cleanup_wait_status_received)
```

nikias, 2024-09-21:
Thanks to @fbrandstetter we have a fix for this and I already pushed it to master. Let me know if it works for you.

P5-2005, 2024-09-22:
yes, last commit fix the issue with last ibridge restore, thank you

closed 2024-09-22 by P5-2005