libimobiledevice/idevicerestore #681: Thank you for developing this!

arsaccol, 2024-09-23:
My sister was trying to fall back to her older iPhone XR after her newer phone got stolen, and in trying to resolve some issues with Apple support involving the XR's broken FaceID sensor, almost bricked it trying to mess around with iOS recovery. We weren't able to fix her iPhone with either the "Apple Devices" application on Windows, and with an old Mac she had lying around, as both were giving different errors over multiple tries. She was going to be without a phone, as this was her backup.

But I was able to finally flash iOS 18 with idevicerestore! The version on Ubuntu's apt gave me some errors, but after building the dependencies and finally the tool itself in its latest version, I was able to get the phone to work! 

Perhaps this isn't what a real GitHub should read like, but I understand how thankless open source development can be, despite all the hard work developers put in it, so I really wanted to take the time to express my thanks. So, bless you all!

nikias, 2024-09-24:
Thank you for the feedback!

baahver, 2024-10-14:
I use this post to compliment you too! 
I had an iPhone 12 that gave me several errors. I could not open it again, nor use it. 
Apple Devices app on windows was not able to restore this phone. I tried several times, with different cables, pc's, etc. But none of them had success. Apple chat helped me a lot, but no success.

Then I installed idevicerestore from git on Manjaro Linux. I just had to find some packages and install. 
```
git clone https://github.com/libimobiledevice/idevicerestore.git
cd idevicerestore
./autogen.sh
sudo make
sudo make install
idevicerestore --latest ## gave me an error that I had to erase too.
sudo idevicerestore --erase --latest
fixed it and I could restore to 17.7, but 18x was possible too. That one I updated on the working phone again!! 

Thank you very much too!!