libimobiledevice/idevicerestore #682: iPhone14,7 restore 18.1, ERROR: tss_send_request: Unhandled status code 168

yinyajiang, 2024-09-24:
`idevicerestore.exe -d -d -y -e F:\Firmware\iPhone14,7_18.1_22B5045g_Restore.ipsw

idevicerestore 1.0.0-git-48350d6-dirty (libirecovery 1.2.0, libtatsu 1.0.3)
idevice_event_cb: device 000931a822a1401e (udid: 00008110-000931A822A1401E) connected in normal mode
Found device in Normal mode
ECID: 2587872995655710
Identified device as d27ap, iPhone14,7
Device Product Version: 18.1
Device Product Build: 22B5045g
Extracting BuildManifest from IPSW
IPSW Product Version: 18.1
IPSW Product Build: 22B5045g Major: 22
Device supports Image4: true
Variant: Developer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in normal mode... 65 20 11 ae 46 70 10 2b 69 ee c0 93 09 66 bf 67 94 2c 5b b8 e4 52 35 e7 b6 dd cd e5 d2 5c 05 3e
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 79 ce 61 1a 6d a2 68 6e c5 20 c3 8f 56 53 65 e9 2f 97 31 82
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry ANS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,AudioBootChime
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,HapticAssets
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreSecurePageTableMonitor
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,RestoreTrustedExecutionMonitor
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,SecurePageTableMonitor
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,SystemVolumeCanonicalMetadata
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,TrustedExecutionMonitor
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'BMU,DigestMap' as it doesn't have RestoreRequestRules and is not Trusted
DEBUG: tss_request_add_ap_tags: Skipping 'BMU,FirmwareMap' as it doesn't have RestoreRequestRules and is not Trusted
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
[... 1225 lines cut ...]
* using HTTP/1.1
> POST /TSS/controller?action=2 HTTP/1.1
Host: gs.apple.com
User-Agent: InetURL/1.0
Accept: */*
Cache-Control: no-cache
Content-type: text/xml; charset="utf-8"
Content-Length: 20804

< HTTP/1.1 200
< Server: Apple
< Date: Tue, 24 Sep 2024 07:26:52 GMT
< Content-Type: text/html
< Content-Length: 46
< Connection: keep-alive
< Strict-Transport-Security: max-age=31536000; includeSubdomains
< X-Frame-Options: SAMEORIGIN
< Host: gs.apple.com
<
* Connection #0 to host gs.apple.com left intact
TSS server returned: STATUS=168&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 168
ERROR: TSS request failed (status=168, message=An internal error occurred.)
ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device`

nikias, 2024-09-28:
it seems like `eUICC,EID` and `eUICC,RootKeyIdentifier` are missing and the `Nonce`values for `eUICC,Gold` and `eUICC,Main`.
Can you post an output of `ideviceinfo -k FirmwarePreflightInfo` ?

irainbw, 2024-09-29:
> FirmwarePreflightInfo

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>CertID</key>
        <integer>3559316616</integer>
        <key>ChipID</key>
        <integer>1515745</integer>
        <key>ChipSerialNo</key>
        <data>
        RybIyg==
        </data>
        <key>FusingStatus</key>
        <integer>3</integer>
        <key>Nonce</key>
        <data>
        pTLahj3e6UrDLw/XCIVh4ul8xX0=
        </data>
        <key>PkHash</key>
        <data>
        1CbciL9lkyTMwvJb9x6qQeI7I6o67JQ7Hrld7ZfedPJEIrHjwZx7sn8Up4AomCeM
        </data>
        <key>SKeyHash</key>
        <data>
        AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
        </data>
        <key>SKeyStatus</key>
        <integer>0</integer>
        <key>VendorID</key>
        <integer>3</integer>
</dict>
</plist>
```

irainbw, 2024-09-29:
> it seems like `eUICC,EID` and `eUICC,RootKeyIdentifier` are missing and the `Nonce`values for `eUICC,Gold` and `eUICC,Main`. Can you post an output of `ideviceinfo -k FirmwarePreflightInfo` ?

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>CertID</key>
        <integer>3559316616</integer>
        <key>ChipID</key>
        <integer>1515745</integer>
        <key>ChipSerialNo</key>
        <data>
        RybIyg==
        </data>
        <key>FusingStatus</key>
        <integer>3</integer>
        <key>Nonce</key>
        <data>
        pTLahj3e6UrDLw/XCIVh4ul8xX0=
        </data>
        <key>PkHash</key>
        <data>
        1CbciL9lkyTMwvJb9x6qQeI7I6o67JQ7Hrld7ZfedPJEIrHjwZx7sn8Up4AomCeM
        </data>
        <key>SKeyHash</key>
        <data>
        AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
        </data>
        <key>SKeyStatus</key>
        <integer>0</integer>
        <key>VendorID</key>
        <integer>3</integer>
</dict>
</plist>
```

Trenki, 2025-01-20:
Hello, have you solved this problem? It seems I'm encountering a similar issue.