libimobiledevice/idevicerestore #684: 2020 M1 Macbook Air fails to enter restore mode

bms-1984, 2024-09-30:
I noticed in the latest commit you started calling this an image personalization error, which is eerily reminding me of the original reason I'm trying to do a restore (image personalization error in software update) Anyway, does anyone actually know what this means/what I might be able to do about it?

```
/usr/local/bin/idevicerestore -e -d UniversalMac_15.0_24A335_Restore.ipsw 
idevicerestore 1.0.0-219-g511261e (libirecovery 1.2.0-6-gd55c5f8, libtatsu 1.0.3-6-g263f3b3)
irecv_event_cb: device 001c541236d2001e (udid: N/A) connected in DFU mode
Found device in DFU mode
ECID: 7973736553775134
Identified device as j313ap, MacBookAir10,1
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 15.0
IPSW Product Build: 24A335 Major: 24
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Performing macOS restore
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in dfu mode... 79 56 83 f8 3c 4e 80 75 cc 65 9a f3 b5 63 cb b6 1e 4c 54 19 f9 10 54 db a4 6d 6c 2c bd 25 ef 1c 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 23 75 ba 86 05 18 c3 14 df 13 8c 0f 63 de 1b 96 be f2 6f 05 
Received SHSH blobs
Getting SepNonce in dfu mode... 23 75 ba 86 05 18 c3 14 df 13 8c 0f 63 de 1b 96 be f2 6f 05 
Received SHSH blobs
Getting SepNonce in dfu mode... 23 75 ba 86 05 18 c3 14 df 13 8c 0f 63 de 1b 96 be f2 6f 05 
Received SHSH blobs
NOTE: No path for component iBSS in TSS, will fetch from build_identity
Extracting iBSS.j313.RELEASE.im4p (Firmware/dfu/iBSS.j313.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Tag found
Sending iBSS (736479 bytes)...
[==================================================] 100.0%
Waiting for device to disconnect...
irecv_event_cb: device 001c541236d2001e (udid: N/A) disconnected
Waiting for device to reconnect...
irecv_event_cb: device 001c541236d2001e (udid: N/A) connected in Recovery mode
Nonce: 79 56 83 f8 3c 4e 80 75 cc 65 9a f3 b5 63 cb b6 1e 4c 54 19 f9 10 54 db a4 6d 6c 2c bd 25 ef 1c 
Personalizing IMG4 component Ap,LocalPolicy...
Tag found
Sending Ap,LocalPolicy (3041 bytes)...
[==================================================] 100.0%
INFO: executing command: lpolrestore
iBoot boot-stage=1
DEBUG: Ap,RestoreCIO is loaded by iBoot Stage 1 and iBoot.
NOTE: No path for component Ap,RestoreCIO in TSS, will fetch from build_identity
Extracting t8103ciofw.im4p (Firmware/t8103ciofw.im4p)...
Personalizing IMG4 component Ap,RestoreCIO...
Tag found
Sending Ap,RestoreCIO (173306 bytes)...
[==================================================] 100.0%
INFO: executing command: firmware
DEBUG: Ap,RestoreTMU is loaded by iBoot Stage 1 and iBoot.
NOTE: No path for component Ap,RestoreTMU in TSS, will fetch from build_identity
Extracting t8103tmufw.im4p (Firmware/t8103tmufw.im4p)...
Personalizing IMG4 component Ap,RestoreTMU...
Tag found
Sending Ap,RestoreTMU (21487 bytes)...
[==================================================] 100.0%
INFO: executing command: firmware
DEBUG: RestoreANS is loaded by iBoot Stage 1 and iBoot.
[... 112 lines cut ...]
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (180365557 bytes)...
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.j313ap.im4p (Firmware/all_flash/DeviceTree.j313ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (53069 bytes)...
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j313.RELEASE.im4p (Firmware/all_flash/sep-firmware.j313.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (5484999 bytes)...
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.mac13g (kernelcache.release.mac13g)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (28589691 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 001c541236d2001e (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 001c541236d2001e (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Device reconnected in recovery mode, most likely image personalization failed.
```

bms-1984, 2024-10-04:
I must have accidentally had library leakage between Debian packages and source-built ones because I rebuilt every package from Git on a fresh live USB and this runs perfectly. Closing issue now!

closed 2024-10-04 by bms-1984

L4Juan, 2024-11-21:
> I must have accidentally had library leakage between Debian packages and source-built ones because I rebuilt every pay leakage between Debian packages and source-built ones because I rebuckage from Git on a fresh live USB and this runs perfectly. Closing issue now!

Hi! Could you briefly explain how you did this? Did you have to install all the dependencies in the clean usb?

fmatrixch, 2024-11-24:
I have the same problem too