libimobiledevice/idevicerestore #687: Feature Request: Add "Fix Battery" Option to Force Battery Health Data Resynchronization After Firmware Restore

D33F4ULT, 2024-10-13:
I’d like to request a new feature to be added to the idevicerestore tool. Specifically, I'm looking for a "Fix Battery" option that forces iOS to resynchronize and retrieve fresh battery health data from the Battery Management System (BMS) during or after a firmware restore.

Background:
In tools like 3uTools, there is a useful feature called "Fix Battery" that is commonly used after replacing an iPhone battery. This option forces iOS to fetch accurate battery health information (capacity, cycle count, etc.) from the BMS chip, which can fix incorrect or missing battery health statistics in iOS.

Currently, when replacing an iPhone battery or after a firmware restore, iOS may not automatically refresh the battery health status, leading to inaccurate data being displayed in Settings. This feature is particularly useful in cases where:
 - The battery was recently replaced.
 - Battery health data was corrupted or not displayed properly.
 
Proposal:
I propose adding a flag (e.g., --fix-battery) in idevicerestore that will:

Force the iPhone to reset and re-read battery health data from the battery’s BMS after a firmware restore.
Potentially clear any cached or outdated battery health data, ensuring iOS retrieves the correct information from the battery.
This feature would be a valuable addition for users dealing with battery replacement and battery health display issues on iPhones. Any insights on the feasibility of this request or recommendations for how it might be implemented in libimobiledevice would be appreciated.

Thank you for your consideration!

nikias, 2024-10-14:
I would add it if knew how it works...

TheRealBoss, 2024-10-31:
Found this on twitter:

Edit % battery health:
- Edit: /var/MobileSoftwareUpdate/Hardware/Battery/Library/Preferences/com.apple.batteryhealthdata.plist (% health)
- Delete: /var/MobileSoftwareUpdate/Hardware/Battery/Library/Preferences/com.apple.batterydata.cyclecount.plist
Might be what you are looking for? I need it aswell.

Johnny314, 2024-11-11:
did you find anything on that @nikias  @TheRealBoss  @D33F4ULT

TheRealBoss, 2024-11-15:
JC uses a tool separate from their main program called jcupdate.exe (in tools/jcupdateb folder). It uses libimobiledevice libraries like libplist, libirecovery etc and the .exe itself seems to be modified and made executable version of idevicerestore itself.
It has a few extra arguments, like -b for batteryfix and -i (for password to start the .exe ?!). So you can't really run it without knowing the arguments needed. But since I'm still a novice in disassembling, I still don't know much.

nikias, 2024-11-15:
@TheRealBoss where can I find the tool? Thanks

TheRealBoss, 2024-11-15:
By downloading JC repair tool and using the battery fix tool once. It will then download the tool.

AiXanadu, 2024-12-14:
@nikias  @TheRealBoss 
The principle is that when it is marked to repair the battery health, if the current operation is CREATE_FILESYSTEM, then exit and re-flash the device. Because the file system is destroyed, the battery data will be rebuilt.
But I think you will have a better way.