libimobiledevice/idevicerestore #690: Iphone 6s 15.8.3 problem

Fskadi, 2024-10-23:
Hello, all works well till half of loading bar 

Mounting filesystems (16)
ERROR: Unable to receive message from FDR 0x79eaac001390 (-4). 0/2 bytes
ERROR: Unable to receive message from FDR 0x79eaac001c70 (-4). 0/2 bytes
ERROR: Unable to receive message from FDR 0x79eaac0012b0 (-4). 0/2 bytes
ERROR: Unable to receive message from FDR 0x79eaac000e30 (-4). 0/2 bytes
ERROR: Unable to receive message from FDR 0x79eaac0011d0 (-4). 0/2 bytes
Unknown operation (80)
Unknown data request 'PersonalizedData' received

this is where it gets bad, what can i do, idevicerestore is downloaded via yay on archlinux, the image is signed

Fskadi, 2024-10-24:
i recompiled from github, now on version 1.0.0, same thing

Fskadi, 2024-10-24:
i thought it might need a computer reboot, i did, tried again, same thing happens so i don't think i'm doing something wrong

nikias, 2024-10-24:
What does `idevicerestore --version` show?

Fskadi, 2024-10-24:
idevicerestore 1.0.0

nikias, 2024-10-24:
Yes, then you didn't compile the latest code, or it's using an older build.

Fskadi, 2024-10-24:
my bad, i forgot to uninstall the yay version, uninstalled everything, git cloned and worked all the way to finish, i have installed ios on my iphone, thank you!!

nikias, 2024-10-24:
Glad to hear!

closed 2024-10-24 by nikias