libimobiledevice/idevicerestore #691: Attempting data recovery

E3lleyyaa, 2024-10-30:
```
Got status message
Status: Restore Finished
```
After the firmware update, instead of switching on, the device rebooted and entered a new mode where it shows a different boot screen with the message ‘Attempting data recovery.’ As far as I understand, this process cannot be skipped and installs the firmware a second time, correct?Sorry for image quality(webcam)
[image]

@nikias @doronz88

nikias, 2024-10-30:
No, this essentially rewrites the encrypted data partition so you don't lose use your data.

E3lleyyaa, 2024-10-30:
@nikias Is this something new, or has it always been there? It’s the first time I’ve seen it, even though I’ve used it many times. Is there no way to skip it or handle the condition?

E3lleyyaa, 2024-10-30:
@nikias Thanks for the information! Yes, that's correct—the device was manually switched to recovery mode. I'll try again from normal mode.