libimobiledevice/idevicerestore #693: TSS request failed (status=94, message=This device isn't eligible for the requested build.)

RetroSynthwave, 2024-10-31:
so im getting TSS error 94 on all my platforms linux, mac os , windows on restoring, Ive tried different commits still getting error code 94 but if i do an update to latest iOS, the iphone updated to the latest version.. its the damn TSS that doesn't want to restore my iphone 14.3 .. i know its the cable because i have two iphones 7 global and few different cables they are MFi certified cables  
its their any commit i can compile and bypass the TSS ?
──(kali㉿Kali-Elemi)-[~]
└─$ idevicerestore --erase -dd --restore-mode --ticket "/home/kali/downr1n/blobs/iPhone9,1-14.3.shsh2" "/home/kali/ipsw/iPhone_4.7_P3_14.3_18C66_Restore.ipsw" 
Using ApTicket found at /home/kali/downr1n/blobs/iPhone9,1-14.3.shsh2 length 9905
irecv_event_cb: device 000a08813a130026 (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
found device with ECID 000a08813a130026
Setting to configuration 1
Setting to interface 0:0
Identified device as d10ap, iPhone9,1
Extracting BuildManifest from IPSW
Product Version: 14.3
Product Build: 18C66 Major: 18
opening device 05ac:1281...
found device with ECID 000a08813a130026
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is F73Z40R4HG6W
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/home/kali/ipsw/iPhone_4.7_P3_14.3_18C66_Restore/038-83137-079.dmg'
Getting device's ECID for TSS request
Found ECID 2824100885233702
Getting ApNonce in recovery mode... c9 28 e7 48 33 74 25 97 84 32 90 e3 a0 02 d9 b8 b0 9a 3e 28 02 6a 9b e5 22 43 6b bb 9d 11 5f 87 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 85 89 8e 7c 85 39 6d 9d d2 c2 96 fd 08 02 2c f7 1f 16 16 24 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,SystemVolumeCanonicalMetadata
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
[... 583 lines cut ...]
                <key>Digest</key>
                <data>
                </data>
        </dict>
        <key>ApNonce</key>
        <data>
        ySjnSDN0JZeEMpDjoALZuLCaPigCapvlIkNru50RX4c=
        </data>
        <key>@ApImg4Ticket</key>
        <true/>
        <key>ApSecurityMode</key>
        <true/>
        <key>ApProductionMode</key>
        <true/>
        <key>SepNonce</key>
        <data>
        hYmOfIU5bZ3Swpb9CAIs9x8WFiQ=
        </data>
</dict>
</plist>
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

RetroSynthwave, 2024-11-04:
Last login: Mon Nov  4 03:46:49 on ttys021
niteblue@Jacquess-Mini ~ % idevicerestore --erase -dd /Users/niteblue/iPhone_4.7_P3_14.3_18C66_Restore.ipsw
idevicerestore 1.0.0-git-04a3f49
idevice_event_cb: device 000a08813a130026 (udid: e0a9dfeb742fd9dfa50b824095581fe732970f9e) connected in normal mode
Found device in Normal mode
ECID: 2824100885233702
Identified device as d10ap, iPhone9,1
Device Product Version: 14.3
Device Product Build: 18C66
Extracting BuildManifest from IPSW
IPSW Product Version: 14.3
IPSW Product Build: 18C66 Major: 18
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in normal mode... 74 36 e3 31 22 ac 18 7a fc d5 a2 9b af d0 17 cf 3a 2f 79 e0 38 5b 28 69 ff 0f b4 f9 d8 c3 f9 f8 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... f9 11 86 84 e3 00 5d 65 41 49 cf dd 56 d8 07 d2 a3 e5 55 84 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,SystemVolumeCanonicalMetadata
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Homer
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
[... 658 lines cut ...]
* using HTTP/1.1
> POST /TSS/controller?action=2 HTTP/1.1
Host: gs.apple.com
User-Agent: InetURL/1.0
Accept: */*
Cache-Control: no-cache
Content-type: text/xml; charset="utf-8"
Content-Length: 10673

< HTTP/1.1 200 
< Server: Apple
< Date: Mon, 04 Nov 2024 12:22:30 GMT
< Content-Type: text/html
< Content-Length: 69
< Connection: keep-alive
< Host: gs.apple.com
< Strict-Transport-Security: max-age=31536000; includeSubdomains
< X-Frame-Options: SAMEORIGIN
< 
* Connection #0 to host gs.apple.com left intact
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
niteblue@Jacquess-Mini ~ %

nikias, 2024-11-04:
You cannot bypass the TSS requirement with this device. The only chance you would have is to use future restore and you have shsh blobs with saved generator value. If you don't know what that is then you probably won't have that and you cannot restore.

RetroSynthwave, 2024-11-07:
> You cannot bypass the TSS requirement with this device. The only chance you would have is to use future restore and you have shsh blobs with saved generator value. If you don't know what that is then you probably won't have that and you cannot restore.

not like bypass, TSS, but i remember during summer TSS wasen't a problem, but it was failed to enter  restore mode, my blob is invalid, its was generated from downr1n tether-down,  when i got it from someone, it was already on the ios 15.3. so i have two iphone 7 global neither of them were shsh blobs saves from  ios 14.. helped me out in here nikias, ive been trying for months to untethered ios 14.. which commits do i need, so i can compile them! and ive trying to compile future restore but it gives me an error, i choose idevicerestore commit e083987 on 97% compiling .. i know with compiling commit e083987. idevices can enter into restore mode for those who dont have valid shsh blobs like mine.. ..