libimobiledevice/idevicerestore #695: iPad7,11 iPadOs 18.1 image personalization failed.

reimerp, 2024-11-18:
I tried this tool, concrats to develop this, since itnse is a bit tricky to ose an less verbose when it comes to errors.

I tried on a ipad 7 10" (2019) which was locked in boot loop. I can only use 18.1 (which is out for some days and seems the cause of the loop) and maybe this is still unsupported (?) or maybe TDD answer changed (There are some no entry fon in response in the logs)
I was using Args: `-l -e -y -d -d`
Here is my log:

```
idevicerestore
idevicerestore 1.0.0-223-g61a76ce (libirecovery 1.2.1-5-g7a5e619, libtatsu 1.0.4-2-g1d31ade)
irecv_event_cb: device 000a24e0217b0026 (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
found device with ECID 000a24e0217b0026
Setting to configuration 1
Setting to interface 0:0
ECID: 2855294820089894
Identified device as j171ap, iPad7,11
Device Product Version: N/A
Device Product Build: N/A
* Host api.ipsw.me:443 was resolved.
* IPv6: 2606:4700:20::681a:89c, 2606:4700:20::681a:99c, 2606:4700:20::ac43:47bb
* IPv4: 172.67.71.187, 104.26.9.156, 104.26.8.156
*   Trying [2606:4700:20::681a:89c]:443...
* Connected to api.ipsw.me (2606:4700:20::681a:89c) port 443
* ALPN: curl offers h2,http/1.1
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384 / X25519 / id-ecPublicKey
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=ipsw.me
*  start date: Oct 26 00:50:32 2024 GMT
*  expire date: Jan 24 00:50:31 2025 GMT
*  subjectAltName: host "api.ipsw.me" matched cert's "*.ipsw.me"
*  issuer: C=US; O=Google Trust Services; CN=WE1
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
*   Certificate level 0: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA256
*   Certificate level 1: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA384
*   Certificate level 2: Public key type EC/secp384r1 (384/192 Bits/secBits), signed using sha256WithRSAEncryption
* using HTTP/2
* [HTTP/2] [1] OPENED stream for https://api.ipsw.me/v4/device/iPad7,11
* [HTTP/2] [1] [:method: GET]
* [HTTP/2] [1] [:scheme: https]
* [HTTP/2] [1] [:authority: api.ipsw.me]
* [HTTP/2] [1] [:path: /v4/device/iPad7,11]
* [HTTP/2] [1] [user-agent: InetURL/1.0 idevicerestore/1.0.0-223-g61a76ce]
* [HTTP/2] [1] [accept: */*]
> GET /v4/device/iPad7,11 HTTP/2

Host: api.ipsw.me

User-Agent: InetURL/1.0 idevicerestore/1.0.0-223-g61a76ce

Accept: */*

* old SSL session ID is stale, removing
< HTTP/2 200 

< date: Mon, 18 Nov 2024 18:21:36 GMT

< content-type: application/json

< vary: Accept-Encoding

< vary: Origin

< access-control-allow-methods: GET, OPTIONS

< access-control-allow-origin: *

< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'

[... 1000 lines cut ...]
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.j171ap.im4p (Firmware/all_flash/DeviceTree.j171ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (40853 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j171.RELEASE.im4p (Firmware/all_flash/sep-firmware.j171.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (1435840 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad7c (kernelcache.release.ipad7c)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (16714250 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 000a24e0217b0026 (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 000a24e0217b0026 (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Device reconnected in recovery mode, most likely image personalization failed.

```

fmatrixch, 2024-11-24:
I have the same problem too

reimerp, 2024-12-21:
The "device reconnected in recovery" seems that it should boot to "upgrade" mode (apple logo with progress bar) instead it boots to recovery mode (screen which shows usb cable).
3uTool having same issue (18% bug) and the cause of this seems a faulty device. It was reported battery but I already loaded battery to almost full. It could be a generic flash issue (end of life or bootloader corrupt) so it seems not a software issue in my case

LexNastin, 2024-12-28:
Same issue but on MacBook 18,3

LexNastin, 2024-12-28:
Found solution! Mismatched deps somewhere. Rebuilt on clean Ubuntu (making sure to build every single dep made by libimobiledevice instead of apt installing any of them) like in issue #684, and it works now!k

reimerp, 2024-12-28:
I used docker from #562, which is building every dep from source