libimobiledevice/idevicerestore #697: ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)

DaBlincx, 2024-11-24:
I'm trying to "update" an iPhone 12 currently on 18.1 to 18.1 because it is boot-looping. iTunes just gave some error 2003 so I tried idevicerestore. It fails with this error:
```
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
```
This also happens if I manually specify the shsh blob for 18.1 which i saved using blobsaver. 

I read in issue #615 that there was an issue with libtatsu and it has been fixed in 1.0.3, but I'm using 1.0.4 and it's still broken.

E3lleyyaa, 2024-11-28:
I have the same with Iphone 13 pro.
has been used the latest lib versions.

E3lleyyaa, 2024-11-28:
@nikias

dmmc12, 2025-01-02:
```
$ pkg-config --modversion libtatsu-1.0
1.0.4-8-g7e1647b
```

+1 to @DaBlincx message ^ 

```

$ idevicerestore -l
Found device in Recovery mode
Identified device as d321ap, iPhone11,2
The following firmwares are currently being signed for iPhone11,2:
  [1] 18.2 (build 22C152)
Select the firmware you want to restore: 1
Selected firmware 18.2 (build 22C152)
Downloading firmware (https://updates.cdn-apple.com/2024FallFCS/fullrestores/072-42099/B947F49A-70F5-4388-8D5E-2472A1733D15/iPhone11,2,iPhone11,4,iPhone11,6_18.2_22C152_Restore.ipsw)

downloading: 99%

Verifying 'iPhone11,2,iPhone11,4,iPhone11,6_18.2_22C152_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 18.2
Product Build: 22C152 Major: 22
INFO: device serial number is [REDACTED]
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 092-10261-200.dmg.aea
[==================================================] 100.0%
Found ECID [REDACTED]
Getting ApNonce in recovery mode... [REDACTED]
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... [REDACTED]
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

nikias, 2025-01-02:
Are you using the latest build of idevicerestore? The first line it prints is the version of idevicerestore, followed by the version of libirecovery and libtatsu. If it doesn't print it it suggests that it's an older version

dmmc12, 2025-01-02:
```
$ idevicerestore --version
idevicerestore 1.0.0
```

Thanks for following up so quickly! I can confirm that I'm running the latest version of `idevicerestore`. Interestingly, while troubleshooting the TSS failure when connecting to `gs.apple.com`, I noticed a certificate issue when using `curl` to test the domain.

```
$ ping gs.apple.com
curl -v https://gs.apple.com
PING gs.apple.com.v.aaplimg.com (17.111.103.15) 56(84) bytes of data.
64 bytes from 17.111.103.15 (17.111.103.15): icmp_seq=1 ttl=46 time=227 ms
64 bytes from 17.111.103.15 (17.111.103.15): icmp_seq=2 ttl=46 time=146 ms
64 bytes from 17.111.103.15 (17.111.103.15): icmp_seq=3 ttl=46 time=169 ms
^C
--- gs.apple.com.v.aaplimg.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 145.992/180.475/226.550/33.893 ms
*   Trying 17.111.103.15:443...
* Connected to gs.apple.com (17.111.103.15) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.0 (OUT), TLS header, Certificate Status (22):
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (OUT), TLS header, Unknown (21):
* TLSv1.2 (OUT), TLS alert, unknown CA (560):
* SSL certificate problem: self-signed certificate in certificate chain
* Closing connection 0
curl: (60) SSL certificate problem: self-signed certificate in certificate chain
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
```

Initially, I updated and reinstalled the CA certificates on my machine:

```
sudo apt update
sudo apt install --reinstall ca-certificates
sudo update-ca-certificates
```

Unfortunately, the issue persisted. To resolve this, I manually added the certificate:

```
echo | openssl s_client -showcerts -servername gs.apple.com -connect gs.apple.com:443 2>/dev/null | openssl x509 > apple_cert.pem
sudo cp apple_cert.pem /usr/local/share/ca-certificates/apple_cert.crt
sudo update-ca-certificates
```

After updating the certificates, I retried the restoration with debug and logging enabled:

```
idevicerestore -e -l -y -dd > stdout.txt 2> stderr.txt
```

This appears to have resolved the issue, allowing me to restore my phone. However, it’s possible the problem was due to a latent issue that I can’t fully explain. 

I truly appreciate this though—without access to a Mac and with boot looping occurring more frequently, this tool has been incredibly helpful to me. Thank you!

E3lleyyaa, 2025-01-14:
@dmmc12 Unfortunately, it didn’t help me. Still  Unable to send TSS request...

PedroxionX, 2025-03-05:
@E3lleyyaa did you find any way to fix the problem?