libimobiledevice/idevicerestore #703: Can't restore ipsw on First Generation iPhone 2G

Hookedon3D, 2024-12-18:
It errors out and I cant' seem to restore a firmware. The iPhone used to be jailbroken and all I get is the apple logo now. I'd like to be able to restore any iphone firmware on it and make it functional again. This is the error I keep getting.

idevicerestore -s '/home/prometheus/Downloads/iPhone1,1_3.1.2_7D11_Restore.ipsw' 
Found device in Recovery mode
Identified device as m68ap, iPhone1,1
Extracting BuildManifest from IPSW
Product Version: 3.1.2
Product Build: 7D11 Major: 7
INFO: device serial number is 87752F7DWH8
Device supports Image4: false
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/home/prometheus/Downloads/iPhone1,1_3.1.2_7D11_Restore/018-6028-014.dmg'
Extracting iBEC.m68ap.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (105844 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-636.66
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo.s5l8900x.img3...
Not personalizing component RestoreLogo...
Sending RestoreLogo (10740 bytes)...
ramdisk-size=RELEASE
Extracting 018-6134-014.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (12615028 bytes)...
Extracting DeviceTree.m68ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (42228 bytes)...
Extracting kernelcache.release.s5l8900x...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (4077492 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 12
Device 537af94f998005f9b25429efaf9d5ece69cf9ddf has successfully entered restore mode
Hardware Information:
BoardID: 0
ChipID: 35072
UniqueChipID: 2786696772911
ProductionMode: true
Previous restore exit status: 0x100
USB log is available:
USBL8028 GINTMSK=a00c3800 GNPTXSTS=08080100 DAINT=00000000
DAINTMSK=0015000b DCFG=000c1c04 DCTL=00000000 DSTS=000ac800
GRXFSIZ=0000011b GNPTXFSIZ=0100011b DPTXSIZ(1)=0300021b 
GAHBCFG=0000002b, GUSBCFG=00002c08, GRSTCTL=80000000 
     DEPCTL   DEPINT   DEPTSIZ  DEPDMA
EP80 00008800 00000000 00000000 09aaf000
EP00 80008000 00000000 20080040 09a4bd40
EP81 00099a00 00000010 20000000 09b2f330
EP83 000a8200 00000000 00000000 00000000
EP85 00000000 00000000 00000000 00000000
EP02 80098200 00000000 02008000 09bd9000
EP04 000a8200 00000000 00000000 00000000

intOut:epNum:2 rcvd:0 ioLen:32768
startEndpointOut:DOEPDMA 09bae000 
startEndpointOut:DOEPTSIZ(2) 02008000 
startEndpointOut:Finish starEndpointOUT(2) 
handleEndpointIO:ep:02
handleEndpointIO:ep:81
startEndpointIn:epNum:1 sent:0 ioLen:371
startEndpointIn:DMA addr 09b2d01c 
startEndpointIn:DIEPTSIZ(1) 20080173 
startEndpointIn:Finish startEndpointIN(1) 
[... 235 lines cut ...]
ASR: done
ASR STATUS: metadata
ASR: Validating source...
AppleUSBDeviceMux::handleConnectResult new session 0xc250f980 established 12345<-lo0->49171 12345<-usb->20
ASR: done
ASR: Warning: the target volume may not be bootable
ASR: Erasing target device /dev/disk0s1...
ASR: done
ASR: Retrieving scan information...done
ASR: Validating sizes...
ASR: done
ASR STATUS: restore
void AppleUSBDeviceMux::handleMuxTCPInput(__mbuf*) received reset, closing 0xc250f980
ASR: asr: \nAttempt to read() offset 0x0 from socket failed with errno of 54
ASR: asr: Failed to read the stream: Connection reset by peer
ASR: __decompressChunk: backing store read at offset 393 length 2271 failed with error 54
ASR: __decompressChunk() error: 54__decompressChunk: backing store read at offset 6165 length 1167 failed with error 54
ASR: __decompressChunk() error: 54__decompressChunk: backing store read at offset 10833 length 1167 failed with error 54
ASR: __decompressChunk() error: 54__decompressChunk: backing store read at offset 15501 length 1167 failed with error 54
ASR: __decompressChunk() error: 54
ASR STATUS: fail
ASR: Could not restore - Connection reset by peer
ASR STATUS: fail

ERROR: Unable to restore device

suchac500, 2025-01-13:
hello!
+1
Exactly same problem, any solution? 
I have tried multiple cables, pc´s and OS. :/

Hookedon3D, 2025-01-13:
Unfortunately, not yet. I have tried basically everything but didn't get anywhere. I was hoping someone with more knowledge would respond with a solution.

suchac500, 2025-01-13:
haha thanks for the fast reply. I made registration just because of this topic :-D

anyway same here, for me i had funcional iphone 2g with data from the (jailbroken) previous owner, i made restore via setting in the iphone. Did reboot with loading bar and thats it with "connect to itunes" screen. 

Firstly I thought the battery is weak, so i change it. No improvement there. 
itunes in W11 wont find iphone in DFU (error 2005), WindowsXP on old laptop will write me usb device is not recognazible. On my own pc i have 2 good usb where is iphone recognized by itunes or other tools in recovery mode. I will put it to DFU and im basically done.

I tried to used VM´s 
XP VM - will recognize ipone every time, but redsnow will get stuck "waiting for reboot"
W7 VM - even wont find iphone
Mac - with legacy ios toolkit will find it, but will get timeout about 60-70% when uploading first item.

I would suggest HW error, but when iphone was working normally before restore. Its quite wierd.

suchac500, 2025-01-13:
Legacy toolkit isnt working for you? 

I made a few good restores on 3g/3gs with it without problem.

It looks like that im not able to push any data to the iphone. Would understand bad cable but dont think this will be the case. 
Im farest point is basically Legacy IOS toolkit, when im getting error at 62.1% during uploading WTF package. 

But in windows with other app im also getting ASR FAIL status and getting denied at 34%.

Also when i do some check iphone write me no ios inside.

Hookedon3D, 2025-01-13:
I just downloaded it trying to figure out how to run it on Linux right now.

suchac500, 2025-01-13:
let me know it was flawless experience for me but on 3g/3gs

suchac500, 2025-01-13:
[image]

suchac500, 2025-01-13:
okey so here is the update: i dont know how, BUT IT PASSED.
It was pain in the as.. however.

I used Virtual Machine Big Sur mac, I did literally everything...
Always i had plenty of errors, during uploading WTF package, or for timetout for DFU device. 

it always failed. I just now tried to one more go, before give up. 
1) I put it on the charger for while, but dont think it changed something because i did even before with no help. But still recommend.
2) I changed (again) USB cable, i had 3 every behave different, i didnt suspect cable because before that i used all of them for 3g restore and it was good. But would recommend you check which is more stable or did you farest.
3) I have tried second usb (as many times before). Now it worked from some miracle reason. I gave it a try and now WTF package was sent sucessfully and rest went without single error. I selected their IPSW 3.1.3 and then restore/download.

Really dont understand why, for love of god, 2000x before that it gave me error. And then without anyspecific reason it past through.

Let me know how did it went for u, or DM me, I will see if I will be able to help.
Finally after three days i can go sleep. :D

suchac500, 2025-01-13:
[image]
Also some screens. 
I got these error massages through the whole process frome 62.1-97%, also when it passed with 100% it timeout immidiately after. So before today i have never were more far than this step.

When i saw loading bar with progress i was so happy... ufffffffff. (because before that i saw only empty and then i got error)

Hookedon3D, 2025-01-13:
OMG, dude, you are Godsent! After a few tries I was able to restore 3.13.3 and I am on the homescreen. It worked! lol

suchac500, 2025-01-13:
Yeah what is funny, i tried it thousand times one week before... I was still stuck at sending patched WTF.

I did everything same as before, but today we are lucky lol.

Take care mate!

Hookedon3D, 2025-01-13:
Thanks, mate, you too! lol

closed 2025-01-13 by Hookedon3D