libimobiledevice/idevicerestore #704: Cant restore iPhone14,2 IOS 18.2

E3lleyyaa, 2024-12-19:
```

idevicerestore.exe -l -y -P --debug
idevicerestore 1.0.0-223-g61a76ce (libirecovery 1.2.1-5-g7a5e619, libtatsu 1.0.4-4-g7661cad)
progress: 0 0.000000
idevice_event_cb: device 0004799e2e69801e (udid: 00008110-0004799E2E69801E) connected in normal mode
progress: 0 0.100000
Found device in Normal mode
ECID: 1259620197302302
progress: 0 0.200000
Identified device as d63ap, iPhone14,2
Device Product Version: 18.1.1
Device Product Build: 22B91
* Host api.ipsw.me:443 was resolved.
* IPv6: (none)
* IPv4: 104.26.9.156, 104.26.8.156, 172.67.71.187
*   Trying 104.26.9.156:443...
* Connected to api.ipsw.me (104.26.9.156) port 443
* ALPN: curl offers http/1.1
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384 / x25519 / id-ecPublicKey
* ALPN: server accepted http/1.1
* Server certificate:
*  subject: CN=ipsw.me
*  start date: Oct 26 00:50:32 2024 GMT
*  expire date: Jan 24 00:50:31 2025 GMT
*  subjectAltName: host "api.ipsw.me" matched cert's "*.ipsw.me"
*  issuer: C=US; O=Google Trust Services; CN=WE1
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
*   Certificate level 0: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA256
*   Certificate level 1: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA384
*   Certificate level 2: Public key type EC/secp384r1 (384/192 Bits/secBits), signed using sha256WithRSAEncryption
* using HTTP/1.x
> GET /v4/device/iPhone14,2 HTTP/1.1
Host: api.ipsw.me
User-Agent: InetURL/1.0 idevicerestore/1.0.0-223-g61a76ce
Accept: */*

* Request completely sent off
< HTTP/1.1 200 OK
< Date: Thu, 19 Dec 2024 09:09:48 GMT
< Content-Type: application/json
< Transfer-Encoding: chunked
< Connection: keep-alive
< Vary: Accept-Encoding
< Vary: Origin
< Access-Control-Allow-Methods: GET, OPTIONS
< Access-Control-Allow-Origin: *
< Content-Security-Policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< X-Content-Type-Options: nosniff
< X-Data-Location: database
< X-Frame-Options: DENY
< X-Frame-Options: DENY
< X-Xss-Protection: 1; mode=block
< Strict-Transport-Security: max-age=15552000; preload
< Last-Modified: Thu, 19 Dec 2024 07:56:37 GMT
< Cache-Control: max-age=14400
< CF-Cache-Status: HIT
< Age: 2384
< Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=rcZBq%2FUufwYwkbRT8e1c50pDh%2F9os8qK%2F%2FpYltaIT5wvZyZ8vYrXZxV%2BDcc%2FdJUFGC3V62J%2Bcfz1nb8XsnsLqhdXS0kwW4ZIQj8s86Y6ISnZPLrke%2Fc2R0mneMkq"}],"group":"cf-nel","max_age":604800}
< NEL: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
< Server: cloudflare
< CF-RAY: 8f464184cb175b88-VIE
< alt-svc: h3=":443"; ma=86400
< server-timing: cfL4;desc="?proto=TCP&rtt=28231&min_rtt=28062&rtt_var=10644&sent=4&recv=6&lost=0&retrans=0&sent_bytes=2827&recv_bytes=748&delivery_rate=104055&cwnd=251&unsent_bytes=0&cid=4d0e1227c1f476c3&ts=57&x=0"
<
* Connection #0 to host api.ipsw.me left intact
NOTE: Running non-interactively, automatically selecting latest available version
Selected firmware 18.2 (build 22C152)
Downloading firmware (https://updates.cdn-apple.com/2024FallFCS/fullrestores/072-43504/D0406442-E480-48DB-B130-3EDA1BE95E3D/iPhone14,2_18.2_22C152_Restore.ipsw)
* Host updates.cdn-apple.com:443 was resolved.
[... 192 lines cut ...]
DEBUG: found component siof (SIO)
DEBUG: found component trst (StaticTrustCache)
DEBUG: found component isys (SystemVolume)
DEBUG: found component wchf (WCHFirmwareUpdater)
DEBUG: found component ibot (iBoot)
DEBUG: found component ibdt (iBootData)
DEBUG: creating stashbag...
Device requires stashbag.
******************************************************************************
* Please enter your passcode on the device.  The device will store a token   *
* that will be used after restore to access the user data partition.  This   *
* prevents an 'Attempting data recovery' process occurring after reboot that *
* may take a long time to complete and will _also_ require the passcode.     *
******************************************************************************
Stashbag created.
Getting ApNonce in normal mode... a1 7f 45 06 61 08 0d cc ff 25 af ac f3 7a 36 7f 47 2f 77 5f 17 c9 a4 00 76 55 ea bd e6 1b 60 dd
Getting SepNonce in normal mode...
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 00 b6 ab 0b c0 d1 74 a8 d2 ac fd 82 34 a4 14 c8 f4 c3 9c a7
TSS server returned: STATUS=168&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 168
ERROR: TSS request failed (status=168, message=An internal error occurred.)
ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
```

E3lleyyaa, 2024-12-19:
@nikias Could you please tell me what might be the issue?

E3lleyyaa, 2025-01-13:
@doronz88 Hi, perhaps you know something about this error and how to avoid it?

doronz88, 2025-01-13:
Sounds like a ticket error. You can try with `pymobiledevice3` or Apple Configurator to see if it works.
If it does fix the issue, we'll need to dump the ticket request and compare the two

E3lleyyaa, 2025-01-13:
> Sounds like a ticket error. You can try with `pymobiledevice3` or Apple Configurator to see if it works. If it does fix the issue, we'll need to dump the ticket request and compare the two

Sure, I will try. 
The strangest thing is that everything works fine with the iPhone 14, and I also tried reinstalling the current version on the iPhone 8, and there were no issues anywhere. However, with the iPhone 13 Pro, regardless of the firmware, this error always occurs.

Trenki, 2025-01-20:
> > Sounds like a ticket error. You can try with `pymobiledevice3` or Apple Configurator to see if it works. If it does fix the issue, we'll need to dump the ticket request and compare the two听起来像是票错误。您可以尝试使用`pymobiledevice3`或 Apple Configurator 看看是否有效。如果确实解决了问题，我们需要转储票证请求并比较两者
> 
> Sure, I will try.  当然，我会尝试。 The strangest thing is that everything works fine with the iPhone 14, and I also tried reinstalling the current version on the iPhone 8, and there were no issues anywhere. However, with the iPhone 13 Pro, regardless of the firmware, this error always occurs.最奇怪的是，在 iPhone 14 上一切正常，我也尝试在 iPhone 8 上重新安装当前版本，没有任何问题。然而，对于iPhone 13 Pro，无论固件如何，总是会出现此错误。

Hello, have you solved this problem? It seems I'm encountering a similar issue.