libimobiledevice/idevicerestore #706: iPad Mini 4 - No device found! - Virtual Box

patienttruth, 2025-01-04:
I'm not sure if I'm barking up the wrong tree, but I feel like I'm real close to getting this to work...

I'm running `sudo usbmuxd -fvv` and I get the following when I connect the device

```
[1774.242714] [000038f7] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[1774.242861] [000038f7] libusb: debug [linux_get_device_address] scan 2-1
[1774.242944] [000038f7] libusb: debug [linux_get_device_address] bus=2 dev=13
[1774.242951] [000038f7] libusb: debug [udev_hotplug_event] udev hotplug event. action: add.
[1774.242955] [000038f7] libusb: debug [linux_enumerate_device] busnum 2 devaddr 13 session_id 525
[1774.242960] [000038f7] libusb: debug [linux_enumerate_device] allocating new device for 2/13 (session 525)
[1774.243060] [000038f7] libusb: debug [linux_get_parent_info] dev 0x75cb38004160 (2-1) has parent 0x5d6cf786c360 (usb2) port 1
[1774.243101] [000038f6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[1774.243108] [000038f6] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 0ms
[1774.243113] [000038f6] libusb: debug [usbi_wait_for_events] poll() returned 1
[1774.243117] [000038f6] libusb: debug [handle_event_trigger] event triggered
[1774.243121] [000038f6] libusb: debug [handle_event_trigger] hotplug message received
[1774.243127] [000038f6] libusb: debug [libusb_get_device_descriptor]  
[1774.270479] [000038f7] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[1774.270500] [000038f7] libusb: debug [linux_get_device_address] scan 2-1
[1774.270561] [000038f7] libusb: debug [linux_get_device_address] bus=2 dev=13
[1774.270565] [000038f7] libusb: debug [udev_hotplug_event] udev hotplug event. action: bind.
```
Seems like a good sign...

However when I run `ideviceinfo` in another terminal I get `ERROR: No device found!` and this from `usbmuxd`
```
[23:09:07.431][4] Client 10 accepted: ideviceinfo[15586]
[23:09:07.431][5] Client 10 command len 440 ver 1 msg 8 tag 1
[23:09:07.431][5] Client 10 output buffer got tag 1 msg 8 payload_length 221
[23:09:07.433][4] Client 10 connection closed
[23:09:07.433][4] Client 10 is going to be disconnected: /proc/15586/cmdline[15586]
```

The device is stuck in restore mode after I tried resetting it with Apple Devices. I have a second iPad that is not stuck in restore mode, however I get the same type of logs.

I have no idea what version of iOS they're running, but from reading posts it seems they should work.  Any suggestion for a next step?

Thanks!

nikias, 2025-01-04:
ideviceinfo doesn't work in restore mode, that's why it just closes the connection. You need to reset the device. Try the hard reboot button combination for the device (should be power key + home button for that model) and it should get back to recovery mode.

patienttruth, 2025-01-04:
> ideviceinfo doesn't work in restore mode, that's why it just closes the connection. You need to reset the device. Try the hard reboot button combination for the device (should be power key + home button for that model) and it should get back to recovery mode.

Thank you, I'll give these another try soon.

I may be using the terms imprecisely. For my understanding, how do you tell the difference between restore and recovery?

The device that I started with (via Apple Devices) only boots to a screen showing the USB plugging into a laptop with the URL.../restore.

The other iPad will boot to the pin screen, and if I hold top key and home it reboots to a similar screen that shows "iTunes" logo instead of a laptop, otherwise it is similar with the same URL.

My goal is to reset both devices, am I approaching this wrong?

patienttruth, 2025-01-07:
Nikias, I have tested with a functional iPhone and am able to connect, so I think my builds worked correctly.

I reset the iPad by holding top key and home key until the plug and iTunes logo showed up. I ran `idevicerestore --erase --latest
`
It says:
```
idevicerestore 1.0.0-226-gbb5591d (libirecovery 1.2.1-9-g0f0928a, libtatsu 1.0.4-8-g7e1647b)
ERROR: Unable to discover device mode. Please make sure a device is attached.
```

Here is the output from `usbmuxd -fvv`

```
[17:00:05.989][4] Client 10 accepted: idevicerestore[22949]
[17:00:05.989][5] Client 10 command len 438 ver 1 msg 8 tag 1
[17:00:05.989][5] Client 10 output buffer got tag 1 msg 8 payload_length 278
[17:00:05.989][5] Client 10 now LISTENING
[17:00:16.743][4] Client 10 connection closed
[17:00:16.743][4] Client 10 is going to be disconnected: idevicerestore[22949]
```

I tried making the logs even more verbose, but there is nothing obvious to me in them. Any suggestions? Do I need to have unlocked the device before resetting?

I'd read that sometimes a mac can do a better job at factory resetting an iPad so I may try when I see my brother soon who has one, but my hope finding this excellent project was to avoid the dependency on the hardware, if possible.