libimobiledevice/idevicerestore #708: idevicerestore and iphones using linux

wshekrota, 2025-01-16:
I am using Ubuntu 24.04 the very most recent LTS drop. My theory was to get as close to where you are submitting your working code.
I have gone down the path of trying to build from scratch but get tangled in a versioning issue of libplist I believe. The build is much too tangled to build by hand just IMHO. So dropping this path I moved back to Ubuntu 24.04 LTS and its current packages.
So by just exposing your necessary code for idevicerestore using apt I get a system that almost works.
I get the code  into the restore phase about 50% at which it always loops waiting a response. I saw online where you have told people its a known bug that was fixed and to build from scratch.
I would much rather know how your working code travels into the linux ecosystem. Given the right build it should just work like your copy. My thoughts were it starts in SID Debian?
Thanks for any directions or ideas I'm quite excited about this package.
-Walt

AiXanadu, 2025-01-18:
libplist-master/autogen.sh --without-cython --disable-static PACKAGE_VERSION=2.6.0

I use the source code downloaded directly from zip, and then compile it by specifying the version.

wshekrota, 2025-01-18:
Hard to remember I think I got through that but versions were flagged.What I hadLibimobiledevice  Libimobiledevice-glue  LibtatsuThere were versions called out in two places the other in libtatsu.Checking for libtatsu-1.0 >= 1.0.3… no ( on the autogen.sh)I became untrusting of my forward progress and gave up…The question I had asked was. Did the working code of libimobiledevice get submitted to linux?ThanksWaltSent from my iPad

wshekrota, 2025-01-18:
What I see is I am forced to dl multiple sources.Libimobiledevice will require that libtatsu and that glue lib also.WaltSent from my iPad

wshekrota, 2025-01-18:
I am having mismatches libplist libtatsu etc
Are these all bundled in one somewhere so as to assure me I have a valid version collection?
If so can I get a pointer to that.
Thanks.
-Walt

AiXanadu, 2025-01-21:
You build it as follows

libplist-master/autogen.sh --without-cython --disable-static PACKAGE_VERSION=2.6.0
libimobiledevice-glue-master/autogen.sh --without-cython --disable-static PACKAGE_VERSION=1.3.0
libusbmuxd-master/autogen.sh --disable-static PACKAGE_VERSION=2.1.0
libtatsu-master/autogen.sh --without-cython --disable-static PACKAGE_VERSION=1.0.3
libimobiledevice-master/autogen.sh --without-cython --disable-static PACKAGE_VERSION=1.3.1
libirecovery-master/autogen.sh --disable-static PACKAGE_VERSION=1.2.0
libideviceactivation-master/autogen.sh --disable-static PACKAGE_VERSION=1.1.1
idevicerestore-master/autogen.sh PACKAGE_VERSION=1.0.1
usbmuxd-master/autogen.sh PACKAGE_VERSION=1.1.2

wshekrota, 2025-01-21:
Nah like I was suspicious that ends up tangled.It was good down to recovery then i had problems with libxml and libzip also noted tatsu required 1.0.4. By recovery I think.Anything like this rarely manually compiles.My LTS is 24.04 absolutely current stable.I also had to backup libusb to a previous version. I think 1.0The solution as I previously suggested is to get it in the linux ecosystem system properly then use that distro tools in a stable mannerI do appreciate the answer though…an hour laterWaltSent from my iPad

wshekrota, 2025-01-22:
I finally got everything compiled. In some cases needed -dev libs.On the idevicerestore this build gets to the white apple screen with empty progress bar. Ends there on waiting for device to enter rescue mode.At least the build from apt in 24.04 got to 50% progress there.I have 3 devices to do now I have two that loop into restore mode ie usb cable to computer screen. Trying to reset this manually loops back to same restore mode.Is there a command line to bring phone back to non dfu mode so can start again?ThanksWalt Sent from my iPad

wshekrota, 2025-01-24:
Ok I got this to compile and restore completely on ubuntu 24.04 Noble Numbat the current LTS.I used your template of packages below with the following changes.. PACKAGE_VERSIONS were bumpedLibirecovery 1.2.1Libtatsu 1.0.4Also required dev libs  libzip-dev and libxml2-dev.Happy sailing.-WaltSent from my iPad