libimobiledevice/idevicerestore #709: Failed to Update iPhone 11 Pro Max to iOS 18.2.1，ERROR: tss_send_request: Unhandled status code 168

Trenki, 2025-01-20:
I've been trying to solve this problem for several days, but still haven't found a solution. Thank you so much for your help.

I used the `idevicerestore --latest --d` command and printed out the following information:
```
idevicerestore 1.0.0 (libirecovery 1.2.1-5-g7a5e619-dirty, libtatsu 1.0.4)
idevice_event_cb: device 0011490e0184802e (udid: 00008030-0011490E0184802E) connected in normal mode
Found device in Normal mode
ECID: 4865399107911726
Identified device as d431ap, iPhone12,5
Device Product Version: 18.2
Device Product Build: 22C152
* Host api.ipsw.me:443 was resolved.
* IPv6: (none)
* IPv4: 172.67.71.187, 104.26.8.156, 104.26.9.156
*   Trying 172.67.71.187:443...
* Connected to api.ipsw.me (172.67.71.187) port 443
* ALPN: curl offers h2,http/1.1
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384 / x25519 / id-ecPublicKey
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=ipsw.me
*  start date: Dec 24 00:48:53 2024 GMT
*  expire date: Mar 24 01:48:23 2025 GMT
*  subjectAltName: host "api.ipsw.me" matched cert's "*.ipsw.me"
*  issuer: C=US; O=Google Trust Services; CN=WE1
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
*   Certificate level 0: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA256
*   Certificate level 1: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA384
*   Certificate level 2: Public key type EC/secp384r1 (384/192 Bits/secBits), signed using sha256WithRSAEncryption
* using HTTP/2
* [HTTP/2] [1] OPENED stream for https://api.ipsw.me/v4/device/iPhone12,5
* [HTTP/2] [1] [:method: GET]
* [HTTP/2] [1] [:scheme: https]
* [HTTP/2] [1] [:authority: api.ipsw.me]
* [HTTP/2] [1] [:path: /v4/device/iPhone12,5]
* [HTTP/2] [1] [user-agent: InetURL/1.0 idevicerestore/1.0.0]
* [HTTP/2] [1] [accept: */*]
> GET /v4/device/iPhone12,5 HTTP/2
Host: api.ipsw.me
User-Agent: InetURL/1.0 idevicerestore/1.0.0
Accept: */*

* Request completely sent off
* old SSL session ID is stale, removing
< HTTP/2 200
< date: Mon, 20 Jan 2025 12:58:14 GMT
< content-type: application/json
< vary: Accept-Encoding
< vary: Origin
< access-control-allow-methods: GET, OPTIONS
< access-control-allow-origin: *
< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< x-content-type-options: nosniff
< x-data-location: database
< x-frame-options: DENY
< x-frame-options: DENY
< x-xss-protection: 1; mode=block
< strict-transport-security: max-age=15552000; preload
< last-modified: Mon, 20 Jan 2025 11:57:50 GMT
< cache-control: max-age=14400
< cf-cache-status: HIT
< age: 1402
< report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=ia8CMlfB9fVG0mOCZ62nRtGqP5baFutvwUhiGL8h5ILo2q9mvfiIFkwnSfV9k2NkpKAzpKGdSCGKv3GHgtZsphfk80qybCJsha90TKNDD4sZXE%2FuRiRHtRomQJUB"}],"group":"cf-nel","max_age":604800}
< nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
< server: cloudflare
< cf-ray: 904f3c1f389d057f-IAD
< alt-svc: h3=":443"; ma=86400
< server-timing: cfL4;desc="?proto=TCP&rtt=332251&min_rtt=323039&rtt_var=106879&sent=6&recv=7&lost=0&retrans=0&sent_bytes=3413&recv_bytes=774&delivery_rate=10574&cwnd=33&unsent_bytes=0&cid=ebf77b6a5e1aada9&ts=387&x=0"
<
* Connection #0 to host api.ipsw.me left intact
The following firmwares are currently being signed for iPhone12,5:
  [1] 18.2.1 (build 22C161)
Select the firmware you want to restore: 1
Selected firmware 18.2.1 (build 22C161)
Verifying 'iPhone12,3,iPhone12,5_18.2.1_22C161_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
IPSW Product Version: 18.2.1
IPSW Product Build: 22C161 Major: 22
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Checking if device requires stashbag...
DEBUG: found component anef (ANE)
DEBUG: found component aopf (AOP)
DEBUG: found component avef (AVE)
DEBUG: found component msys (Ap,SystemVolumeCanonicalMetadata)
DEBUG: found component logo (AppleLogo)
DEBUG: found component chg0 (BatteryCharging0)
DEBUG: found component chg1 (BatteryCharging1)
DEBUG: found component batF (BatteryFull)
DEBUG: found component bat0 (BatteryLow0)
DEBUG: found component bat1 (BatteryLow1)
DEBUG: found component glyP (BatteryPlugin)
DEBUG: found component dtre (DeviceTree)
DEBUG: found component gfxf (GFX)
DEBUG: found component ispf (ISP)
DEBUG: found component illb (LLB)
DEBUG: found component lpw0 (LowPowerWallet0)
DEBUG: found component lpw1 (LowPowerWallet1)
DEBUG: found component pmpf (PMP)
DEBUG: found component recm (RecoveryMode)
DEBUG: found component rsep (RestoreSEP)
DEBUG: found component rtsc (RestoreTrustCache)
DEBUG: found component sepi (SEP)
DEBUG: found component siof (SIO)
DEBUG: found component trst (StaticTrustCache)
DEBUG: found component isys (SystemVolume)
DEBUG: found component wchf (WCHFirmwareUpdater)
DEBUG: found component ibot (iBoot)
DEBUG: creating stashbag...
Device does not require stashbag.
Getting ApNonce in normal mode... 3f ff c3 bf 78 87 74 ec ea a9 0e 7d 96 1a 6c ed 63 93 c9 44 f7 f1 6b d9 70 35 61 eb 38 1c 9a 99
Getting SepNonce in normal mode...
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... c8 6f 47 53 a0 57 97 f8 ae 78 68 1d 5d 1c ac 96 6b ad f9 2e
TSS server returned: STATUS=168&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 168
ERROR: TSS request failed (status=168, message=An internal error occurred.)
ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
```

Trenki, 2025-02-06:
@nikias Hello, could you help me take a look at this problem? I've been trying to solve it for several weeks without success. Thank you very much.

leana8959, 2025-08-15:
I have the same issue flashing `iPad_Pro_A12X_A12Z_18.6_22G86_Restore.ipsw` to my iPad. Are there updates to this?