libimobiledevice/idevicerestore #711: Unable to get SHSH blobs for this device

Ali-1980, 2025-02-13:
idevicerestore -u ********** iPhone12,3,iPhone12,5_18.3.1_22D72_Restore.ipsw 
idevicerestore 1.0.0-226-gbb5591d (libirecovery 1.2.1-9-g0f0928a, libtatsu 1.0.4-8-g7e1647b)
Found device in Normal mode
ECID: **********
Identified device as d421ap, iPhone12,3
Device Product Version: 18.3.1
Device Product Build: 22D72
Extracting BuildManifest from IPSW
IPSW Product Version: 18.3.1
IPSW Product Build: 22D72 Major: 22
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Checking if device requires stashbag...
Device does not require stashbag.
Getting ApNonce in normal mode... 45 78 3f 32 c1 b1 35 d0 56 c1 51 03 15 03 18 80 b7 86 75 0e 25 e9 f9 1d 92 31 7e cb da 2d 75 b7 
Getting SepNonce in normal mode... 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 91 87 a4 1c 82 85 8e f5 06 99 4f 68 ac 1b fb 9b 6e 5b 9a 63 
ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

nikias, 2025-02-14:
This looks like a connection issue to me. Can you try to run with `-dd` to see if we can see any libcurl/libtatsu error message?

Ali-1980, 2025-02-14:
Recovery mode working fine, but normal mode getting this issue 

Part 1:
------------------------------------------------------------------
opening device 05ac:1881 @ 0x02112100...
Setting to configuration 1
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Setting to interface 0:0
error opening interface: 0xe00002c5
Failed to set interface, error -8, retrying...
Failed to set interface, error -8
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'Ap,HapticAssets' as it is not trusted
DEBUG: Applying restore request rules for entry Ap,SystemVolumeCanonicalMetadata
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'AudioCodecFirmware' as it is not trusted
DEBUG: tss_request_add_ap_tags: Skipping 'BMU,DigestMap' as it doesn't have RestoreRequestRules and is not Trusted
DEBUG: tss_request_add_ap_tags: Skipping 'BMU,FirmwareMap' as it doesn't have RestoreRequestRules and is not Trusted
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=false to TSS entry
DEBUG: Adding ESEC=false to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding EPRO=false to TSS entry
[... 345 lines cut ...]
*  SSL certificate verify ok.
* using HTTP/1.1
> POST /TSS/controller?action=2 HTTP/1.1
Host: gs.apple.com
User-Agent: InetURL/1.0
Accept: */*
Cache-Control: no-cache
Content-type: text/xml; charset="utf-8"
Content-Length: 16778

< HTTP/1.1 200 
< Server: Apple
< Date: Thu, 13 Feb 2025 18:32:32 GMT
< Content-Type: text/html
< Content-Length: 46
< Connection: keep-alive
< Host: gs.apple.com
< Strict-Transport-Security: max-age=31536000; includeSubdomains
< X-Frame-Options: SAMEORIGIN
< 
* Connection #0 to host gs.apple.com left intact
TSS server returned: STATUS=168&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 168
ERROR: TSS request failed (status=168, message=An internal error occurred.)
ERROR: Unable to get SHSH blobs for this device

Ali-1980, 2025-03-09:
@nikias  

Can you help check this ? thanks

Ali-1980, 2025-03-16:
updated , fixed

closed 2025-03-16 by Ali-1980