libimobiledevice/idevicerestore #712: Restore fail AppleTV 4K 1gen. to 18.3

Slyrfecso, 2025-02-24:
feketecsongor@iPhone ~ % idevicerestore /Users/feketecsongor/Downloads/APPLE_TV/Restore_IPSWS/AppleTV6\,2_18.3_22K557_Restore_Csongor.ipsw 
Found device in DFU mode
Identified device as j105aap, AppleTV6,2
Extracting BuildManifest from IPSW
Product Version: 18.3
Product Build: 22K557 Major: 22
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 044-55199-062.dmg
[==================================================] 100.0%
Found ECID 7142608600803386
Getting ApNonce in dfu mode... fa xxx 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... xxx 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.j105a.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Sending iBSS (618683 bytes)...
[==================================================] 100.0%
Nonce: fa xxx
Extracting iBEC.j105a.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (618683 bytes)...
[==================================================] 100.0%
ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.
ERROR: Unable to place device into recovery mode from DFU mode

nikias, 2025-02-24:
Are you using the most recent code? I don't see the version being printed out by idevicerestore, so I can assume it is an older build.

Slyrfecso, 2025-02-25:
idevicerestore 1.0.0
You have right, but I don't know where can I find the new version for osx?
Thanks.

Slyrfecso, 2025-02-25:
feketecsongor@iPhone limd-build %  idevicerestore /Users/feketecsongor/Downloads/APPLE_TV/Restore_IPSWS/AppleTV6\,2_18.3_22K557_Restore_Csongor.ipsw 
idevicerestore 1.0.0-226-gbb5591d (libirecovery 1.2.1-11-g6462ea1, libtatsu 1.0.4-8-g7e1647b)
Found device in DFU mode
ECID: 7142608600803386
Identified device as j105aap, AppleTV6,2
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 18.3
IPSW Product Build: 22K557 Major: 22
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 044-55199-062.dmg
[==================================================] 100.0%
Getting ApNonce in dfu mode... a9 32 fa 00 9e 0f 22 44 20 e4 ca 5f 4f 64 6b 53 66 13 9e ce a7 03 ab 77 a5 8e f6 be 7b 9d 7d b8 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... eb 56 b9 31 dc fe 0e 7c f7 2b fc e5 22 c2 bb e3 d0 9f cc d3 
Received SHSH blobs
Extracting iBSS.j105a.RELEASE.im4p (Firmware/dfu/iBSS.j105a.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Sending iBSS (618871 bytes)...
[==================================================] 100.0%
Nonce: a9 32 fa 00 9e 0f 22 44 20 e4 ca 5f 4f 64 6b 53 66 13 9e ce a7 03 ab 77 a5 8e f6 be 7b 9d 7d b8 
Extracting iBEC.j105a.RELEASE.im4p (Firmware/dfu/iBEC.j105a.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (618871 bytes)...
[==================================================] 100.0%
INFO: device serial number is C0HXCLGUJ1WF
Recovery Mode Environment:
iBoot build-version=iBoot-11881.80.57
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo~appletv.im4p (Firmware/all_flash/applelogo~appletv.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (10877 bytes)...
Extracting AppleAVE2FW_H9.im4p (Firmware/ave/AppleAVE2FW_H9.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1199771 bytes)...
Extracting arm64SURamDisk2.dmg.trustcache (Firmware/arm64SURamDisk2.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (12190 bytes)...
ramdisk-size=0x20000000
Extracting arm64SURamDisk2.dmg (arm64SURamDisk2.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (96475136 bytes)...
Extracting DeviceTree.j105aap.im4p (Firmware/all_flash/DeviceTree.j105aap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (25361 bytes)...
Extracting sep-firmware.j105a.RELEASE.im4p (Firmware/all_flash/sep-firmware.j105a.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (782570 bytes)...
Extracting kernelcache.release.j105a (kernelcache.release.j105a)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (16547055 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
feketecsongor@iPhone limd-build %

Slyrfecso, 2025-03-01:
@nikias 
I read somewhere, that USBMUXD 1.1.2 update can solve my problem.
Can you help me please how can I install this:
[https://layers.openembedded.org/layerindex/recipe/350775/](url)

nikias, 2025-03-02:
For you it fails earlier than the stage it requires usbmuxd, and for restoring an Apple TV, even a really old usbmuxd will work just fine.
In your case it fails differently; it starts in DFU mode loads iBSS and iBEC which will bring it to Recovery mode; and for some reason it fails, and goes back to DFU mode. I have no idea why. Is the Apple TV (the attached screen) actually showing anything, at least an Apple Logo, when you try this?

Slyrfecso, 2025-03-02:
I treid to restore 13.4.8 again, maybe I see something on the screen, but nothing.
**Unfortunatelly I got this error now, but that was not before:**

```
idevicerestore 1.0.0-229-gb05d50c (libirecovery 1.2.1, libtatsu 1.0.4)
Found device in DFU mode
ECID: 7142608600803386
Identified device as j105aap, AppleTV6,2
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 13.4.8
IPSW Product Build: 17M61 Major: 17
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in dfu mode... 36 17 94 ab 03 3d 3d 4e 20 9d 44 7f b9 63 ac 32 bd 7a 18 19 27 a9 6e fd 40 7a 1d e8 7d 46 de 45 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... f0 44 d0 0d e9 86 88 c4 27 c2 3c f8 3e 82 12 59 0a 05 8f f7 
Received SHSH blobs
ERROR: Variant 'Recovery Customer Install' not found in BuildManifest
```

**That was when finished well:**
That's why I think if it worked with 13.4.8, it should work with 18.3 too.
```
Sending RestoreKernelCache （31369204 bytes）...
Waiting for device to enter restore mode...
About
to restore device...
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device ffffffffffffffffffffffffffffffff0000002b has successfully entered restore mode
Hardware Information:
BoardID: 2
ChipID: 32785
UniqueChipID: 7142608600803386
ProductionMode: true
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
Not sending NORData... Quitting...
Cleaning up...
DONE

```

Slyrfecso, 2025-03-02:
@nikias 
I see the idevicerestore version is updated to 1.0.0-229,
maybe something has been changed?

Slyrfecso, 2025-03-05:
@nikias 
I downgraded idevicerestore(1.0.0) and libirecovery[6dc8c86] to older versions and now I could restore my AppleTV6,2 (A1842) to 13.4.8.
I think the solution is in your builds for 18.3

```
feketecsongor@Fekete-iMac idevicerestore % /Users/feketecsongor/Downloads/APPLE_TV/gaster pwn
usb_timeout: 5
usb_abort_timeout_min: 0
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Stage: RESET
ret: true
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Stage: SETUP
ret: true
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Stage: SPRAY
ret: true
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Stage: PATCH
ret: true
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Now you can boot untrusted images.
feketecsongor@Fekete-iMac idevicerestore % /Users/feketecsongor/Downloads/APPLE_TV/gaster reset
usb_timeout: 5
usb_abort_timeout_min: 0
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
Found the USB handle.
feketecsongor@Fekete-iMac idevicerestore % idevicerestore /Users/feketecsongor/Downloads/APPLE_TV/Restore_IPSWS/AppleTV6\,2_13.4.8_17M61_RestoreFinal.ipsw
Found device in DFU mode
Identified device as j105aap, AppleTV6,2
Extracting BuildManifest from IPSW
Product Version: 13.4.8
Product Build: 17M61 Major: 17
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/Users/feketecsongor/Downloads/APPLE_TV/Restore_IPSWS/AppleTV6,2_13.4.8_17M61_RestoreFinal/038-50852-064.dmg'
Found ECID XXX
Getting ApNonce in dfu mode... 07 5e 3d 7d f8 49 38 be 61 33 97 16 e5 d1 ce 55 cc a6 14 5c f1 91 f6 dc 80 20 5d 9c dc 33 f5 c8 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 51 66 46 3c b4 f6 6e de 61 49 eb 93 31 17 f0 8c c3 e5 f4 98 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.j105a.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Sending iBSS (488413 bytes)...
[==================================================] 100.0%
Nonce: 07 5e 3d 7d f8 49 38 be 61 33 97 16 e5 d1 ce 55 cc a6 14 5c f1 91 f6 dc 80 20 5d 9c dc 33 f5 c8 
Extracting iBEC.j105a.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (488413 bytes)...
[==================================================] 100.0%
INFO: device serial number is XXX
Recovery Mode Environment:
iBoot build-version=iBoot-5540.140.12
iBoot build-style=RELEASE
Sending RestoreLogo...
[... 96 lines cut ...]
[==================================================] 100.0%
Requesting FUD data (36)
Found FUD component 'AOP'
Found FUD component 'RestoreTrustCache'
Found FUD component 'StaticTrustCache'
Sending FUD image list
Extracting aopfw-j105aaop.im4p...
Personalizing IMG4 component AOP...
Sending FUD data for AOP...
Extracting 038-50870-064.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending FUD data for RestoreTrustCache...
Extracting 038-50852-064.dmg.trustcache...
Personalizing IMG4 component StaticTrustCache...
Sending FUD data for StaticTrustCache...
Updating gas gauge software (47)
Updating Stockholm (55)
Fixing up /var (17)
Modifying persistent boot-args (25)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Got status message
Status: Restore Finished
```

nikias, 2025-03-06:
I just realized that you are using an ipsw that doesn't seem to be officially available. Where did you get this?

Slyrfecso, 2025-03-06:
@nikias 
Yes, AppleTV4K doesn't have official, because the device doesn't have official USB port. It has a secret lightning port inside the LAN port, if you want to connect to commputer, you need Goldeneye cable and Alex DSCS cable for DFU.
The used restore IPSW is modified and combined from official ATV 4K 1. gen. OTA and official ATV HD ipsw.
Whit this script can anybody create restore IPSW for ATV4K.
`https://github.com/verygenericname/TV4K_IPSW_Script`
It seems simple, but it's not.
Only signed version will work, at this time, 13.4.8 and 18.3 are official signed, that is why I try.

Everything I know, I learned from Zenzeq, he is also a genius, he has a post on Reddit for ATV 4K 1.gen.
He told the restore will work whit idevicerestore. Futurerestore also work but whit linux and whit the right dependencies.
I am very persistent, I don't give up, that's why I try everything.
[https://www.reddit.com/r/jailbreak/comments/1f8of4b/how_to_restore_an_apple_tv_4k_first_generation/](url)

Slyrfecso, 2025-03-07:
@nikias 
I added more time, nothing chaged
cond_wait_timeout(&client->device_event_cond, &client->device_event_mutex, 18000**0**);

Slyrfecso, 2025-03-07:
@nikias 
I haver idea, what happening, if I skip the brake:
```
	if (client->mode != MODE_RESTORE) {
		mutex_lock(&client->device_event_mutex);
		info("Waiting for device to enter restore mode...\n");
		cond_wait_timeout(&client->device_event_cond, &client->device_event_mutex, 2000000);
//		if (client->mode != MODE_RESTORE || (client->flags & FLAG_QUIT)) {
//			mutex_unlock(&client->device_event_mutex);
//			error("ERROR: Device failed to enter restore mode.\n");
//			if (client->mode == MODE_UNKNOWN) {
//				error("Make sure that usbmuxd is running.\n");
//			} else if (client->mode == MODE_RECOVERY) {
//				error("Device reconnected in recovery mode, most likely image personalization failed.\n");
//			}
//			return -1;
//		}
		mutex_unlock(&client->device_event_mutex);
	}
```

```
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (16547055 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 0019602a286ba83a (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 0019602a286ba83a (udid: N/A) connected in DFU mode
DONE
[==================================================] 100.0%
feketecsongor@Fekete-iMac idevicerestore % 
```

The idevicerestore program is not going to the next step.
Any idea why?

Slyrfecso, 2025-03-07:
@nikias 

I skipped full section above and i wanted to know in which mode is the device.
```
   if (client->mode == MODE_UNKNOWN) {
        error("MODE_UNKNOWN.\n");
    }
```
```
Sending RestoreKernelCache (16547055 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 0019602a286ba83a (udid: N/A) disconnected
MODE_UNKNOWN.
DONE
[==================================================] 100.0%
feketecsongor@Fekete-iMac idevicerestore % 
```

I don't know why I see the device is disconected, because I skipped all section!
**That's why I want to try it USBMUXD 1.1.2**
Please help me where can I find it.
Thanks.

Slyrfecso, 2025-03-14:
@nikias 
I found something with -dd mode:
I have seen that has been resolved in 2019 post, but I got it.
Can you help me please?

```
NOTE: Unable to find BbChipID node
NOTE: Unable to find BbProvisioningManifestKeyHash node
NOTE: Unable to find BbActivationManifestKeyHash node
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
NOTE: Unable to find BbSkeyId node
```

Thanks.

nikias, 2025-03-14:
> [@nikias](https://github.com/nikias) I found something with -dd mode: I have seen that has been resolved in 2019 post, but I got it. Can you help me please?
>
> ```
> NOTE: Unable to find BbChipID node [...]

The Apple TV does not have a baseband, so all these values do not exists. This is not the reason.

nikias, 2025-03-14:
So let me understand this. The old idevicerestore does just work with both 13.4.8  and 18.3 ?
The new one does work with 18.3 but not 13.4.8? Because I am a bit lost now...

In any case if it doesn't leave DFU mode (or goes from DFU to DFU), then it didn't accept one of the images provided, so something must be wrong with the image personalization.

Slyrfecso, 2025-03-14:
@nikias 
Sorry maybe I wrote a bit too much. Only 13.4.8 can restore with new idevicerestore and old too.
Can you look inside the two ipsw, maybe you'll notice something?

fragtion, 2025-08-22:
@Slyrfecso 
```
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
```
Please, how did you solve this part? Is it an issue with the IPSW, or something else? Tnx

Serg55ksv, 2026-01-09:
@hongxuanzheng
Try it   -  https://drive.google.com/file/d/1xqQFJQZayz-g7Nez-k35I9szQ8NYVEOz/view

But this file doesn't work for me....