libimobiledevice/idevicerestore #713: Restore AppleTV 4K A1842 13.4.8

mikeromanchenko, 2025-02-26:
I'm trying to restore Apple TV 4K to ipsw 13.4.8 with idevicerestore and getting ERROR: could not retrieve device serial number. Can't continue.

How should I fix it? Please help me.

Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/Users/mikeromanchenko/Desktop/ATV/AppleTV/038-50852-064.dmg'
Found ECID 625622116476986
Getting ApNonce in dfu mode... 08 14 20 5b 39 af 43 6b f0 41 ac 9f b7 fc 65 6d 7c 8f 4c 4a c0 11 14 38 07 69 f3 a5 f9 80 f0 4e 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... a7 18 3a 87 d5 90 48 3a 05 1c ce 3b e3 8c 0a 76 ca 43 36 18 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.j105a.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Sending iBSS (488413 bytes)...
[==================================================] 100.0%
Nonce: 08 14 20 5b 39 af 43 6b f0 41 ac 9f b7 fc 65 6d 7c 8f 4c 4a c0 11 14 38 07 69 f3 a5 f9 80 f0 4e 
Extracting iBEC.j105a.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (488413 bytes)...
[==================================================] 100.0%
ERROR: could not retrieve device serial number. Can't continue.

Slyrfecso, 2025-02-28:
Hi,
Please run "idevicerestore -v".
Very important which version do you use.
I restored 13.4.8 well with idevicerestore but 18.3 not.

Slyrfecso, 2025-02-28:
I reinstalled Limd-build, after I got this versions:
idevicerestore 1.0.0-226-gbb5591d (libirecovery 1.2.1-11-g6462ea1, libtatsu 1.0.4-8-g7e1647b)

Slyrfecso, 2025-02-28:
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L https://is.gd/limdmacos
bash ./limd-build-macos.sh

mikeromanchenko, 2025-02-28:
> curl -o ./limd-build-macos.sh -L https://is.gd/limdmacos

I had TLS error%

mikeromanchenko@iMac-5K-Mike-Romanchenko ~ % curl -o ./limd-build-macos.sh -L https://is.gd/limdmacos
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:15 --:--:--     0
curl: (35) TLS connect error: error:00000000:lib(0)::reason(0)

Is it problem with my connect? Or I did something wrong?

Slyrfecso, 2025-02-28:
I tried and don't got error.

```
feketecsongor@Fekete-iMac ~ % mkdir -p limd-build
feketecsongor@Fekete-iMac ~ % cd limd-build
feketecsongor@Fekete-iMac limd-build % curl -o ./limd-build-macos.sh -L https://is.gd/limdmacos
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 35896  100 35896    0     0  69377      0 --:--:-- --:--:-- --:--:--  217k
```

Maybe you have to use MacPort:
sudo port install idevicerestore

mikeromanchenko, 2025-02-28:
Successfully download script, but can't run it with bash. I have error when run it:
./limd-build-macos.sh: line 1: syntax error near unexpected token `<'

Slyrfecso, 2025-02-28:
I don't know why you got this error, but I think you have to upadte idevicerestore.
The best if you update limd-build, because that update all dependencies.
Maybe you can try MacPort or Homebrew.
Whit these you can install idevicerestore to.
If everithing works you have to see same version linke me.
After try again the restore.

I hope other people are reading this post and could help to you!

nikias, 2025-02-28:
> Successfully download script, but can't run it with bash. I have error when run it: ./limd-build-macos.sh: line 1: syntax error near unexpected token `<'

Seems like you didn't download it correctly. The script is hosted here, maybe you can access it: https://gist.github.com/nikias/84c79469a1d0f16ff95250f0d51858c3

mikeromanchenko, 2025-03-01:
Thanks, downloaded the script. But when ran it I had an error with building libimobiledevice-glue:
ERROR: build failed for libimobiledevice-glue

[libimobiledevice-glue_build.log](https://github.com/user-attachments/files/19037205/libimobiledevice-glue_build.log)

What I did wrong?

nikias, 2025-03-01:
Doesn't really look like your fault, but on my machines and through github actions it builds just fine.
What's your clang version?
`clang --version`

mikeromanchenko, 2025-03-02:
> Ok I fixed it. Run the script again, it will pull updated code and then it should compile fine!

Thank you! Now successfully install glue, but now I have error with ifuse.

[ifuse_build.log](https://github.com/user-attachments/files/19042853/ifuse_build.log)

Slyrfecso, 2025-03-02:
I don't think it matters, try the restore now.

mikeromanchenko, 2025-03-02:
> I don't think it matters, try the restore now.
> brew install ifuse

Ok, tried to restore and had error with BuildManifest:

Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in dfu mode... 32 a4 6a 48 35 33 1f 8a 1b 1e 17 ad f8 bd da 00 79 3c da a3 c7 f9 74 a7 f3 7b 31 8c 81 6c 9b 80 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... f6 8a fd 3f 3b c5 a5 e5 fa d2 1a 8a 72 4a 66 7d 06 0a c2 86 
Received SHSH blobs
ERROR: Variant 'Recovery Customer Install' not found in BuildManifest

How to fix it?

Slyrfecso, 2025-03-02:
:))))
I have now the same problem.
That was not before.
I asked @nikias I hope hi will help us.

mikeromanchenko, 2025-03-02:
May be we can use previous versions? But I don't know how to choose it and install.

Slyrfecso, 2025-03-02:
Yes, I find this in other post:
[(https://github.com/UnbendableStraw/homepod-restore/issues/8#issuecomment-2519323812)]

Slyrfecso, 2025-03-06:
@nikias 
So I modified easly the IPSW, @mikeromanchenko see in the above post, how to.
I restored well with the new idevicerestore the Apple TV 4K 1842 to TVOS 13.4.8

```
feketecsongor@Fekete-iMac ~ % /Users/feketecsongor/Downloads/APPLE_TV/gaster pwn
usb_timeout: 5
usb_abort_timeout_min: 0
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Stage: RESET
ret: true
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Stage: SETUP
ret: true
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Stage: SPRAY
ret: true
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Stage: PATCH
ret: true
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
CPID: 0x8011
Found the USB handle.
Now you can boot untrusted images.
feketecsongor@Fekete-iMac ~ % /Users/feketecsongor/Downloads/APPLE_TV/gaster reset
usb_timeout: 5
usb_abort_timeout_min: 0
[IOKit] Waiting for the USB handle with VID: 0x5AC, PID: 0x1227
Found the USB handle.
feketecsongor@Fekete-iMac ~ % idevicerestore /Users/feketecsongor/Downloads/APPLE_TV/ipsw-file-extraction/AppleTV6.2_13.4.8_17M61_RestoreFinal_Csongor.ipsw
idevicerestore 1.0.0-229-gb05d50c (libirecovery 1.2.1-11-g6462ea1, libtatsu 1.0.4)
Found device in DFU mode
ECID: xxx
Identified device as j105aap, AppleTV6,2
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 13.4.8
IPSW Product Build: 17M61 Major: 17
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-50852-064.dmg
[==================================================] 100.0%
Getting ApNonce in dfu mode... fa 4b 02 9e 10 5e 47 41 fa 02 68 c2 55 cd bd 3c b6 cb 14 42 b4 2c 24 11 eb 5a 5a fc c9 20 e7 aa 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 4a 38 e2 70 1a dc a4 55 7e c8 63 f6 91 9d 14 7b 4a 3f 31 69 
Received SHSH blobs
Extracting iBSS.j105a.RELEASE.im4p (Firmware/dfu/iBSS.j105a.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Sending iBSS (488413 bytes)...
[==================================================] 100.0%
Nonce: fa 4b 02 9e 10 5e 47 41 fa 02 68 c2 55 cd bd 3c b6 cb 14 42 b4 2c 24 11 eb 5a 5a fc c9 20 e7 aa 
Extracting iBEC.j105a.RELEASE.im4p (Firmware/dfu/iBEC.j105a.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (488413 bytes)...
[==================================================] 100.0%
INFO: device serial number is XXX
Recovery Mode Environment:
iBoot build-version=iBoot-5540.140.12
[... 275 lines cut ...]
Checkpoint completed id: 0x67B (unknown) result=0
Checkpoint started   id: 0x1600 (unknown)
Checkpoint completed id: 0x1600 (unknown) result=0
Checkpoint started   id: 0x63F (unknown)
Checkpoint completed id: 0x63F (unknown) result=0
Checkpoint started   id: 0x640 (unknown)
Checkpoint completed id: 0x640 (unknown) result=0
Checkpoint started   id: 0x669 (unknown)
Checkpoint completed id: 0x669 (unknown) result=0
Checkpoint started   id: 0x642 (unknown)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checkpoint completed id: 0x642 (unknown) result=0
Checkpoint started   id: 0x67C (unknown)
Checkpoint completed id: 0x67C (unknown) result=0
Checkpoint started   id: 0x1614 (unknown)
Checkpoint completed id: 0x1614 (unknown) result=0
Checkpoint started   id: 0x647 (unknown)
Checkpoint completed id: 0x647 (unknown) result=0
Checkpoint started   id: 0x648 (unknown)
Got status message
Status: Restore Finished
ReverseProxy[Ctrl]: (status=2) Terminated
```

mikeromanchenko, 2025-03-07:
@Slyrfecso Could you give me link to download modified 13.4.8 ipsw? I can unzip it and change BuildManifest, but don't know how to zip it to ipsw.

Slyrfecso, 2025-03-07:
@mikeromanchenko 
Very simple: go to finder and select all files and folders after right click and choose compress.
When finished you will find xy.zip file next to them. Rename it to ipsw and use it restore.

mikeromanchenko, 2025-03-07:
@Slyrfecso Thanks! I created root ipsw folder instead of all folder and files. Now it's ok, but I have new error: Device failed to enter restore mode.

How to fix it? You had the same error when tried to restore to 18.3. But I use 13.4.8.

mikeromanchenko@iMac-5K-Mike-Romanchenko ~ % idevicerestore -e /Users/mikeromanchenko/Desktop/ATV/AppleTV6\,2_13.4.8_17M61_Restore.ipsw 
idevicerestore 1.0.0-229-gb05d50c (libirecovery 1.2.1-12-g4a7e52f, libtatsu 1.0.4-8-g7e1647b)
Found device in DFU mode
ECID: 625622116476986
Identified device as j105aap, AppleTV6,2
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 13.4.8
IPSW Product Build: 17M61 Major: 17
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-50852-064.dmg
[==================================================] 100.0%
Getting ApNonce in dfu mode... cb 52 0e a7 3e ab 6f 1b 26 26 bd ac 52 e7 37 31 d6 c6 44 65 f2 4e 56 f9 38 86 e5 e7 1f 4e a3 17 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 81 47 ab 73 2e 92 42 0f 94 96 cc dc e6 e5 5c 5a bc 66 0b 41 
Received SHSH blobs
Extracting iBSS.j105a.RELEASE.im4p (Firmware/dfu/iBSS.j105a.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Sending iBSS (488413 bytes)...
[==================================================] 100.0%
Nonce: cb 52 0e a7 3e ab 6f 1b 26 26 bd ac 52 e7 37 31 d6 c6 44 65 f2 4e 56 f9 38 86 e5 e7 1f 4e a3 17 
Extracting iBEC.j105a.RELEASE.im4p (Firmware/dfu/iBEC.j105a.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (488413 bytes)...
[==================================================] 100.0%
Recovery Mode Environment:
iBoot build-version=iBoot-5540.140.12
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo~appletv.im4p (Firmware/all_flash/applelogo~appletv.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (10259 bytes)...
Extracting 038-50782-064.dmg.trustcache (Firmware/038-50782-064.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (9014 bytes)...
ramdisk-size=0x20000000
Extracting 038-50782-064.dmg (038-50782-064.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (72816016 bytes)...
Extracting RestoreDeviceTree.j105aap.im4p (Firmware/all_flash/RestoreDeviceTree.j105aap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (111974 bytes)...
Extracting sep-firmware.j105a.RELEASE.im4p (Firmware/all_flash/sep-firmware.j105a.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (2000462 bytes)...
Extracting kernelcache.release.j105a (kernelcache.release.j105a)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (11897160 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
mikeromanchenko@iMac-5K-Mike-Romanchenko ~ %

mikeromanchenko, 2025-03-07:
Actually I think that reason is hardware ATV. If I connect it only with power cable (without Goldeneye and DCSD cables) it goes into DFU mode in 5-10 seconds by itself.  Maybe because of this it can't go into recovery mode :(

Slyrfecso, 2025-03-07:
@mikeromanchenko 
Can show me the version of:
`usbmuxd -V`
(usbmuxd 1.1.1-69-g523f700)

AlexDCSD cable version:
APN-591-0567-06-H
Date of manufacture: 05/August/2021

libirecovery 1.2.1-12-g4a7e52f (this is yours)
libirecovery 1.2.1-11-g6462ea1 (this is my, I will try to restore whit yours, and I will check maybe that is the problem.)

mikeromanchenko, 2025-03-07:
> usbmuxd -V

usbmuxd 1.1.1 

without 69-g523f700

DSCD cable is the same

I know that is DCSD is necessary for DFU mode, but also goes itself without it. I understood it this way: it lights slowly and after 5-10 sec it lights rapidly, and gaster pwn works whiteout cables.

Slyrfecso, 2025-03-07:
@mikeromanchenko 
I tried with the new libirecovery 1.2.1-12-g4a7e52f and the restore worked well with 13.4.8
I couldn't update libtatsu to 1.0.4-8-g7e1647b because my is from brew installed,
when I removed and installed from git, that not linked.
I try to solve it tomorrow.
Very intersting your video, sure you need the cables, when you try restoring.

KingberLiu, 2025-05-18:
> 您好，请运行 “idevicerestore -v”。使用哪个版本非常重要。我使用 idevicerestore 很好地恢复了 13.4.8，但 18.3 还没有。

Hello, can you give me a copy of the ipsw file of 13.4.8 for Apple tv 4k? Mine keeps building failing.

Slyrfecso, 2025-05-19:
Hi, send me a mail and send you the file.
info@csongorfekete.com

HelloThere295, 2025-07-09:
Hello, I try to restore my ATV 4K with idevicerestore 1.0.0-241-g038a493 (libirecovery 1.2.1-13-g638056a, libtatsu 1.0.5). Also edited the manifest file to skip the error of "Variant 'Recovery Customer Install' not found in BuildManifest" but finally run into the error of below,
**Could not read data (-256). Aborting.
Unable to restore device**

How to fix this? Thanks for any help here. Below the full output on my screen,

vlou@Vs-MacBook-Pro ~ % idevicerestore -e /Users/vlou/Downloads/AppleTV6\,2_13.4.8_17M61_Restore

idevicerestore 1.0.0-241-g038a493 (libirecovery 1.2.1-13-g638056a, libtatsu 1.0.5)
Found device in DFU mode
ECID: 955501511944250
Identified device as j105aap, AppleTV6,2
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 13.4.8
IPSW Product Build: 17M61 Major: 17
Uploading [                                                  ]   0.0% Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
##################################[ WARNING ]#################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, cancel now and restart without -e or --erase        #
# command line switch.                                                       #
# Type YES and press ENTER to continue, or hit CTRL+C to cancel.             #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce Getting ApNonce in DFU mode... 81 be cf b7 f0 c7 89 97 81 0c 81 24 71 50 26 aa 73 58 83 40 b3 89 ef 98 50 63 f9 80 5b 84 1f 6e
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 62 5e 94 a0 1a 50 48 6c d5 2b 62 87 f4 0e 62 62 27 b1 ec da
Received SHSH blobs
Uploading [                                                  ]   0.0% Extracting iBSS.j105a.RELEASE.im4p (Firmware/dfu/iBSS.j105a.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Sending iBSS (488413 bytes)...
Uploading [==================================================] 100.0% 
Uploading [                                                  ]   0.0% Nonce: 81 be cf b7 f0 c7 89 97 81 0c 81 24 71 50 26 aa 73 58 83 40 b3 89 ef 98 50 63 f9 80 5b 84 1f 6e
Extracting iBEC.j105a.RELEASE.im4p (Firmware/dfu/iBEC.j105a.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (488413 bytes)...
Uploading [==================================================] 100.0% 
INFO: device serial number is C07WT9MNJ1WF
Uploading [                                                  ]   0.0% Recovery Mode Environment:
iBoot build-version=iBoot-5540.140.12
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo~appletv.im4p (Firmware/all_flash/applelogo~appletv.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (10259 bytes)...
Uploading [==================================================] 100.0% 
Extracting 038-50782-064.dmg.trustcache (Firmware/038-50782-064.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (9014 bytes)...
ramdisk-size=0x20000000
Extracting 038-50782-064.dmg (038-50782-064.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (72816016 bytes)...
Extracting RestoreDeviceTree.j105aap.im4p (Firmware/all_flash/RestoreDeviceTree.j105aap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (111974 bytes)...
Extracting sep-firmware.j105a.RELEASE.im4p (Firmware/all_flash/sep-firmware.j105a.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (2000462 bytes)...
Extracting kernelcache.release.j105a (kernelcache.release.j105a)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (11897160 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device ffffffffffffffffffffffffffffffff00000002 has successfully entered restore mode
Hardware Information:
BoardID: 2
ChipID: 32785
UniqueChipID: 955501511944250
ProductionMode: true
Previous restore exit status: 256
Starting Reverse Proxy
ReverseProxy[Ctrl]: (status=1) Ready
Device will not use nonce slots.
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
ReverseProxy[Ctrl]: (status=2) Terminated
Could not read data (-256). Aborting.
Unable to restore device

Slyrfecso, 2025-08-24:
@fragtion @HelloThere295 @KingberLiu 
Hi Dimitri,
I think your IPSW file is not combined well, therefor you got the image personalization failed.
Sometime better if you use other PC or MAC, because one dependencie isn't installed well or linking aren't correct.
Maybe you can try container of idevicerestore, that install every dependencies well.

**I see you tried many thinks and worked lot of on that, therefore I share whit as my IPSW.**
[https://drive.google.com/file/d/1xqQFJQZayz-g7Nez-k35I9szQ8NYVEOz/view?usp=sharing](url)

MY OPINION
Half year was very intresting for me from 0 to restoring this device.
I learned lot of from hardware and softwares to, but my device was defeat, because the hdmi output isn't worked.
I learned linux, signing method, IPSW creation, hardware soldering.
I restored well and over IP and XCode tried many thinks.

Now I sold everything, device and cables too and I collecting money to new one:)

huozong88-dev, 2026-04-15:
> Hi,
> Please run "idevicerestore -v".
> Very important which version do you use.
> I restored 13.4.8 well with idevicerestore but 18.3 not yet.

Can you share the ipsw file of 13.4.8 for recovery? Thank you very much huozong88@ gimal.com