libimobiledevice/idevicerestore #715: Error in dfu.c: error: call to undeclared function 'irecv_init'

DerekBraeden, 2025-03-02:
Hello,

I'm trying to get idevicerestore to build, but Terminal gives the following error in idevicerestore-dfu.c regarding making a call to an undeclared function (irecv_init)): "ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]"

```
/usr/local/Cellar/libzip/1.11.3/include/zip.h:72:42: note: expanded from macro 'ZIP_DEPRECATED'
   72 | #define ZIP_DEPRECATED(x) __attribute__((deprecated(x)))
      |                                          ^
2 warnings generated.
  CC       idevicerestore-normal.o
  CC       idevicerestore-dfu.o
dfu.c:87:2: error: call to undeclared function 'irecv_init'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
   87 |         irecv_init();
      |         ^
1 error generated.
make[2]: *** [idevicerestore-dfu.o] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

As a result, build of idevicerestore does not complete and I obviously cannot install it.

Brief system info (let me know if you need anything else): 

- Homebrew 4.4.22
- Mac Pro 2019
- Intel i9
- MacOS Sequoia 15.3.1
- Installed everything to the default directories for MacOS Intel.

nikias, 2025-03-02:
You are not trying to build the latest code. On macOS, try this script:
```
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L https://is.gd/limdmacos
bash ./limd-build-macos.sh
```

DerekBraeden, 2025-03-02:
Thank you. I had tried that route as well; macFUSE throws the following error during installation: 

`installer: Error - The macFUSE installation package is not compatible with this version of macOS`

Again, I'm running MacOS Sequoia 15.3.1

DerekBraeden, 2025-03-02:
ifuse build fails and throws the following errors (log attached) 

[ifuse_build.log](https://github.com/user-attachments/files/19045256/ifuse_build.log)

Recovery Mode was entered and reduced security enabled, as well

nikias, 2025-03-03:
ifuse is not a required component, but optional. I think I will remove it from the build script since it is not obvious enough that it is not required...

DerekBraeden, 2025-03-03:
Thank you. Please let me know when the build script has been updated to remove ifuse

Slyrfecso, 2025-03-03:
@DerekBraeden 
You can use the script without ifuse if you got the error at the end of install, try to write in the Terminal:
idevicerestore

DerekBraeden, 2025-03-03:
Perfect, thank you!

closed 2025-03-03 by DerekBraeden