libimobiledevice/idevicerestore #716: Restore fails on iPhone 16e

skortela, 2025-03-05:
idevicerestore fails with iPhone 16e.
Lastest idevicerestore with https://github.com/libimobiledevice/libirecovery/pull/135

`[13:12:58.146][8272] : idevicerestore 1.0.1 (libirecovery 1.2.1, libtatsu 1.0.4)
[13:12:58.149][8568] : [libusbmuxd] device_monitor: device watcher started
[13:12:58.149][8568] : [libusbmuxd] device_monitor: connecting to usbmuxd...
[13:12:58.150][8568] : [libusbmuxd] device_monitor: connected to usbmuxd
[13:12:58.150][8568] : [libusbmuxd] get_next_event: got MESSAGE_DEVICE_ADD
[13:12:58.194][8568] : idevice_event_cb: device 000115480aa2801c (udid: 00008140-000115480AA2801C) connected in normal mode
[13:12:58.194][8272] : Found device in Normal mode
[13:12:58.348][8272] : ECID: 304874136961052
[13:12:58.348][8272] : Identified device as v59ap, iPhone17,5
[13:12:58.410][8272] : Device Product Version: 18.3
[13:12:58.410][8272] : Device Product Build: 22D8063
[13:12:58.410][8272] : Extracting BuildManifest from IPSW
[13:12:58.430][8272] : IPSW Product Version: 18.3.1
[13:12:58.430][8272] : IPSW Product Build: 22D8075 Major: 22
[13:12:58.472][8272] : Device supports Image4: true
[13:12:58.473][8272] : Variant: Customer Erase Install (IPSW)
[13:12:58.473][8272] : This restore will erase all device data.
[13:12:58.473][8272] : Checking IPSW for required components...
[13:12:58.501][8272] : All required components found in IPSW
[13:12:58.501][8272] : Getting ApNonce in normal mode... 0d f7 d2 55 f8 14 1a 93 4b f5 7a 73 86 b1 47 94 c0 1d 2e 33 58 35 38 f1 60 3e 1c 55 55 7b 1f fb 
[13:12:58.753][8272] : Getting SepNonce in normal mode... [13:12:58.753][8272] : fd [13:12:58.753][8272] : f5 [13:12:58.753][8272] : ca [13:12:58.753][8272] : f4 [13:12:58.753][8272] : 78 [13:12:58.753][8272] : 31 [13:12:58.753][8272] : 4d [13:12:58.753][8272] : 35 [13:12:58.753][8272] : d1 [13:12:58.753][8272] : 80 [13:12:58.753][8272] : 07 [13:12:58.753][8272] : 82 [13:12:58.753][8272] : 5d [13:12:58.753][8272] : db [13:12:58.753][8272] : 68 [13:12:58.753][8272] : 82 [13:12:58.753][8272] : ad [13:12:58.753][8272] : 03 [13:12:58.754][8272] : a7 [13:12:58.754][8272] : ac [13:12:58.754][8272] : 
[13:12:58.754][8272] : Trying to fetch new SHSH blob
[13:13:01.655][8272] : ERROR: Unable to send TSS request
ERROR: Unable to find required BbSNUM in parameters
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
[13:13:01.657][8272] : ERROR: Unable to get SHSH blobs for this device`

Ali-1980, 2025-03-09:
status=94 , becoz firmware stop sign, use the new one

iFred09, 2025-03-11:
> status=94 , becoz firmware stop sign, use the new one

no, he try to restore to iOS 18.3.1, which is the latest version.

nirhen, 2025-03-12:
hi - i also have this issue 

 ERROR: Unable to find required BbSNUM in parameters
Request URL set to https://gs.apple.com/TSS/controller?action=2
) Sending TSS request attempt 1... TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
ERROR: Unable to get SHSH blobs for this device

using the new  18.3.2 from today did not help - same error 
i've been told that flashing from recovery mode does work fine

nirhen, 2025-03-23:
@iFred09 - say 
have you worked arround this issue - or fixed it by any chance ? 
for now i do - normal to recovery - and than flash works

nikias, 2025-03-23:
Hi, you can try to comment out these lines:

https://github.com/libimobiledevice/idevicerestore/blob/2ee5021f36a26f7344b63d3321cda78899edc319/src/idevicerestore.c#L2410-2421

nirhen, 2025-03-24:
@nikias - worked like magic :)
thanks !!

nikias, 2025-03-24:
Great. I will detect the Apple baseband and skip this part for these new style devices.

nikias, 2025-03-27:
@nirhen you don't happen to have - or are able to generate - a full log output of the restore process (preferrably with `-d` command line switch)?

nikias, 2025-03-27:
Other than that I pushed commit a5905b7 that will make it work for the iPhone 16e.