libimobiledevice/idevicerestore #720: Got an error

vitorcrdias, 2025-03-17:
Restore doesn't work with: iTunes, 3uTools, etc. 
Model Iphone: XR
iPhone 11,8 (whorever it means).
CHIP ID: 8020
iOS version: 111881.80.57

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>@Locality</key>
        <string>en_US</string>
        <key>@HostPlatformInfo</key>
        <string>windows</string>
        <key>@VersionInfo</key>
        <string>libauthinstall-293.1.16</string>
        <key>@UUID</key>
        <string>6038BCFF-83E8-424D-80FD-3C152410E7FF</string>
        <key>ApECID</key>
        <integer>2934889246097466</integer>
        <key>UniqueBuildID</key>
        <data>
        0oUk7rpXuPDa5T1HvCZuPbdnNDM=
        </data>
        <key>ApChipID</key>
        <integer>32800</integer>
        <key>ApBoardID</key>
        <integer>12</integer>
        <key>ApSecurityDomain</key>
        <integer>1</integer>
        <key>ANE</key>
        <dict>
                <key>Digest</key>
                <data>
                DsLnOcXoqnc0kDUjn6NEdGYJKMNAPhTU0QE1QdvrJojl+PTThk7DdUl9yGqe
                5Gtm
                </data>
                <key>Trusted</key>
                <true/>
                <key>EPRO</key>
                <true/>
                <key>ESEC</key>
                <true/>
        </dict>
        <key>AOP</key>
        <dict>
                <key>Digest</key>
                <data>
                NcLP1hSRFEsBMV8iy3qY9MFbIA2KWULjlTTs01qIe0TUNA2ACDmYe22ZjJB8
                wsak
                </data>
                <key>Trusted</key>
                <true/>
                <key>EPRO</key>
                <true/>
                <key>ESEC</key>
                <true/>
        </dict>
        <key>AVE</key>
        <dict>
                <key>Digest</key>
                <data>
                2NeNJHFYDqqDKeqBcpYviwMz609tZqHXjo/imuZIGYMTUxsZbZRyib5SJKf/
                ADr8
                </data>
                <key>Trusted</key>
                <true/>
                <key>EPRO</key>
                <true/>
                <key>ESEC</key>
[... 843 lines cut ...]
                </data>
        </dict>
        <key>ApNonce</key>
        <data>
        wok9aOpzH1aVodZnVOhs1m5agJmyuKraC1mncFvg/i8=
        </data>
        <key>@ApImg4Ticket</key>
        <true/>
        <key>ApSecurityMode</key>
        <true/>
        <key>ApProductionMode</key>
        <true/>
        <key>SepNonce</key>
        <data>
        IPP8r3p+1iWGVQCNPQmeX5N0JsQ=
        </data>
</dict>
</plist>
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

Thank all of you guys for any help!

vitorcrdias, 2025-03-17:
I tryed to let xml code structured here, but I'm newbie in GitHub lol

nirhen, 2025-03-18:
@vitorcrdias Hi
I'm having similar issue 
What is the IPSW you are trying to restore ? 
What is the current version installed on the device ?
if you try to flash from recovery mode (e.g - not in normal)- does the issue reproduce ?