libimobiledevice/idevicerestore #725: Device failed to enter restore mode

omya3qno, 2025-03-30:
Hoping to revive a borked iPhone 14.

```
$ idevicerestore --version
idevicerestore 1.0.1

$ idevicerestore -R -dd iPhone14,7_18.3.2_22D82_Restore.ipsw
....
Sending RestoreKernelCache (18044678 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device XYZ (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device XYZ (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
```

Trying the restore from an iMac, I get a mix of errors 21, 2006 and 4013.

Is there anything left to try before I get this phone back to the store?

Danny-Dunn, 2025-04-05:
Hello! I think this is a very similar issue to mine in #724. I'm getting errors 2006 and 4012 when I try to restore from a Mac. Would love to know if you find anything out.

Abubakar-Waheed, 2025-07-26:
Did anyone got it fixed? I got same issue