libimobiledevice/idevicerestore #727: ios18.4 restore fail

irainbw, 2025-04-02:
When I use iPhone 11 to flash 18.4, an error occurs

Here is some information:

  
< HTTP/1.1 200 OK
< x-amz-id-2: z7XrpOYa7pl5CBMvS+zKxM21AJhsDPwMwhFzUW7lo18g4L6j223vrf4eYReVfNqa2qohPgT/zMRd3o7MOH7dnw==
< x-amz-request-id: KFJZ8PCF6RS3RP0D
< Date: Wed, 02 Apr 2025 06:35:23 GMT
< Last-Modified: Wed, 26 Mar 2025 16:01:30 GMT
< x-amz-server-side-encryption: AES256
< x-amz-version-id: 0asM_TNqRN.m2tRFl2eK2s77jmINlLcs
< Accept-Ranges: bytes
< Content-Type: binary/octet-stream
< Content-Length: 241
< Server: AmazonS3
< Cache-Control: max-age=600, public
< Via: http/1.1 usatl4-edge-mx-020.ts.apple.com (acdn/268.14469), http/1.1 usatl4-edge-fx-006.ts.apple.com (acdn/268.14469)
< X-Cache: hit-fresh, hit-fresh
< CDNUUID: 447cfe15-ca89-4333-9a7e-59d70b8c4c47-16139479003
< Strict-Transport-Security: max-age=31536000; includeSubdomains
< Etag: "a7d89656c5c01f23ca0870d9a7e815cc"
< Age: 176
< Connection: keep-alive
<
* Connection #1 to host fcs-keys-pub-prod.cdn-apple.com left intact
Connecting to URLAsset data port 49160
ERROR: Failed to start service connection for URLAsset on port 49160
ERROR: restore_send_url_asset: Unable to connect to service client
ERROR: Unable to send URLAsset data
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
No data to read
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to send data to ASR. Sent 0 of 131072 bytes.
Unable to send filesystem payload chunk, retrying...
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
ERROR: Unable to restore device

full info:

[iphone11_18.4.txt](https://github.com/user-attachments/files/19562511/iphone11_18.4.txt)

nirhen, 2025-04-07:
hi - i  also have the same issue with iphone 15 pro - 
ipsw 18.4 does not flash from normal mode - logs showing the same thing 
old ipsw shows no issue 

```
Done sending NORData
Tag found
Tag found
0
ReverseProxy[Ctrl]: Connection closed
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to restore device
```

oddly enough - if you start from restore mode the device flashes with success

nirhen, 2025-04-08:
BTW - 
issue - https://github.com/libimobiledevice/idevicerestore/issues/721
is a duplicate of this one

zakaryaboukherroub-crypto, 2025-12-25:
i want to by pass icloud on an iphone 11.
Someone know how to do?