libimobiledevice/idevicerestore #729: Restore failed with macOS IPSW greater than 18.4 on Apple silicon Mac (M1)

raccoon2k, 2025-04-17:
When using idevicerestore to restore my Macbook Air M1 with a macOS build version greater than 18.4 (18.4 and 18.4.1), after sending restoreKernelCache, the MacBook restarts into DFU again, therefore the restore process fails.

If using any macOS IPSW before 18.4, the restore process  proceeds normally

How should i debug this problem ?

Ubuntu 24.04.2 
idevicerestore 1.0.0-231-ga5905b7 (libirecovery 1.2.1-13-g638056a, libtatsu 1.0.4-8-g7e1647b)
usbmuxd 1.1.1-69-g523f700

############# 
...
Sending RestoreSEP (5681746 bytes)...
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.mac13g (kernelcache.release.mac13g)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (29455927 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 000e25ec3001e (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 000e25ec3001e (udid: N/A) connected in DFU mode
ERROR: Device failed to enter restore mode.

kFYatek, 2025-04-24:
The macOS version is actually 15.4, but I can confirm that it happens for me as well. macOS 15.3.2 is the newest version I can restore.

raccoon2k, 2025-04-25:
> The macOS version is actually 15.4, but I can confirm that it happens for me as well. macOS 15.3.2 is the newest version I can restore.
Yeah, I somehow mixed up the macOS version with the iOS version—sorry about that.

raccoon2k, 2025-04-25:
> The macOS version is actually 15.4, but I can confirm that it happens for me as well. macOS 15.3.2 is the newest version I can restore.<

Yeah, I somehow mixed up the macOS version with the iOS version—sorry about that.

zer010gic, 2025-07-22:
I am also experiencing this issue for M1.  Restores on M4 appear to function, though missing the restore progress screen.

zer010gic, 2025-07-23:
I have tested on an M2 Mini, and it works for the M2.  So it may be isolated to the M1.  I will see if I can do other tests as well.

zer010gic, 2025-07-27:
> > I have tested on an M2 Mini, and it works for the M2. So it may be isolated to the M1. I will see if I can do other tests as well.
> 
> What worked for you can guide please I'm stuck on failed to enter restore mode issue
> 
> [#746](https://github.com/libimobiledevice/idevicerestore/issues/746)

Not sure what I am doing differently, I am running it in Arch Linux off a Raspberry Pi.  I have used both the AUR install for idevicerestore, as well as compiling versions.

Here are the versions I am running.
idevicerestore 1.0.0-232-g8061f08-dirty (libirecovery 1.2.1-13-g638056a, libtatsu 1.0.5)
usbmuxd 1.1.1-69-g523f700

raccoon2k, 2025-09-13:
> Please re-test with latest code, I pushed a commit that hopefully fixes this.

The error was fixed after updating to the latest commit (Tested using ipsw 15.6.1 on my M1 Mac). Thank you ~

closed 2025-09-13 by raccoon2k