libimobiledevice/idevicerestore #732: Boot into Recovery or DFU mode without Resetting iPad

jbrooks2-KCPS, 2025-04-30:
Not really an issue but more of a question/feature request. Is there a way to just make idevicerestore boot the device into recovery or dfu mode and not restore the device? I don't know if this is possible but for my needs it would be pretty nice to have.

nikias, 2025-04-30:
Well, there is `ideviceenterrecovery` which puts a device that is in normal mode into recovery mode, but entering DFU mode automatically requires special hardware. So this is not something we can add to idevicerestore.

jbrooks2-KCPS, 2025-04-30:
Oh thank you!! I totally didn't know about that

im getting this when i run that though, it doesn't put the device into recovery mode sadly - 

Putting device with UDID 00008030-000A59322650A02E into recovery mode...
Telling device with udid 00008030-000A59322650A02E to enter recovery mode.
Failed to enter recovery mode.
Device is successfully switching to recovery mode.
Device 00008030-000A59322650A02E successfully entered recovery mode.