libimobiledevice/idevicerestore #733: M4 Mac Mini

ZachBacon, 2025-05-16:
I'm running into an issue where I'm trying to restore an m4 mac mini in DFU mode, but I'm getting ```ERROR: Unable to discover device type```

when I use my git head compiles of the software/libraries and then running this command

```idevicerestore -P -d -ey ~/Downloads/UniversalMac_15.5_24F74_Restore.ipsw```

I'm currently at a loss of what I need to do since I don't have another mac computer to do an dfu restore.

ledlamp, 2025-06-14:
The only way is to set up a macOS virtual machine. Ugh

rascalldev, 2025-06-17:
I ran into this issue recently due to a failed 3rd party SSD. Managed to get my m4 mini restored by using this compilation workflow [BarbossHack/libimobiledevice-fedora](https://github.com/BarbossHack/libimobiledevice-fedora) on Fedora 42.

Tried Sonoma on VMWare (Win & Linux), Windows iTunes, and an old Intel Mac on Catalina, which all failed to restore the M4 mini. This is the only method that worked for me without using an additional Apple Silicon mac.

falchion10, 2025-06-17:
idevicerestore is currently broken for 15.4+ restores. if you want to DFU restore you'd need to do it to 15.3.2 and then OTA update to latest. https://github.com/libimobiledevice/idevicerestore/issues/729

a13547, 2025-06-23:
> > I'm writing this comment on an m4 mini I restored using:
> > ```
> > * idevicerestore 1.0.0-git-e1f5865 (libirecovery 1.2.1-13-g638056a, libtatsu 1.0.5)
> > 
> > * UniversalMac_15.5_24F74_Restore.ipsw
> > ```
> 
> How exactly did you do that, could you write the steps out for me?

Nothing special. Just compile the whole setup (libimobiledevice idevicerestore usbmuxd… etc)from source code and done. I can even flash macOS 26 ipsw. Maybe the key is not to use any component from distros  .

nikias, 2025-09-06:
Please re-test with latest code, I pushed a commit that hopefully fixes this.

adftrew345, 2025-09-10:
> Please re-test with latest code, I pushed a commit that hopefully fixes this.

Hi Nik, can confirm (Macbook Air M4,UniversalMac_15.6_24G84_Restore) is working 

had the same error " ERROR: Unable to discover device type" till i fixed usbmuxd. 

Also i compiled the sourcecode, couldnt revive my macbook tho, had to use the -erase option. 

Dont know how i got here in the first place, i tried to uninstall a audio interface driver using onyx, after freezing up all my apps where gone and after turning in off and on again i got stuck in a bootloop 

eh thank you so much for developing this tool, i just opted for a macbook for music production reasons, never thought it would be such a pain in the ass. hopefully the eu is gonna force apple to let one restore its own macbook with a usb stick or a windows pc

ledlamp, 2025-09-14:
> Nothing special. Just compile the whole setup (libimobiledevice idevicerestore usbmuxd… etc)from source code and done. I can even flash macOS 26 ipsw. Maybe the key is not to use any component from [...]

I did all this but how come it still using old versions??

idevicerestore 1.0.0-254-g8d1e798 (libirecovery 1.2.1, libtatsu 1.0.3)