libimobiledevice/idevicerestore #734: Restore Error 'Failed to install splat' on iPhone 13 Pro

yasavvym8, 2025-05-25:
Hello all,

I am having some difficulty with a successful firmware update on my iPhone 13 Pro. Initially, I attempted to update iOS 18.3.2 to iOS 18.5 via itunes, but then my phone never finished the update. Inevitably it got stuck in recovery mode. Now when trying to use idevicerestore to flash iPhone14,2_18.5_22F76_Restore.ipsw, I get the following output towards the end of the update. Does anyone have any insight on this? 

ramrod_display_set_granular_progress_forced: 95.000000
ASR VERIFY PROGRESS: 80%
ramrod_display_set_granular_progress_forced: 95.000000
ASR VERIFY PROGRESS: 81%
ramrod_display_set_granular_progress_forced: 96.000000
ASR VERIFY PROGRESS: 82%
ramrod_display_set_granular_progress_forced: 96.000000
ASR VERIFY PROGRESS: 83%
ramrod_display_set_granular_progress_forced: 96.000000
ASR VERIFY PROGRESS: 84%
ramrod_display_set_granular_progress_forced: 96.000000
ASR VERIFY PROGRESS: 85%
ramrod_display_set_granular_progress_forced: 96.000000
ASR VERIFY PROGRESS: 86%
ramrod_display_set_granular_progress_forced: 97.000000
ASR VERIFY PROGRESS: 87%
ramrod_display_set_granular_progress_forced: 97.000000
ASR VERIFY PROGRESS: 88%
ramrod_display_set_granular_progress_forced: 97.000000
ASR VERIFY PROGRESS: 89%
ramrod_display_set_granular_progress_forced: 97.000000
ASR VERIFY PROGRESS: 90%
ramrod_display_set_granular_progress_forced: 97.000000
ASR VERIFY PROGRESS: 91%
ramrod_display_set_granular_progress_forced: 98.000000
ASR VERIFY PROGRESS: 92%
ramrod_display_set_granular_progress_forced: 98.000000
ASR VERIFY PROGRESS: 93%
ramrod_display_set_granular_progress_forced: 98.000000
ASR VERIFY PROGRESS: 94%
ramrod_display_set_granular_progress_forced: 98.000000
ASR VERIFY PROGRESS: 95%
ramrod_display_set_granular_progress_forced: 98.000000
ASR VERIFY PROGRESS: 96%
ramrod_display_set_granular_progress_forced: 99.000000
ASR VERIFY PROGRESS: 97%
ramrod_display_set_granular_progress_forced: 99.000000
ASR VERIFY PROGRESS: 98%
ramrod_display_set_granular_progress_forced: 99.000000
ASR VERIFY PROGRESS: 99%
ramrod_display_set_granular_progress_forced: 99.000000
ASR VERIFY PROGRESS: 100%
ramrod_display_set_granular_progress_forced: 100.000000
read-back digest verification of /mnt1/apfs_invert_asr_img succeeded
sock  23: closed
restore_apfs_image_async : finished restoring an APFS image
[AsyncCP][End] Checkpoint restore_system_image finished with result: 0.
[AsyncCP][WaitEnd] Wait checkpoint restore_system_image success.
entering mount_filesystems
Not counting data volume as required.
Not counting user volume as required.
IODeviceTree:/arm-io/sep/iop-sep-nub/xART found
We should have an xART partition.
Attempting to preload_apfs_keys
entering force_unmount_partition
Data not mounted (unmount ignored)
entering ramrod_execute_command_with_config: /sbin/mount_apfs
executing /sbin/mount_apfs -R -o rdonly /dev/disk2s8 /mnt2
waiting for child to exit
child exited
exit status: 0
/dev/disk2s8 mounted on /mnt2
Calling APFS_FSCTL_UNMOUNT_CRYPTO_HINT on /dev/disk2s8
Preloading of APFS keys succeeded even when DATA_VOLUME_ACCESS_RESTRICTED is set to true...
entering ramrod_probe_media_internal
entering wait_for_device: 'EmbeddedDeviceTypeRoot'
[... 340 lines cut ...]
restore-step-user-progress = 100
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[00:32:32.0895-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x1103067B:65;0x110306A6:129;0x110306C2:133}
restore-step-names = {0x1103067B:perform_restore_installing;0x110306A6:install_splat;0x110306C2:cleanup_send_crash_logs}
restore-step-uptime = 344
restore-step-user-progress = 100
[00:32:32.0896-GMT]{3>6} CHECKPOINT END: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x1103067B:65;0x110306A6:129}
restore-step-names = {0x1103067B:perform_restore_installing;0x110306A6:install_splat}
restore-step-uptime = 344
restore-step-user-progress = 100
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[00:32:32.0896-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103067B:65;0x110306A6:129;0x11030648:134}
restore-step-names = {0x1103067B:perform_restore_installing;0x110306A6:install_splat;0x11030648:cleanup_send_final_status}
restore-step-uptime = 344
restore-step-user-progress = 100

ERROR: Unable to successfully restore device
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(failed to install splat)
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device

MilanBabuljak, 2025-06-27:
I'm having the same issue when trying to downgrade from iOS 26 back to 18

egorpol, 2026-10-11:
Same `install_splat` / `Failed preallocation due to space` failure here — fixed without data loss. Posting details in case it helps others.

**Setup**
- iPhone 13 (iPhone14,5 / d17ap), 128 GB
- Update 26.6.1 (23G83) → 27.0.1 (24A446)
- Finder failed twice with `Unknown error (100). 06A6.0064`, phone stuck on Apple logo
- idevicerestore `1.0.0-278-g4b3e847` (libirecovery 1.3.1-12-g93c117c, libtatsu 1.0.5-6-ge7d6ad1), Intel Mac, macOS 26

**Failure (Finder log, both attempts)**
```
Transfer: Failed preallocation of 6012534784 bytes for /mnt9/cryptex1/current/os.dmg: Unknown error: -1.
Transfer: Failed preallocation due to space. No hope in this succeeding, bailing.
CHECKPOINT FAILURE:(FAILURE:100) [0x06A6] install_splat [0]D(failed to install splat)
```
By then the new system volume was already installed and the Data volume mounted fine. Only the Cryptex was missing.

**Why it happens.** The device's `space_checks` step reads `MinimumSystemPartition` from the build identity and estimates:

```
needed ≈ MinimumSystemPartition × 1.05 + cryptex size − current System − current Preboot
```

(8572 × 1.05 = 9000.6 MB matches the logged `spaceForRestore(9437813145)` exactly.)

| | Attempt 1 | Attempt 2 |
|---|---|---|
| SpaceNeededForRestore | 2109 MB | 6052 MB |
| ContainerFreeSpace | 2507 MB | 6450 MB |
| Check | "Sufficient" | "Sufficient" |
| Cryptex prealloc (5734 MiB) | failed | failed |

The check passes with roughly 400 MB to spare, but the real write still fails. The Data volume is mounted "for free space checks/cleanup", yet purgeable space is apparently only cleaned when the check fails. iOS Settings had shown 8–10 GB "available", which includes purgeable data.

**Fix.** Unpacked the IPSW to a folder (idevicerestore accepts a directory) and changed only `BuildIdentities[Customer Upgrade Install (IPSW)].Info.MinimumSystemPartition` from 8572 to 10400. That forces the estimate to about 7970 MB, above the ~6450 MB free. The Erase identity was left untouched. Then:

```
idevicerestore -d --variant "Customer Upgrade Install (IPSW)" --logfile run.log ./IPSW_MOD
```

The device requested `BuildIdentityDict` during the restore, so the edited value reaches it. `install_splat` → `result=0`, `Status: Restore Finished`; the phone booted with all data intact.

**Notes**
- I couldn't capture the device-side space-check output for the successful run, so "failed check → cleanup → install proceeds" is inferred, not proven.
- In `restore.c`, `MinimumSystemPartition` → `SystemPartitionSize` is only set for macOS restores. For iOS the value only reaches the device via the requested build identity.
- Choose your value from your own log: aim for `needed` about 1–1.5 GB above `ContainerFreeSpace`. The approach comes from [this r/applehelp guide](https://www.reddit.com/r/applehelp/comments/1wsvwrm/guide_fixed_iphone_error_100_06a6_install_splat/), where an iPhone 13 mini was fixed with 8569 → 9956. Confirmed here on a standard iPhone 13.
- Never use `-e`. Check the early `This restore will update the device without erasing user data.` line before anything is sent.

Thanks to the author of that Reddit guide for the idea, and to the idevicerestore maintainers for the tool.