libimobiledevice/idevicerestore #739: M1 MBA: Device failed to enter restore mode.

halian, 2025-06-21:
My M1 MacBook Air started bootlooping and eventually showing the (!) screen after I tried to erase it from within the Mac OS 26 “Tahoe” beta, which was causing it to overheat. I don't have another Mac that can run Apple Configurator, so I've been trying `idevicerestore`, but my laptop will not reboot into restore mode once everything is sent over, no matter what version of Mac OS I try; it always reboots into DFU or recovery mode. Debug-level logs follow.

```
idevicerestore 1.0.0-231-ga5905b7-dirty (libirecovery 1.2.1-13-g638056a, libtatsu 1.0.5)
irecv_event_cb: device 001c19983a28a01e (udid: N/A) connected in DFU mode
Found device in DFU mode
ECID: 7909440949362718
Identified device as j313ap, MacBookAir10,1
Device Product Version: N/A
Device Product Build: N/A
* Host api.ipsw.me:443 was resolved.
* IPv6: 2606:4700:20::681a:89c, 2606:4700:20::ac43:47bb, 2606:4700:20::681a:99c
* IPv4: 172.67.71.187, 104.26.9.156, 104.26.8.156
*   Trying [2606:4700:20::681a:89c]:443...
* ALPN: curl offers h2,http/1.1
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384 / X25519MLKEM768 / id-ecPublicKey
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=ipsw.me
*  start date: Jun 19 04:47:53 2025 GMT
*  expire date: Sep 17 05:47:40 2025 GMT
*  subjectAltName: host "api.ipsw.me" matched cert's "*.ipsw.me"
*  issuer: C=US; O=Google Trust Services; CN=WE1
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
*   Certificate level 0: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA256
*   Certificate level 1: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA384
*   Certificate level 2: Public key type EC/secp384r1 (384/192 Bits/secBits), signed using sha256WithRSAEncryption
* Connected to api.ipsw.me (2606:4700:20::681a:89c) port 443
* using HTTP/2
* [HTTP/2] [1] OPENED stream for https://api.ipsw.me/v4/device/MacBookAir10,1
* [HTTP/2] [1] [:method: GET]
* [HTTP/2] [1] [:scheme: https]
* [HTTP/2] [1] [:authority: api.ipsw.me]
* [HTTP/2] [1] [:path: /v4/device/MacBookAir10,1]
* [HTTP/2] [1] [user-agent: InetURL/1.0 idevicerestore/1.0.0-231-ga5905b7-dirty]
* [HTTP/2] [1] [accept: */*]
> GET /v4/device/MacBookAir10,1 HTTP/2
Host: api.ipsw.me
User-Agent: InetURL/1.0 idevicerestore/1.0.0-231-ga5905b7-dirty
Accept: */*

* Request completely sent off
< HTTP/2 200 
< date: Sat, 21 Jun 2025 04:23:05 GMT
< content-type: application/json
< cf-ray: 9530ba80eb975c5f-MIA
< vary: Accept-Encoding
< vary: Origin
< access-control-allow-methods: GET, OPTIONS
< access-control-allow-origin: *
< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< x-content-type-options: nosniff
< x-data-location: database
< x-frame-options: DENY
< x-frame-options: DENY
< x-xss-protection: 1; mode=block
< strict-transport-security: max-age=15552000; preload
< last-modified: Sat, 21 Jun 2025 03:32:42 GMT
< cache-control: max-age=14400
< cf-cache-status: HIT
< age: 278
< report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=NvnvQL7UFmYp4vBkTeaLcgbCdDAN79aLgQTGUXg1FR%2BV3hVgtAUJXgLsKmkOH%2Fkb%2FpoIIRolgT4DI5k5yUyJJ5DVciKsPgKMv8Z15w8CuhRP7tGp%2FugaG0PcQYztUt03XC4MmqCnMyCp"}],"group":"cf-nel","max_age":604800}
< nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
< server: cloudflare
< alt-svc: h3=":443"; ma=86400
< server-timing: cfL4;desc="?proto=TCP&rtt=24047&min_rtt=22315&rtt_var=9605&sent=6&recv=7&lost=0&retrans=0&sent_bytes=4477&recv_bytes=1814&delivery_rate=181223&cwnd=252&unsent_bytes=0&cid=69b2c6da49c33d38&ts=50&x=0"
< 
* Connection #0 to host api.ipsw.me left intact
The following firmwares are currently being signed for MacBookAir10,1:
  [1] 15.5 (build 24F74)
[... 233 lines cut ...]
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (163153534 bytes)...
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.j313ap.im4p (Firmware/all_flash/DeviceTree.j313ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (51282 bytes)...
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j313.RELEASE.im4p (Firmware/all_flash/sep-firmware.j313.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (7500112 bytes)...
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.mac13g (kernelcache.release.mac13g)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (24191888 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 001c19983a28a01e (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 001c19983a28a01e (udid: N/A) connected in DFU mode
ERROR: Device failed to enter restore mode.

```

halian, 2025-06-21:
Duplicates #696. I got it to work by building `usbmuxd` from source instead of using the version of the AUR (please for the love of $DEITY do a formal release) and restored 15.0 successfully.

Abubakar-Waheed, 2025-07-22:
@halian I'm tired of trying and gets the same issue as yours. Built the usbmuxd from source as well. Can you guide how did you solve it or other recommendations?