libimobiledevice/idevicerestore #747: Restore Error 'Failed to install splat'

irainbw, 2025-07-30:
Hello all,

I encountered some problems when flashing 18.5 on iPhone 11. The log prompt (failed to install splat)

some info:

ramrod_dump_mounted_filesystem_info:**********DUMPING MOUNTED FILESYSTEMS********
ramrod_dump_mounted_filesystem_info: 3 filesystems are mounted
tmpfs is mounted at /mnt5
devfs is mounted at /dev
/dev/md0 is mounted at /
ramrod_dump_mounted_filesystem_info: *********DONE DUMPING MOUNTED FILESYSTEMS********
invert_apfs_image : inverting : /System/Library/Filesystems/apfs.fs/apfs_invert /dev/disk2 0 apfs_invert_asr_img
entering ramrod_execute_command_with_config: /System/Library/Filesystems/apfs.fs/apfs_invert
executing /System/Library/Filesystems/apfs.fs/apfs_invert -d /dev/disk2 -s 1 -n apfs_invert_asr_img -f
ASR: *** Mounting outer volume (/dev/disk2 s1)...
ASR: nx_mount:1308: disk2 initializing cache w/hash_size 32768 and cache size 65536
ASR: nx_mount:1454: disk2 container cleanly-unmounted flag set.
ASR: nx_mount:1630: disk2 checkpoint search: largest xid 37, best xid 37 @ 73
ASR: nx_mount:1657: disk2 stable checkpoint indices: desc 72 data 142
ASR: spaceman_datazone_init:611: disk2 allocation zone on dev 0 for allocations of 1 blocks starting at paddr 4096000
ASR: spaceman_datazone_init:611: disk2 allocation zone on dev 0 for allocations of 2 blocks starting at paddr 32768
ASR: spaceman_datazone_init:611: disk2 allocation zone on dev 0 for allocations of 3 blocks starting at paddr 65536
ASR: spaceman_datazone_init:611: disk2 allocation zone on dev 0 for allocations of 4 blocks starting at paddr 98304
ASR: spaceman_scan_free_blocks:4106: disk2 scan took 0.007333 s, trims took 0.000000 s
ASR: spaceman_scan_free_blocks:4110: disk2 13753997 blocks free in 35 extents, avg 392971.34
ASR: spaceman_scan_free_blocks:4119: disk2 13753997 blocks trimmed in 35 extents (0 us/trim, 35000000 trims/s)
ASR: spaceman_scan_free_blocks:4122: disk2 trim distribution 1:9 2+:10 4+:2 16+:3 64+:2 256+:9
ASR: spaceman_fxc_print_stats:477: disk2 dev 0 smfree 13753997/15624989 table 32/186 blocks 13753997 1:429812:11250545 100.00% range 31332:15593657 99.79% scans 1
ASR: *** Getting image dstream info...
ASR:     apfs_invert_asr_img: dstream_id=16, size=7377780736
ASR: *** Mounting inner volume (apfs_invert_asr_img)...
ASR: nx_mount:1308:  initializing cache w/hash_size 32768 and cache size 65536
ASR: nx_mount:1630:  checkpoint search: largest xid 452, best xid 452 @ 5
ASR: nx_mount:1657:  stable checkpoint indices: desc 4 data 9
ASR: *** Copying inner volume extentref tree into outer volume...
ASR:     Copied 45682 original extents and created 0 new extents
ASR: *** Copying inner volume fsroot tree into outer volume...
ASR:     Copied 1134073 fs_root records plus 0 new file extents
ASR: *** Cleaning up unused blocks...
ASR:     Freed 139630 data blocks and kept 1661586 data blocks
ASR: *** Updating the superblock...
ASR: *** Deleting old fs_root...
ASR: *** Deleting old extentref tree...
mDNS [      ]: no usable interfaces found
ASR: *** Finishing transaction...
ASR: *** Unmounting...
ASR: sanity_check_alloced_blocks:730: disk2s1 fs_alloc_count mismatch: fs root nodes 1 extent 1 omap 363 snap_meta 1 doc_id 0 prev_doc_id 0 fext: 0 pfkur: 0 er: 0 udata: 1661586 fs_alloc_count 1702536 != count 1661953
ASR: *** Success!
waiting for child to exit
child exited
exit status: 0
invert_apfs_image : succeeded inverting : /System/Library/Filesystems/apfs.fs/apfs_invert /dev/disk2 0 apfs_invert_asr_img
ASR succeed on initial attempt
IODeviceTree:/arm-io/sep/iop-sep-nub/xART found
We should have an xART partition.
entering mount_partition
entering ramrod_execute_command_with_config: /sbin/mount_apfs
executing /sbin/mount_apfs -R /dev/disk2s3 /mnt7
waiting for child to exit
child exited
exit status: 0
/dev/disk2s3 mounted on /mnt7
xART mounted read-write
entering ramrod_init_gigalocker
IODeviceTree:/arm-io/sep/iop-sep-nub/xART found
We should have an xART partition.
entering ramrod_execute_command_with_config: /usr/libexec/seputil
executing /usr/libexec/seputil --gigalocker-init
seputil: Gigalocker file (/mnt7/744A9F09-1B82-5C78-A173-00E5E8BCBCFA.gl) exists
[... 235 lines cut ...]
restore-step-user-progress = 98
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[00:35:14.0984-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x1103067B:66;0x1103132C:112;0x110306A6:130;0x110306C2:134}
restore-step-names = {0x1103067B:perform_restore_installing;0x1103132C:await_update_update_veridian;0x110306A6:install_splat}
restore-step-uptime = 330
restore-step-user-progress = 98
[00:35:14.0984-GMT]{3>6} CHECKPOINT END: RESTORED:[0x06C2] cleanup_send_crash_logs
restore-step-ids = {0x1103067B:66;0x1103132C:112;0x110306A6:130}
restore-step-names = {0x1103067B:perform_restore_installing;0x1103132C:await_update_update_veridian;0x110306A6:install_splat}
restore-step-uptime = 330
restore-step-user-progress = 98
Considering saving tolerated failures: true /mnt4/lastOTA/ota_tolerated_failures.plist false
[00:35:14.0984-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103067B:66;0x1103132C:112;0x110306A6:130;0x11030648:135}
restore-step-names = {0x1103067B:perform_restore_installing;0x1103132C:await_update_update_veridian;0x110306A6:install_splat}
restore-step-uptime = 330
restore-step-user-progress = 98

ERROR: Unable to successfully restore device
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(failed to install splat)
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device

themacintoshnerd, 2025-08-04:
Bumping this I have an unrestorable device thanks to this issue (iPad Air 5, iOS 26b4)

CrusherNotDrip, 2025-08-24:
I am also currently experiencing this problem. (iPhone 11 Pro Max, iOS 26b3 and now attempting to get out of restore mode by either reupdating to 26b3 or above so I don't lose my data as I have no backups :[ )

redoninho, 2025-10-25:
@CrusherNotDrip 
did you manage to fix it? I have a iphone 16 pro 256gb on 26.0.1 and i'm having a similar issue. I was thinking of updating to a beta version of the ios 26 update in hopes it will work and i can recover my data. I am in the same position as you

CrusherNotDrip, 2025-10-25:
> [@CrusherNotDrip](https://github.com/CrusherNotDrip) did you manage to fix it? I have a iphone 16 pro 256gb on 26.0.1 and i'm having a similar issue. I was thinking of updating to a beta version of the ios 26 update in hopes it will work and i can recover my data. I am in the same position as you

no unfortunately i haven't and i just went off by resetting my phone
only thing i lost was all my photos but the rest was backed up to icloud so

redoninho, 2025-10-25:
Thats a shame. I will take it into apple but don't really expect much. I have 11 years of photos that I would ideally like to save. Its pathetic that a trillion dollar company has such a common issue that wipes peoples phones. I have never heard of any other device doing something like this