libimobiledevice/idevicerestore #755: Mac Mini m4 failed to enter restore mode

Rowb, 2025-09-18:
**Environment**
- **Host OS**: Windows 11  
- **CPU**: AMD Ryzen 5700x  
- **Shell**: MSYS2 MinGW64  

**idevicerestore**
```
idevicerestore 1.0.0-255-ga0cec3b (libirecovery 1.3.0, libtatsu 1.0.5-2-gfc57e26)
```

**Library versions**
```
libusbmuxd-2.0 2.1.1-2-g93eb168
libimobiledevice 1.3.0-304-g54f6437
libirecovery 1.3.0
libtatsu 1.0.5-2-gfc57e26
libplist 2.7.0-3-g91ba426
```

---

**What I tried**
- Different USB cables (USB-C ↔ USB-C and USB-C ↔ USB-A)
- Different IPSW versions for my Mac (M4 / UniversalMac)

---

**Observed behavior**
- Restore process starts correctly (iBSS, iBEC, ramdisk, SEP, kernelcache all sent successfully).
- At the stage **“Waiting for device to enter restore mode…”**, device reboots, LED changes from red → white, Apple logo appears with empty progress bar.
- Process then fails with:
Device failed to enter restore mode.
Device reconnected in Recovery mode, most likely image personalization failed.
- Device stays stuck at Apple logo with empty bar.

---
```
09:22:29.565    <Info> idevicerestore 1.0.0-255-ga0cec3b (libirecovery 1.3.0, libtatsu 1.0.5-2-gfc57e26)
09:22:29.585    <Info> Found device in Port DFU mode
09:22:29.601    <Info> ECID: 1289542151176081965
09:22:29.601    <Info> Identified device as j773gap, Mac16,10
09:22:29.602    <Info> Device Product Version: N/A
09:22:29.602    <Info> Device Product Build: N/A
09:22:30.075    <Info> The following firmwares are currently being signed for Mac16,10:
09:22:30.076    <Info>   [1] 26.0 (build 25A354)
09:22:30.076    <Info>   [2] 15.6.1 (build 24G90)
09:22:30.077    <Info>   [3] 15.6 (build 24G84)
09:22:30.078    <Info>   [4] 15.5 (build 24F74)
09:22:30.078    <Info>   [5] 15.4.1 (build 24E263)
09:22:30.079    <Info>   [6] 15.4 (build 24E248)
09:22:30.079    <Info>   [7] 15.3.2 (build 24D81)
09:22:30.080    <Info>   [8] 15.3.1 (build 24D70)
09:22:30.081    <Info>   [9] 15.3 (build 24D60)
09:22:30.081    <Info>   [10] 15.2 (build 24C101)
09:22:30.082    <Info>   [11] 15.2 (build 24C2101)
09:22:30.082    <Info>   [12] 15.1.1 (build 24B2091)
09:22:30.083    <Info>   [13] 15.1 (build 24B2083)
09:22:56.447  <Notice> Selected firmware 15.2 (build 24C101)
09:22:56.448    <Info> Verifying 'UniversalMac_15.2_24C101_Restore.ipsw'...
09:24:03.371    <Info> Checksum matches.
09:24:03.377    <Info> Extracting BuildManifest from IPSW
09:24:04.501    <Info> Received USBPortController1,Ticket
09:24:04.502    <Info> Creating Ace3Binary
09:24:04.505    <Info> Sending data (39865 bytes)...
09:24:09.421    <Info> Port DFU restore successful. Continuing.
09:24:09.422    <Info> IPSW Product Version: 15.2
09:24:09.422    <Info> IPSW Product Build: 24C101 Major: 24
09:24:09.425    <Info> Device supports Image4: true
09:24:09.426    <Info> Variant: Customer Erase Install (IPSW)
09:24:09.426    <Info> This restore will erase all device data.
[... 85 lines cut ...]
09:25:10.791    <Info> Personalizing IMG4 component RestoreTrustCache...
09:25:10.792    <Info> Sending RestoreTrustCache (21373 bytes)...
09:25:10.795    <Info> Extracting SmartIOFirmware_ASCv7.im4p (Firmware/SmartIOFirmware_ASCv7.im4p)...
09:25:10.804    <Info> Personalizing IMG4 component SIO...
09:25:10.805    <Info> Sending SIO (1163914 bytes)...
09:25:10.839    <Info> Extracting iBootData.j773g.RELEASE.im4p (Firmware/all_flash/iBootData.j773g.RELEASE.im4p)...
09:25:10.846    <Info> Personalizing IMG4 component iBootData...
09:25:10.847    <Info> Sending iBootData (10471 bytes)...
09:25:10.850    <Info> ramdisk-size=0x20000000
09:25:10.850    <Info> Extracting 044-81195-001.dmg (044-81195-001.dmg)...
09:25:11.161    <Info> Personalizing IMG4 component RestoreRamDisk...
09:25:11.191    <Info> Sending RestoreRamDisk (182462601 bytes)...
09:25:18.106    <Info> Extracting DeviceTree.j773gap.im4p (Firmware/all_flash/DeviceTree.j773gap.im4p)...
09:25:18.112    <Info> Personalizing IMG4 component RestoreDeviceTree...
09:25:18.113    <Info> Sending RestoreDeviceTree (55822 bytes)...
09:25:18.118    <Info> Extracting sep-firmware.j773g.RELEASE.im4p (Firmware/all_flash/sep-firmware.j773g.RELEASE.im4p)...
09:25:18.129    <Info> Personalizing IMG4 component RestoreSEP...
09:25:18.131    <Info> Sending RestoreSEP (5879033 bytes)...
09:25:18.277    <Info> Extracting kernelcache.release.mac16g (kernelcache.release.mac16g)...
09:25:18.320    <Info> Personalizing IMG4 component RestoreKernelCache...
09:25:18.326    <Info> Sending RestoreKernelCache (29319881 bytes)...
09:25:19.808    <Info> Waiting for device to enter restore mode...
09:27:48.527   <Error> Device failed to enter restore mode.
09:27:48.528   <Error> Device reconnected in Recovery mode, most likely image personalization failed.
```

nikias, 2025-09-18:
Ok that makes sense now, I guess that the Windows driver does not support Mac devices.

Abubakar-Waheed, 2025-09-27:
@Rowb My issue is also same. I will try with latest commit this weekend but I tried every possible thing in the universe and error was the same as yours never got solved

#746

Abubakar-Waheed, 2025-10-07:
@Rowb I have tried like 15-20 times using apple configurator and it never works. Can you tell me which IPSW verison, software versions for idevice restore libusb etc you used so I can try the same?

My logs are here #746   I shared the issue like 2 months ago. Mac is currently sitting in cabinet presumly dead.

Rowb, 2025-10-07:
The environment I was working in is described in the issue — it was on Windows.
I also tried on macOS 15.2 (you can see that in the logs).
Using a MacBook, I managed to get it working with version 15.4, if I remember correctly.

The main problem on Windows was that the driver couldn’t detect the DFU reconnection, as mentioned above.
I’m not sure whether that’s a driver issue or if my version of Windows 11 (not the latest one) simply doesn’t support the Mac mini.

Have you tried with the latest version? I think they’ve fixed something related to that.

Rowb, 2025-10-07:
@Abubakar-Waheed The environment I was working in is described in the issue — it was on Windows.
I was trying to restore using the macOS 15.2 IPSW (you can see that in the logs).
Using a MacBook, I managed to get it working with version 15.4, if I remember correctly.

The main problem on Windows was that the driver couldn’t detect the DFU reconnection, as mentioned above.
I’m not sure whether that’s a driver issue or if my version of Windows 11 (not the latest one) simply doesn’t support the Mac mini.

Have you tried with the latest version? I think they’ve fixed something related to that.

Abubakar-Waheed, 2025-11-07:
@StormYudi Can you guide me on how you used the the docker version to solve it. Please lay out step by step so I can test it. I have been stuck with the issue for the past 3 months and laptop is sitting idle there waiting for some miracle.

StormYudi, 2025-11-11:
> > > [@StormYudi](https://github.com/StormYudi) Can you guide me on how you used the the docker version to solve it. Please lay out step by step so I can test it. I have been stuck with the issue for the past 3 months and laptop is sitting idle there waiting for some miracle.
> > 
> > 
> > It is mentioned in the README: https://github.com/libimobiledevice/idevicerestore/blob/master/README.md#docker
> 
> [@nikias](https://github.com/nikias) [@StormYudi](https://github.com/StormYudi) I tried with docker 100% followed the instructions, built using build.sh and after that ran it with the most latest ipsw file and it always says "Failed to found device". While locally it does detect device on "lsusb". It doesn't detect the device at all in the docker mode. Tried consulting claude on that and it tried various commands but it failed.
> 
> I also tried again to do it locally and compiled everything 100% from scratch and it has the most latest version and it still fails at kernalcache.
> 
> So tired of trying I have put at least 80+ hours in total into this issue and it never solves. One thing I know is that it has to do with usbmuxd but I run it correctly with all privileges and latest version but it still doesn't work.

Have you tried compiling using a Debian/Ubuntu virtual machine? (Perhaps the best option is Ubuntu with Docker). I see you're using Fedora; I tried it before, but encountered problems and gave up immediately. However, I don't have many operating systems to choose from on my Raspberry Pi ;)

anticultist, 2026-09-26:
Confirming @themodev's workaround on a **Mac mini M4 (Mac16,10, j773gap)**: with a separate usbmuxd instance (`-p`) and the `UNIX:` socket prefix, a dry run with `-z` gets to restore mode without the "image personalization failed" error. I didn't do an actual restore; I only tested up to `-z`.

**Setup:** Ubuntu 24.04.5 (kernel 7.0), all libs built from git HEAD (2026-09-25):
`idevicerestore 1.0.0-276-g60192e9 (libirecovery 1.3.1-12-g93c117c, libtatsu 1.0.5-6-ge7d6ad1)`, usbmuxd `3ded00c`. The distro usbmuxd was masked. I used a USB-C cable directly into the Mac's middle port, with no hub.

```sh
sudo usbmuxd -f -v -v -p -C /tmp/dfu-usbmuxd-config -S /tmp/dfu-usbmuxd.sock -P /tmp/dfu-usbmuxd.pid
sudo env USBMUXD_SOCKET_ADDRESS=UNIX:/tmp/dfu-usbmuxd.sock \
  idevicerestore -e -z -d UniversalMac_27.0_26A428_Restore.ipsw
```

```
Found device in Port DFU mode
Port DFU restore successful. Continuing.
Variant: Customer Erase Install (IPSW)
...
Sending RestoreKernelCache (33670024 bytes)...
Waiting for device to enter restore mode...
Device is now in restore mode. Exiting as requested.
```

I can post full `-d` logs if that helps.

anticultist, 2026-11-02:
Follow-up: the full restore works too. I swapped in a third-party 1 TB SSD module (Expand Mac mini) and restored macOS 27.0 onto it from Linux. I used the same setup as in my previous comment, just without `-z` this time. The restore ended with Status: `Restore Finished` and `DONE`, and after setup macOS sees the full capacity.

While the filesystem was being sent, the output kept showing `Unknown operation (80)` and similar lines with other numbers (79, 82, 84, 85, 89, 96). These are progress steps that idevicerestore has no name for, and they didn't affect the restore.