libimobiledevice/idevicerestore #756: iPhone 17 Pro flash error

xlzhen-940218, 2025-09-19:
[===============               Sending Baseband TSS request...
[===============                                   ]  31.0% restore_send_firmware_updater_data: Got unknown updater name 'Centauri', trying to discover from device generated request.
Sending Centauri TSS request...
Received Baseband SHSH blobs
Received Wireless1,Ticket
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
[================                                  ]  32.0% Size mismatch when parsing MBN file. Continuing anyway.

nikias, 2025-09-19:
This is currently broken.

pulojrodrigues76-cmyk, 2025-09-20:
> [=============== Sending Baseband TSS request... [=============== ] 31.0% restore_send_firmware_updater_data: Got unknown updater name 'Centauri', trying to discover from device generated request. [...]

pekn, 2025-09-24:
iPhone 17 base model seems to have similar problems.

- New 'Centauri' updater name
- Savage TSS request requires changes (now it's Jasmine)
- Baseband structure has changed: along with baseband ticket, there are some data keys (Misc-HashTableBody-Blob, RestoreSBL1-HashTableBody-Blob, SBL1-HashTableBody-Blob, TME-HashTableBody-Blob) included. It looks like Misc-HashTableBody-Blob goes to multi_image.mbm, RestoreSBL1-HashTableBody-Blob goes to restorexbl_sc.elf, SBL1-HashTableBody-Blob goes to xbl_sc.elf and TME-HashTableBody-Blob goes to signed_firmware_soc_view.elf. However, there are additional changes to signed_firmware_soc_view.elf.. I have not figured out where those changes come from.

nikias, 2025-09-24:
I have it running. Stay put for a commit coming soon.

ze-tomreid, 2025-09-26:
@nikias - thanks for the heads up that this is completed, I'm also interested. I don't see a commit in this repo since the issue was closed? Was it a commit in a different repo that fixed this - if so, where?

Thanks again!
Tom

ze-tomreid, 2025-09-26:
@nikias - thanks, I see the commit now and will check it out on Monday.