libimobiledevice/idevicerestore #757: IPhone 17 Pro firmware restoration failed。

duyi56432, 2025-09-21:
Error Domain=FirmwareRestorer Code=2001 
progress: 0 0.800000
Could not make sure this firmware is suitable for the current device. Refusing to continue.

pekn, 2025-09-26:
This is duplicate of https://github.com/libimobiledevice/idevicerestore/issues/756

nikias, 2025-09-26:
Please check with latest commit.

nikias, 2025-09-29:
Can you drop a diff here? Thanks.

pekn, 2025-09-29:
diff --git a/src/mbn.c b/src/mbn.c
index 2ba5e65..aad19b3 100644
--- a/src/mbn.c
+++ b/src/mbn.c
@@ -263,18 +263,18 @@ void* mbn_stitch(const void* data, size_t data_size, const void* blob, size_t bl
 
 	off_t stitch_offset = data_size - blob_size;
 	if (stitch_offset + blob_size > data_size) {
-		logger(LL_ERROR, "%s: stitch offset (0x%llx) + size (0x%zx) is larger than the destination (0x%zx)\n", __func__, stitch_offset, blob_size, data_size);
+		logger(LL_ERROR, "%s: stitch offset (0x%lx) + size (0x%zx) is larger than the destination (0x%zx)\n", __func__, stitch_offset, blob_size, data_size);
 		return NULL;
 	}
 
-	void* buf = malloc(data_size);
+	unsigned char* buf = malloc(data_size);
 	if (buf == NULL)	{
 		logger(LL_ERROR, "out of memory\n");
 		return NULL;
 	}
 
 	memcpy(buf, data, data_size);
-	logger(LL_DEBUG, "%s: stitching mbn at 0x%llx, size 0x%zx\n", __func__, stitch_offset, blob_size);
+	logger(LL_DEBUG, "%s: stitching mbn at 0x%lx, size 0x%zx\n", __func__, stitch_offset, blob_size);
 	memcpy(buf + stitch_offset, blob, blob_size);
 
 	return buf;
@@ -409,7 +409,7 @@ void* mbn_mav25_stitch(const void* data, size_t data_size, const void* blob, siz
 	}
 
 	if (sect_off + sect_size > data_size) {
-		logger(LL_ERROR, "%s: section (0x%llx+0x%zx) is bigger than the data\n", __func__, sect_off, sect_size);
+		logger(LL_ERROR, "%s: section (0x%lx+0x%zx) is bigger than the data\n", __func__, sect_off, sect_size);
 		return NULL;
 	}
 
@@ -441,9 +441,8 @@ void* mbn_mav25_stitch(const void* data, size_t data_size, const void* blob, siz
 		new_metadata_size + src_header->hash_table_size;
 	size_t new_oem_sig_and_cert_chain_size =
 		src_header->oem_signature_size + src_header->oem_certificate_chain_size;
-	off_t new_oem_sig_and_cert_chain_off = new_metadata_and_hash_table_size +
-																				 header->qti_signature_size +
-																				 header->qti_certificate_chain_size;
+	off_t new_oem_sig_and_cert_chain_off = new_metadata_and_hash_table_size + 
+		header->qti_signature_size + header->qti_certificate_chain_size;
 
 	if (new_metadata_and_hash_table_size > blob_size) {
 		logger(LL_ERROR, "%s: new metadata (0x%zx) and hash table (0x%x) are bigger than the source (0x%zx)\n", __func__, new_metadata_size, src_header->hash_table_size, blob_size);
@@ -465,16 +464,16 @@ void* mbn_mav25_stitch(const void* data, size_t data_size, const void* blob, siz
 		return NULL;
 	}
 
-	void* buf = malloc(data_size);
+	unsigned char* buf = malloc(data_size);
 	if (buf == NULL)	{
 		logger(LL_ERROR, "out of memory\n");
 		return NULL;
 	}
 
 	memcpy(buf, data, data_size);
-	logger(LL_DEBUG, "%s: stitching mbn at 0x%llx (0x%zx bytes)\n", __func__, sect_off, new_metadata_and_hash_table_size);
+	logger(LL_DEBUG, "%s: stitching mbn at 0x%lx (0x%zx bytes)\n", __func__, sect_off, new_metadata_and_hash_table_size);
 	memcpy(buf + sect_off, blob, new_metadata_and_hash_table_size);
-	logger(LL_DEBUG, "%s: stitching mbn at 0x%llx (0x%zx bytes)\n", __func__, sect_off + new_oem_sig_and_cert_chain_off, new_oem_sig_and_cert_chain_size);
+	logger(LL_DEBUG, "%s: stitching mbn at 0x%lx (0x%zx bytes)\n", __func__, sect_off + new_oem_sig_and_cert_chain_off, new_oem_sig_and_cert_chain_size);
 	memcpy(buf + sect_off + new_oem_sig_and_cert_chain_off, blob + new_metadata_and_hash_table_size, new_oem_sig_and_cert_chain_size);
 
 	return buf;
diff --git a/src/restore.c b/src/restore.c
index f2fd9be..f193de8 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -2016,8 +2016,6 @@ static int restore_sign_bbfw(const char* bbfwtmp, plist_t bbtss, const unsigned
 					goto leave;
 				}
 			}
-			free(buffer);
-			buffer = NULL;
 
 			blob_size = 0;
 			blob = (const unsigned char*)plist_get_data_ptr(node, &blob_size);
@@ -2043,18 +2041,23 @@ static int restore_sign_bbfw(const char* bbfwtmp, plist_t bbtss, const unsigned
 				fls = NULL;
 			} else if (bb_chip_id == 0x1F30E1) { // Mav25 - Qualcomm Snapdragon X80 5G Modem
 				fdata = mbn_mav25_stitch(buffer, zstat.size, blob, (size_t)blob_size);
+				fsize = zstat.size;
 				if (!fdata) {
 					logger(LL_ERROR, "Could not stitch %s\n", signfn);
 					goto leave;
 				}
 			} else {
 				fdata = mbn_stitch(buffer, zstat.size, blob, (size_t)blob_size);
+				fsize = zstat.size;
 				if (!fdata) {
 					logger(LL_ERROR, "Could not stitch %s\n", signfn);
 					goto leave;
 				}
 			}
 
+			free(buffer);
+			buffer = NULL;
+
 			zs = zip_source_buffer(za, fdata, fsize, 1);
 			if (!zs) {
 				logger(LL_ERROR, "out of memory\n");

nikias, 2025-09-29:
Fixed with 551dba5

closed 2025-09-29 by nikias

nikias, 2025-09-29:
Yes, I reopened it so people reporting this can report if this now actually works or not :)

nikias, 2025-10-03:
I just yesterday pushed a commit that should fix the baseband restore, and then the activation should also work fine.